lumapu/ahoy #1228: MQTT keeps on reporting value when Inverter went offline (battery usage)

tuxianerDE, 2023-11-13:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

### Version / Git SHA: 
Version: 0.7.36
Github Hash: ba218ed

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

I am using MQTT to report the data to my InfluxDB. I set the interval in the web interface to 5s so that constant data get pushed instead of on demand when they get pulled.

I am operating my HM-1500 with a battery pack, hence it happens that in the middle of the night that batteries run out of power and thus drops to 0. In this case the DTU still publishes data to the MQTT stating the last value in my case last night around 125W of input from WR, which of course is wrong as the connection to the Hub when Offline and of course no data went into the grid.

My question/suggestion: Would it be possible to alter the behaviour that in this setting MQTT stop publishing data i.e. inverter is offline hence no reason to keep on publishing? I do appreciate the regular interval at which data get published especially for the generation of usage data down the road, I am reluctant to the publishing interval back to 0 just for that.

juepi, 2023-12-08:
I have the same issue, however imho the problem should be specified a bit different (i am currently on dev build **0.8.14**): as far as i can tell, the problem is, that AhoyDTU does not update all MQTT inverter related topics to `0` when an inverter goes offline, even when the checkbox _Reset values when inverter status is 'not available'_ is enabled.
I assume Ahoy-DTU will probably keep publishing the last values via MQTT if a fixed (non-zero) send-interval is configured for MQTT (which i guess causes the issue described by @tuxianerDE ).

**Example:**
* Inverter online and producing, `<inverter-name>/ch0/P_AC` constantly updated, `<inverter-name>/available` = `2`
* DC-supply is cut
* after 5 minutes, `<inverter-name>/available` topic changes from `2` to `4`, `<inverter-name>/ch0/P_AC` topic not updated
* after 15min `<inverter-name>/available` topic changes to `0`, `<inverter-name>/ch0/P_AC` **not** updated

Imho Ahoy-DTU should reset/update all MQTT power-related topics of the inverter to `0` when the `available` state changes to 0 when _Reset values when inverter status is 'not available'_ is enabled (discussible imho if an offline-inverter should be assumed to not produce at any time, even if _Reset values..._ not set?)

lumapu, 2024-02-29:
there was a recent change on this "reset values if the inverter isn't available". Maybe you can test again

stefan123t, 2024-10-29:
@tuxianerDE @juepi are you still facing the issue in v0.8.152 or can this be closed based on the changes that lumapu mentioned ?

juepi, 2024-10-30:
Can't check it at the moment as i'm not at home. My MQTT interval is set to '0' and home automation script that manually updates AC power to '0' when the inverter   status is not '2'.
Will manually check it asap and let you know.

Yours,
Juergen

tuxianerDE, 2024-10-30:
Hey,
will not be able to check leave it then to @juepi as I switched to OpenDTU.