lumapu/ahoy #1229: enhancement - Anzeigen der USB-Versorgungsspannung im GUI

Thomas-SW, 2023-11-13:
Hallo alle.

Ich hatte Probleme mit der Funktionalität der Hardware (Wemos D1 mini & NRF24L01+), was definitiv auf einen Fehler in der Stromversorgung zurückzuführen war. Es wurden vom defekten USB-Steckernetzteil nur noch schwankende 4 Volt geliefert...
Hat etwas gedauert, bis ich darauf gekommen bin.

Daher meine Frage, ob eine einfache Anzeige der USB-Versorgungsspannung in der Live-Ansicht mit realisiert werden kann. Die muss weder besonders genau sein noch häufig aktualisiert werden. Beispiel: einfach "V USB = 4,9 V" irgendwo unten im Status anzeigen. Beim ESP8266 könnte dazu A0 (analogRead(A0), kann 0 - 1 Volt messen) mit einem Spannungsteiler (z.B. zwei Widerstände 820k / 180k) beschaltet werden, und natürlich muss ein Modul mit im Code zum Auswerten / Rechnen / Anzeigen hinzugefügt werden...

Leider kann ich nur die Hardware, nicht den Code, daher meine Anfrage hier.

You69Man, 2023-11-13:
Zur Prüfung von USB Versorgung hat sich bei so ein Teil bewährt: https://www.amazon.de/Retoo-Multi-function-Tester-Multimeter-Anzeige/dp/B09QXLS5JP/ref=mp_s_a_1_4?adgrpid=70002040534&hvadid=352703111932&hvdev=m&hvlocphy=9062702&hvnetw=g&hvqmt=e&hvrand=16550981709529472011&hvtargid=kwd-300022150920&hydadcr=25005_1831550&keywords=usb+spannungsmesser&qid=1699909453&sr=8-4

Thomas-SW, 2023-11-13:
So etwas habe ich sogar genutzt (von Drok) - aber die Versorgungsspannung zu verdächtigen muss man auch erst mal drauf kommen. Da der ESP das kann, wäre es ja relativ einfach - und wir sind hier ja auch thematisch bei der Abfrage von elektrischen Werten.

kiu77, 2023-11-14:
Ich denke, man kann nicht für jede Eventualität eine Funktion in die Software einbauen. Eierlegende Wollmilchsäue hat die Evolution schließlich auch nicht zustande gebracht.. Dem Messen von USB-Stromversorgungen (Spannung und Strom) gibt es entsprechende Zwischenstecker für wenig Geld.

Thomas-SW, 2023-11-14:
Mit einem Zwischenstecker kann man aber nicht prüfen, ob am ESP direkt die Spannung stimmt.
Sicherlich ist dies keine wichtige Funktion, aber immerhin wird eine mögliche Fehlerquelle einfach überprüft.

Wenn es ohne Beeinträchtigung der eigentlichen Funktionalität möglich ist, warum also nicht?
Es ist dazu nach meiner Einschätzung kein wirklich großer Aufwand notwendig...

kiu77, 2023-11-14:
Wo genau meinst Du? Am Chip selbst? an welchen Pins?
Und welche? Wirklich 5V? Warum nicht 3,3V - die liegen nämlich am Chip an.
Und das ist nicht alles, was für korrektes Funktionieren erforderlich ist.
Ganz so trivial ist das nicht.
Ich bin nicht der Ansicht, dass es sinnvoll ist, alle denkbaren Funktionen zu implementieren. Die wichtigen Basis-Funktionen sind genug. So unendlich viel Speicher steht da nicht zur Verfügung. Man muss die Ressourcen nach Notwendigkeit verteilen. Spannungsüberwachung gehört m:E. nicht dazu.

Thomas-SW, 2023-11-14:
Wenn die Ressourcen nicht reichen, ist das klar, kann ich aber nicht einschätzen - es soll aber kein Präzisionsmessgerät werden. Und welche Spannung? Die, die anfällig ist: USB 5V auf der Platine direkt- kann aber diskutiert werden. In meiner Erfahrung spielen unzureichende Spannungen in der Versorgung eben eine besondere Rolle, hatte schon zu viele Defekte bzw, Funktionsstörungen damit.

kiu77, 2023-11-14:
Woher willst Du wissen, ob die 5V anfällig sind? Weil das bei Dir so war?
Liest Du hier verstärkt Berichte mit Problemen bei der 5V-Leitung?
Dein Problem ist kein allgemeines Problem, was einer Lösung für die Allgemeinheit benötigen würde.
Ich gebe meine "Stimme" explizit gegen Deinen Wunsch ab.

Thomas-SW, 2023-11-14:
Fehlinterpretation meiner Aussagen.

kiu77, 2023-11-14:
Dieses Statement kannst Du mit viel Erfolg auf Dich selbst anwenden. Was soll das?
Jemand sieht das komplett anders. Das ist alles. 
Damit siehst Du, dass Dein Wunsch nicht auf ungeteilte Zustimmung stößt und
dass es ernstzunehmende Gegenargumente gibt - auf wenn sie Dir nicht gefallen.

Thomas-SW, 2023-11-14:
Es reicht doch, wenn Du etwas dagegen hast und dies auch sagst. Meine Erfahrungen, die sich augenscheinlich auf mehr als dieses Projekt beziehen, solltest Du doch aber bitte nicht in Frage stellen.

kiu77, 2023-11-14:
Sorry, was soll das?
Nicht die Worte verdrehen.
Ich stelle nicht Deine "Erfahrungen" in Frage, sondern Deine Schlussfolgerungen und die Sinnhaftigkeit Deines Vorschlags.
Und damit musst Du leben.
Du hast meine Aussage ja dauernd gekontert und jetzt wirst Du auch noch unleidlich. Das ist nicht in Ordnung.

Thomas-SW, 2023-11-14:
Nein, das wollte ich nicht, bitte nicht falsch verstehen.
Meine Erfahrung ist eben, was Versorgung angeht, eine völlig andere als Du scheinbar gemacht hast, nur das ist mir wichtig zu sagen.

kiu77, 2023-11-14:
Ich habe bisher schon fast 50 ESP32 verbaut und bei mir ist das nicht einmal aufgetreten.
Soviel zu meinen "unhinterfragbaren" Erfahrungen damit.

leliep, 2023-11-14:
Vielleicht einfach ein vernünftiges Netzteil kaufen…? ;-)

Thomas-SW, 2023-11-14:
Nun ja, das ist doch alles richtig und ich stelle das auch gar nicht in Frage.
Sicherlich ist die Wahrscheinlichkeit einer Störung auch sehr überschaubar, aber wie ich schon sagte: wenn trivial und nicht ressourcenfressend, warum denn nicht? Das ist es doch, was ein objektives NoGo wäre. Ist das denn der Fall?

;-) PS.: das Netzteil soll(te) hochwertig sein, und ist es auch, wenn ich das Innenleben ansehe. Sogar echtes CE und VDE, SK II wird eingehalten... Aber hat auch nichts genutzt. Daher war ich auch (wieder einmal mehr) überrascht über dieses unerwartete Problem. Habe schon ~50 Jahe damit zu tun, mir reicht es halt :))

kiu77, 2023-11-14:
Es gibt kein "objektives NoGo".
Aber es gibt schon gar kein "objektives Go".
Dass es Dir "reicht" ist kein akzeptabler Grund.

Thomas-SW, 2023-11-14:
Ich hätte nicht gedacht, dass ein "akzeptabler Grund" vorliegen muss, um etwas mit aufzunehmen.
Hätte erwartet, dass der Spaß am Projekt, Ideen und die technischen Möglichkeiten eher entscheidend sind.
Muss ich meine Erwartungshaltung wohl justieren.

kiu77, 2023-11-14:
wäre ne Idee...

lumapu, 2023-11-18:
ich sehe ein Problem mit erhöhtem Supportaufwand. Auf der anderen Seite verstehe ich auch, warum du die Funktion gerne hättest. Bis auf weiteres würde ich davon absehen weitere Hardware zu definieren - die meisten sind mit den wenigen Drähten schon überfordert.
Wenn ich das einbauen sollte, wären da noch einige andere Dinge, die auch in die Richtung gehen, ich denke gerade zB. an Taster um das Display ein / auszuschalten. Neuestes Feature was ich gehört habe war ein PIR Sensor um das Display bei Näherung einzuschalten. Man sieht die Möglichkeiten sind schier unbegrenzt, die Setup-Seite wird aber zunehmend unübersichtlicher.

Thomas-SW, 2023-11-25:
Hallo lumapo, danke für die sachliche Antwort.
Die Problematik mit der zusätzlichen Hardware und dem Support kann ich durchaus nachvollziehen.
Hatte das vor meiner Frage auch bedacht, und das ist damit bestätigt. Aber immerhin wollte ich das "Problem" und die Idee dazu mal nennen. Vielleicht kommt ja irgendwann einmal die "Eierlegende Wollmilchsau" ;-)
Danke für das Projekt, und frohes Schaffen!

closed 2023-12-29 by lumapu