lumapu/ahoy #1230: Inverter option 'Include inverter to sum of total' not working

SilverSurfer2000, 2023-11-14:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.7

### Github Hash

?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

3 inverters
MQTT enabled
Inverter intervall 5s

### Error description

I unchecked the new option 'Include inverter to sum of total (should be checked by default)' in the inverter settings, but the values are still added to the totals section (AC Power, Yield, etc.). Save and reboot was done.

knickohr, 2023-11-14:
Blöd gefragt, den Reboot haste aber gemacht ?

SilverSurfer2000, 2023-11-14:
Freilich, bereits mehrfach.

closed 2023-11-14 by SilverSurfer2000