lumapu/ahoy #1231: Ahoy mit 6 Wechselrichter

Oxbow6, 2023-11-14:
Hallo 
Ich habe seit ich von 4 auf 6 HM 1500 erweitert habe das Problem dass die Kommunikation nicht mehr so stabiel fubktioniert. Kennt jemand das Problem ? Habe jetzt zum testen den Abfrageintervall auf 60 Sekunden geschraubt. 
Aber das hilft leider nichts. Gibt es sonst noch eine Einstellung mit der ich testen könnte ? 

Gruss

Inverter #0: OST (v10018) is available and is producing
Inverter #1: WEST (v10018) is available and is producing
Inverter #2: WEST2_WR3 (v10018) is not yet available
-> last successful transmission: 2023-11-14, 10:54:11
Inverter #3: OST2_WR4 (v10018) is available and is producing
Inverter #4: OST-WEST_WR5 (v10018) is available and is producing
Inverter #5: OST-WEST_WR6 (v0) is available and is producing

## Hardware
  * [x ] ESP8266

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 7.3.6
Github Hash: _______

### Build & Flash Method:
* [ x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-11-16:
Achtung !

In den neueren Versionen werden mit dem ESP8266 nur noch max. 2 Inverter abgefragt. Ich würde Dir empfehlen gleich auf einen ESP32 zu gehen.

mtet2, 2023-11-17:
In der 0.8.8 kann man beim ESP8266 bis zu 4 Inverter eintragen.
Ich habe noch nicht getestet, ob auch 4 Inverter abgefragt werden. Ich gehe aber mal davon aus. ;-)
> 4 Inverter = ESP32

knickohr, 2023-11-17:
Bis zu 16

DanielR92, 2023-12-03:
Ich geh mal davon aus das @Oxbow6 Frage beantwortet wurde.
Wenn nicht nochmal öffnen. Danke :)

closed 2023-12-03 by DanielR92