lumapu/ahoy #1232: 0.8.7 Edit Inverter keine Änderungen möglich: "error while saving"

elisenjens, 2023-11-14:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.7

### Github Hash

1bc3a0f

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard

### Error description

"error while saving"

mtet2, 2023-11-14:
"error while saving" tritt auch bei mir auf.  (0.8.7 ESP8266)
Und immer noch keine Daten von den Invertern. (in der 0.8.3 Funktioniert es nach wie vor)

Muß man eventuell mit einem "Ahoy-Factory-Reset" alles neu eingeben?
Probiere ich mal, wenn ich wieder zu Hause bin.

mtet2, 2023-11-16:
ESP8266
Nach einem "Ahoy-Factory-Reset" alles neu eingeben geht nicht, da immer noch der "error while saving" Fehler zuschlägt.
Den Sicherungsfile von z.B. 0.8.3 importieren funktioniert.
Daten von den Invertern werden empfangen aber gefühlt dauert es 3 bis 4 mal so lange bis die Daten zu sehen sind.
(verglichen mit der 0.8.3)
Ein Inverter im Keller und der andere auf dem Dach. Die DTU in der Mitte. Nicht ganz optimal, aber ging immer.
Inverter löschen funktioniert in der 0.8.7 auch nicht.

Gubi2023, 2023-11-16:
hast du es mal mit dem Handy versucht? damit kann ich die Änderungen speichern, nur am PC bekomme ich die Fehlermeldung....

mtet2, 2023-11-16:
Das Handy verhält sich bei mir identisch mit dem Browser auf dem Rechner.
Aber einen Versuch war es wert ;-)

lumapu, 2023-11-16:
könnte ich mir auch nicht erklären .. ich schaue was ich machen kann

Gubi2023, 2023-11-16:
velleicht doch ein Browserproblem? siehe https://github.com/lumapu/ahoy/issues/1221#issuecomment-1814003316

lumapu, 2023-11-16:
nein, ich bin grad am Testen. Es kommt unmittelbar `error while saving`, ohne jegliche Kommunikation zu Ahoy (der Browser sollte eigentlich ein Paket rausschicken, tut es aber nicht)

mtet2, 2023-11-16:
Im Firefox 119.0.1 mit "ctrl F5" die Seite neu aufgebaut bringt keine Linderung ;-)

lumapu, 2023-11-16:
Das Problem ist irgendwie beim `AsyncWebserver` wo genau verstehe ich noch nicht. Er empfängt die Nachricht vom Browser immer in zwei Teilen, wobei der erste immer 98 Zeichen lang ist (wenn ich zB. den Invertername ändere, dann sind es trotzdem 98 Zeichen).
Durch den 'Split' kann das empfangene JSON nicht geparst werden und der Fehler kommt. Bleibt spannend

```
I: 98: 
{"cmd":"save_iv","id":0,"ser":19110224281945,"name":"HM1200.1","en":true,"ch":[{"pwr":"405","name"
I: 234:
:"","yld":"0"},{"pwr":"405","name":"","yld":"0"},{"pwr":"405","name":"","yld":"0"},{"pwr":"405","name":"","yld":"0"},{"pwr":"","name":"","yld":""},{"pwr":"","name":"","yld":""}],"pa":"1","freq":"","disnightcom":false,"add2total":true}
```

lumapu, 2023-11-16:
ok, wieder was gelernt, das ist wohl von mir nicht sauber implementiert, da die betreffende Funktion die Info vom `AsyncWebserver` bekommt, ich aber nicht drauf reagiere, hier die Funktion:
```cpp
void onApiPostBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total)
```
die Parameter `index` und `total` waren noch nie nötig - jetzt sind sie es.

lumapu, 2023-11-16:
fixed 😅

mtet2, 2023-11-17:
Mit der 0.8.8 wieder mal toll gefixt. Danke.
Jetzt reagiert die DTU auch flüssig und in meinem Fall, duch das getrennte einstellen der Sendepegel, ein echter Mehrwert,
würden die Werbestrategen sagen ;-)
Mal sehen ob jetzt auch die Uptime wieder mehrere Tage schafft.

closed 2023-12-29 by lumapu