lumapu/ahoy #1235: 0.8.9 (ESP8266) 3 Inverter antworten nicht mehr [Bug]

mtet2, 2023-11-19:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.9

### Github Hash

49ef94c

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

`<div id="content"><div class="tabs tabs-tall "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent" id="json-scrolling-panel" tabindex="0">
  |  
-- | --
generic |  
wifi_rssi | -40
ts_uptime | 1054
ts_now | 1700383038
version | "0.8.9"
build | "49ef94c"
menu_prot | false
menu_mask | 61
menu_protEn | false
esp_type | "ESP8266"
system |  
ssid | "edi_FB"
ap_pwd | "esp_8266"
hidd | false
device_name | "AHOY-DTU3"
dark_mode | true
sched_reboot | false
mac | "BC:DD:C2:7B:18:3A"
hostname | "AHOY-DTU3"
pwd_set | false
prot_mask | 61
sdk | "2.2.2-dev(38a443e)"
cpu_freq | 80
heap_free | 17208
sketch_total | 2568192
sketch_used | 562
wifi_rssi | -40
ts_uptime | 1054
ts_now | 1700383038
version | "0.8.9"
build | "49ef94c"
menu_prot | false
menu_mask | 61
menu_protEn | false
esp_type | "ESP8266"
radioNrf |  
en | true
isconnected | true
dataRate | 2
mqtt |  
enabled | true
connected | true
tx_cnt | 84
rx_cnt | 0
interval | 0
heap_frag | 25
max_free_blk | 11848
core_version | "3.1.2"
flash_size | 4096
reboot_reason | "External System"
schMax | 12
mqtt |  
broker | "192.168.178.49"
clientId | ""
port | "1883"
user | ""
pwd | ""
topic | "ws_dtu3"
interval | "0"
ntp |  
addr | "192.168.178.1"
port | "123"
interval | "50"
sun |  
lat | "52.42561"
lon | "10.63059"
offs | 600
pinout |  
cs | 15
ce | 0
irq | 2
sclk | 0
mosi | 0
miso | 0
led0 | 255
led1 | 255
led_high_active | false
radioNrf |  
en | true
isconnected | true
dataRate | 2
serial |  
show_live_data | false
debug | false
static_ip |  
ip | ""
mask | ""
dns1 | ""
dns2 | ""
gateway | ""
display |  
disp_typ | 0
disp_pwr | false
disp_pxshift | true
disp_rot | 0
disp_cont | 60
disp_clk | 255
disp_data | 255
disp_cs | 255
disp_dc | 255
disp_rst | 255
disp_bsy | 255

</div></div></div></div>`

### Error description

Die Hardware läuft unter der 0.8.8 prima.
Unter der 0.8.9  RX no answer = 100% auf allen 3 Invertern.

Die Sicherungdateien sind identisch:
2023-11-17_15-13-58_v0.8.8_ahoy_setup.json
2023-11-19_07-19-10_v0.8.9_ahoy_setup.json

Wieder zurück auf die 0.8.8 und die Sache läuft wieder.
Auf die Lösung bin ich gespannt ;-)

lille2000, 2023-11-19:
Bei mir ist es auch so: keine Kommunikation mit den Invertern seit 0.8.9.

lille2000, 2023-11-19:
`09:48:07 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 1
09:48:07 I: heuristic test mode
09:48:07 I: (#0) TX 27 CH61 | 15 #0 #0 #0 #0 88 94 57 42 80 01 00 65 59 cb c7 00 00 00 00 00 00 00 00 3d 0a dd 
09:48:07 W: RX other inverter:  82 95 76 26 82 95 76 26 81 27 1a 07 e5 04 4d 03 4a 00 68 00 00 00 00 e6 fb be 
09:48:07 I: (#0) request timeout: 500ms
09:48:07 I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 1
09:48:07 I: heuristic test mode
09:48:07 I: (#1) TX 27 CH75 | 15 #1 #1 #1 #1 88 94 57 42 80 01 00 65 59 cb c7 00 00 00 00 00 00 00 00 3d 0a b9 
09:48:07 W: RX other inverter:  82 85 46 62 82 85 46 62 81 27 1e 07 e5 04 b9 04 de 00 66 00 00 00 00 48 3e b8 
09:48:08 I: (#1) request timeout: 501ms
09:48:22 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 2, s: 0, f: 0, n: 2 | p: 1
09:48:22 I: heuristic test mode
09:48:22 I: (#0) TX 27 CH3 | 15 #0 #0 #0 #0 88 94 57 42 80 01 00 65 59 cb d6 00 00 00 00 00 00 00 00 6d ca 5c 
09:48:22 W: RX other inverter:  82 95 76 26 82 95 76 26 81 27 1a 07 e5 04 4d 03 4a 00 68 00 00 00 00 e6 fb be 
09:48:22 I: (#0) request timeout: 500ms
09:48:22 I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 2, s: 0, f: 0, n: 2 | p: 1
09:48:22 I: heuristic test mode
09:48:22 I: (#1) TX 27 CH23 | 15 #1 #1 #1 #1 88 94 57 42 80 01 00 65 59 cb d6 00 00 00 00 00 00 00 00 6d ca 38 
09:48:22 W: RX other inverter:  82 85 46 62 82 85 46 62 81 27 1e 07 e5 04 b9 04 de 00 66 00 00 00 00 48 3e b8 
09:48:23 I: (#1) request timeout: 501ms
`

GrisuXX, 2023-11-19:
Kann ich betätigen auch mit ESP32

dtuuser, 2023-11-19:
Platform
ESP32-D0WD-V3

nRF24L01+ plus (powerlevel Low)

Antenna
external antenna

Power Stabilization
Elko (~100uF)

Eigenes Netzteil 500mA

------------------------------------------------------------

Bei mir auch keine Antwort. Zurück zur 0.8.5, wieder erreichbar

GrisuXX, 2023-11-19:
PS: Der BUG in #1035 scheint mit der 0.8.7 und 0.8.8 behoben zu sein dafür aber werden die Daten nicht resetet wenn der Status "not available" ist. Und das Häckchenn entsprend gesetzt ist.

mtet2, 2023-11-19:
> Platform ESP32-D0WD-V3
> Bei mir auch keine Antwort. Zurück zur 0.8.5, wieder erreichbar

Teste einfach mal die 0.8.8. Auf dem ESP8266 läuft die ganz gut.

dtuuser, 2023-11-19:
> > Platform ESP32-D0WD-V3
> > Bei mir auch keine Antwort. Zurück zur 0.8.5, wieder erreichbar
> 
> Teste einfach mal die 0.8.8. Auf dem ESP8266 läuft die ganz gut.

Ist installiert und läuft. Hat aber 50% RX Fail

mtet2, 2023-11-19:
> > > Platform ESP32-D0WD-V3
> > > Bei mir auch keine Antwort. Zurück zur 0.8.5, wieder erreichbar
> > 
> > 
> > Teste einfach mal die 0.8.8. Auf dem ESP8266 läuft die ganz gut.
> 
> Ist installiert und läuft. Hat aber 50% RX Fail
> 
> Mal sehen, ob Max Power unter Total noch bei "Nicht erreichbar" genullt wird. War bei der 0.8.5 so
Schau mal unter System nach ob: "reboot reason	Hardware Watchdog" zugeschlagen hat.

Bei mir HM600 (High) im Keller, HM1500 (Min) auf dem Dach und die DTU (ESP8266 0.8.8) in der Mitte.
Damit kommt man gut aus.

[image]
[image]

dtuuser, 2023-11-19:
> Schau mal unter System nach ob: "reboot reason Hardware Watchdog" zugeschlagen hat.
>
> Bei mir HM600 (High) im Keller, HM1500 (Min) auf dem Dach und die DTU (ESP8266 0.8.8) in der Mitte. Damit kommt [...]

Bei mir ESP32 finde ich keinen Reboot Grund unter System.... komisch

Gubi2023, 2023-11-19:
8.10 installiert:  öfter mal was neues... Was bedeuten die ganzen Hiroglyphen? sind die irgendwo erklärt?  C:\Users\GünterCzarnyC:\Users\GünterCzarny\Desktop\2023-11-19 14_49_20-Greenshot.jpg

lumapu, 2023-11-19:
@Gubi2023 du hast den 'privacy mode' eingeschaltet, der ist neu in dieser Version 😉

Gubi2023, 2023-11-19:
oberer Teil ist Version 7.6.5....,.

lumapu, 2023-11-19:
ah danke für die Aufklärung 😉 konnte es im Code nicht nachvollziehen, jetzt verstehe ich auch warum

mtet2, 2023-11-20:
Noch eine Anmerkung zur 0.8.10. Die "Radio statistics for inverter" Prozentrechnung hat sich im Ergebnis halbiert.
Wo in der 0.8.8 die Addition der 3 Prozentwerte noch ca. 100 % ergaben, sind es jetzt nur noch ungefähr 50 %.
Wie ist das zu bewerten?

dtuuser, 2023-11-20:
Bei mir addiert aktuell 86,26

lumapu, 2023-11-20:
an der Stelle wird gerade heftig geschraubt - verzeiht wenn hier der ein oder andere Fehler passiert - wichtig ist, dass die Inverter überhaupt kommunizieren 😅.
Vielen Dank fürs reporten - das bitte immer weiter tun, dann wissen wir schneller wo es gerade klemmt.

dtuuser, 2023-11-21:
Aktuelle 0.8.12 liefert leider keine Daten

Inverter #0: Garten is available but no data was received until now

10:24:03 W: Inverter serial does not match

Gubi2023, 2023-11-21:
komisch, bei mir funzt es tadellos...

mtet2, 2023-11-21:
Anmerkung zur 0.8.12 (ESP8266) Die "Radio statistics for inverter" Prozentrechnung.
Jetzt aber nicht übertreiben;-) 2050 %
Es werden keine Daten vom Inverter 0 angezeigt.
Inverter 3 ist ausgeschaltet (nicht angeschlossen)
[image]
[image]
[image]

[image]
[image]

Gubi2023, 2023-11-21:
bei mir sieht es gar nicht schlecht aus:   
[image]

dtuuser, 2023-11-21:
> komisch, bei mir funzt es tadellos...

Nach 2 mal Reboot jetzt wieder Daten. Aber immer mal einige Minuten kein Empfang.

exo1987, 2023-11-21:
Bei mir das selbe Problem. Nach zweimal Rebooten geht es. Wobei MQTT ständig nur UNDEF auf bestimmten Kanälen schickt. Teilweise kommen mal die Watt Leistungen richtig, aber meistens nur UNDEF

reserve85, 2023-11-22:
Bei mir auch das gleiche. 
Die 0.7.60 läuft bei mir wirklich seit Monaten **tadellos**, die neuen 0.8.xx versionen sind bei mir im Vergleich von der Kommunikation wesentlich schlechter. Ständige Aussetzer und teilweise minutenlang keine Kommunikation, die Successrate geht in den Keller. Egal ob mit oder ohne laufender Nulleinspeisung.
Mir ist natürlich klar, dass es die DEV-Version ist. Bitte nicht als gejammer sehen, ich wollte nur anmerken was ich bei mir beobachten konnte.

exo1987, 2023-11-23:
Ich kann bestätigen, das mein ESP32 AhyoDTU mit 0.7.6.0 am Besten lief. Zwar bekomme ich einen Haufen Alarmmeldungen (DTU command failed), aber zumindest klappt der Abruf bei einer Einstellung von 5 Sek. ca. alle 25 Sekunden. Sobald ich auf höhere Versionen wechsel, habe ich höhere Failed-Raten als Success.

Ollipop030, 2023-11-23:
@reserve85 @exo1987 

Ich die beiden Versionen vor ein paar Tagen verglichen, siehe hier:
https://github.com/lumapu/ahoy/issues/1221#issuecomment-1810169236

Ich hoffe aber, das die 0.8er Versionen sich noch wieder verbessern werden.

reserve85, 2023-11-23:
Bei mir ist das **viel** gravierender, ich bekomme mit der 0.8.12 z.B. nahezu keine Kommunikation zustande. Da läuft quasi gar nichts mehr. Wenn ich das ein paar min laufen lassen, dann habe ich TX Count = 105 und RX Success = 4. Also absolut unbrauchbar. Da brauch ich gar nicht weiterprobieren. mit der 0.7.60 TX Count = 3263 und RX Success = 2962. Beides mit Powerlevel "LOW".

Ollipop030, 2023-11-23:
> Welches Abfrageintervall hast du eingestellt?

Bei mir 6 Sekunden. Die Fails wurden aber mehr, als ich das Powerlevel erhöht habe.

exo1987, 2023-11-23:
Ich habe heute auch mal die 0.7.64 installiert und das Powerlevel auf low mit 6Sek Abfrageintervall gesetzt und bekomme miserable Werte...
[image]
Weiß jmd an was das liegen kann?

exo1987, 2023-11-24:
Danke für deine Rückmeldung. Die AhoyDTU hängt ca. 40cm vom WR entfernt. Auf MIN bekomme ich die stabilste Verbindung, aber leider überzeugen die Werte immer noch nicht ganz. Das ist jetzt 0.7.36... läuft irgendwie noch am stabilsten...

[image]

Ollipop030, 2023-11-24:
Vergrößer mal den Abstand zwischen DTU und Wechselrichter. Bei mir sind es knapp 3 Meter, und das war der Tag heute mit 0.7.60:

[image]

exo1987, 2023-11-24:
Danke für den Input. Habe die DTU nun auch 3-4 Meter entfernt platziert, 0.7.60 aufgespielt und das Power Level auf MIN gestellt. Morgen mal schauen, ob es stabiler läuft!

mtet2, 2023-11-29:
> Danke für den Input. Habe die DTU nun auch 3-4 Meter entfernt platziert, 0.7.60 aufgespielt und das Power Level auf MIN gestellt. Morgen mal schauen, ob es stabiler läuft!

Gibt es schon neue Erkenntnisse?

Ich habe soeben die 0.8.13 getestet. Sie liefert wie die 0.8.11/.12 bei mir (ESP8266) keine brauchbaren Messwerte.
Die 0.8.10 läuft, hat aber öfter reboots: "reboot reason Hardware Watchdog" oder "Exception"

Senderti, 2023-11-29:
Mit meinem HM-1200 und dem ESP32 läuft zuletzt Nur die 0.8.10, alle jüngeren Versionen ergeben keine Kommunikation.

lumapu, 2023-11-30:
wir sind noch an der Kommunikation dran, daher vor dem Update einfach kurz checken, ob's überhaupt verändert wurde. Leider ist zZt. die Zeit zum Entwickeln und Testen sehr kurz - ich bin mir aber sicher, dass wir dieses Jahr noch eine brauchbare Version generieren.
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

Senderti, 2023-12-12:
Danke für den Hinweis, ich teste sofern das changelog irgendwie vielversprechend klingt. Ich habe nur einen Inverter und bislang ist die jüngste funktionierende Version 0.8.10.

closed 2023-12-29 by lumapu