lumapu/ahoy #1236: Falsches Datum/Uhrzeit [Bug]

007-mk, 2023-11-19:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

datum

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT ist eingeschaltet
REST Api wird verwendet

### Debug Serial Log output

```shell
siehe oben
```

### Error description

1) in der Webseite wird ebenfalls das falsche Datum angezeigt.
[image]

2) der Unix Timestamp ist in der restAPI nicht korrekt
[image]

3) MQTT ist er ebenfalls nicht korrekt
Dachterasse
last_success = 2085996714
available = 2
ch1
ch2
ch0
alarm

4) in der Hoymiles S-Miles Cloud werden die Werte unterm richtigen Datum korrekt dargestellt.
[image]

007-mk, 2023-11-19:
Neustart behebt das datumsproblem!

MetaChuh, 2023-11-19:
@007-mk 

> Neustart behebt das Datumsproblem

bedeutet das du kannst dieses ticket als gelöst schließen ?

info:
leider haben weder die inverter, noch die esp‘s eine eingebaute, batteriegepufferte uhr, sodass wenn der gewählte ntp server temporär nicht reagiert, das beobachtete passiert.

007-mk, 2023-11-27:
ggf. kann ein re-triggern des Updates des Datum/Uhrzeit umgesetzt werden. Sehe ich als geringere Prio.

DanielR92, 2023-12-03:
Hallo zusammen. kruze Info:
Die Uhrzeit und Datum weiß der Inverter von Beginn an nicht. 
Das beduetet das der Inverter hier abwarten muss bis die DTU die aktuelle Zeit dem Inverter mitteilt. 

Natührlich kann es passieren, das irgendwann mal die Zeit "weg rennt", hier müsste normalerweiße ca. alle 10sek. ein Update zu einem NTP passieren. Außer der NTP-Server selbst ist nicht mehr erreichbar. Dann funkt Ahoy die bekannte Zeit weiter und driftet irgendwann mal weg.

Bitte dieses Issue wieder öffnen wenn weitere Infos bekannt sind. Da wir hier so etwas noch nicht gehört haben das es bei anderen passiert. Also am besten prüfen ob eine Verbindung zum NTP-Server existiert.

closed 2023-12-03 by DanielR92