lumapu/ahoy #1237: Ahoy 0.8.10 "Inverter serial does not match" HMT-1800-6T

Screambear, 2023-11-20:
## Hardware
  * [ ] ESP32

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### CMT2300A Module

### Antenna:
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller

Hallo,

ich habe beide Module parallel laufen.
Leider bekomme ich von meinem HMT-1800-6T keine Daten. In der Konsole steht immer "Inverter serial does not match". Habe die Serial mehrmals gegengecheckt diese stimmt. Hardware funktioniert. Mit OpenDTU werden die Daten empfangen. Ich mag aber die AhoyDTU irgendwie lieber. Deshalb würde ich mich freuen wenn wir das zum laufen bekommen würden.

lumapu, 2023-11-20:
könntest du einen kurzen Log-Auszug bereitstellen? Habe auch zwei HMS (gleiche Kommunikation wie HMT) daher funktoniert Ahoy hier grundsätzlich.
Freut mich zu hören, dass du Ahoy einsetzen willst.

jogybaer0815, 2023-11-28:
Leider auch bei mir: Hoymiles HMS-1600 wird nicht kommuniziert. Ich habe testweise jetzt OpenDTU aufgespielt und der Kontakt ist da.
Ich würde auch gerne wieder auf Ahoy zurückflashen, da die gesamte Oberfläche toll gelungen ist. 
Was wird genau benötigt? die serielle Ausgabe im Webinterface oder wo wird das Log ausgegeben?

jogybaer0815, 2023-12-01:
Leider hat noch keine Version funktioniert. Das Release 0.7.36 war für mich nicht configurierbar (CMT)
Ich könnte noch weitere Versionen unter 0.8.1 testen, wenn ich den Link zur bin finden würde.

Screambear, 2023-12-01:
Sorry fürs lange nicht melden. Bin derzeit leider bei der OpenDTU angekommen.

Bei mir ging auch noch keine Version mit dem CMT. Habe die selbe Ausgabe wie jogybear. Getestet habe ich 0.7.7 (da kann ich aber nicht sagen ob es an der konfiguration etc. lag) 0.8.8 und 0.8.10

lumapu, 2023-12-08:
könnt ihr nochmal ein log ohne die Sterne hochladen? Dafür in den Einstellungen unter `System` den Haken bei `Privacy Mode` entfernen. Zusätzlich bitten den Haken bei `Print whole traces in Log` setzen.

redscho, 2023-12-08:
> könnt ihr nochmal ein log ohne die Sterne hochladen? Dafür in den Einstellungen unter `System` den Haken bei `Privacy Mode` entfernen. Zusätzlich bitten den Haken bei `Print whole traces in Log` setzen.

Hallo Lukas, hier mein Log...
12:10:53 I: (#0) request timeout: 500ms
12:10:53 I: (#0) TX 868.00Mhz | 56 90 91 72 81 81 00 17 65 02 15 21 0c 14 2d 51 05 
12:11:03 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:03 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 c7 00 00 00 00 00 00 00 00 ec 25 75 
12:11:03 I: (#0) request timeout: 500ms
12:11:03 I: (#0) TX 868.00Mhz | 56 90 91 72 81 81 00 17 65 02 15 21 0c 14 2d 51 05 
12:11:13 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:13 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 d1 00 00 00 00 00 00 00 00 8c c3 e5 
12:11:13 I: (#0) RX  30ms | 0-36dBm | 
12:11:13 W: Inverter serial does not match
12:11:14 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:14 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 d2 00 00 00 00 00 00 00 00 7c d7 02 
12:11:14 I: (#0) RX  37ms | 0-38dBm | 
12:11:14 W: Inverter serial does not match
12:11:15 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:15 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 d3 00 00 00 00 00 00 00 00 ec da 9e 
12:11:15 I: (#0) RX  37ms | 0-39dBm | 
12:11:15 W: Inverter serial does not match
12:11:16 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:16 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 d4 00 00 00 00 00 00 00 00 dc fc 8f 
12:11:16 I: (#0) RX  37ms | 0-36dBm | 
12:11:16 W: Inverter serial does not match
12:11:17 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 30
12:11:17 I: (#0) TX 863.00Mhz | 15 90 91 72 81 81 00 17 65 80 01 00 65 72 f9 d5 00 00 00 00 00 00 00 00 4c f1 13 
12:11:17 I: (#0) request timeout: 500ms
12:11:17 I: (#0) TX 868.00Mhz | 56 90 91 72 81 81 00 17 65 02 15 21 0c 14 2d 51 05

lumapu, 2023-12-08:
hast du den GPIO3 bei dir richtig konfiguriert?

jogybaer0815, 2023-12-08:
Bei mir wird über GPOIO2 und 3 mit dem CMT kommuniziert und das läuft unter OpenDTU.
So habe ich es dann auch unter Ahoy eingestellt und unter der System-Übersicht wird auch (grün?) angezeigt, dass die Kommunikation läuft

lumapu, 2023-12-08:
komisch ist, dass bei RX 0-Bytes empfangen wurden. GPIO2 wird bei Ahoy nicht benötigt, nur GPIO3, der GPIO3 wird bei der Systemübersicht nicht geprüft, daher denke ich liegt hier der Fehler.
(Die Anzeige `(#0) RX  37ms | 0-39dBm | ` ist von meiner Seite her nicht optimal, es heißt 0 bytes mit -39dBm empfangen)

jogybaer0815, 2023-12-08:
> kommt auf deine Hardware an, ich weiß nicht ob `GPIO3` an `GPIO34` deines ESPs hängt. Den Rest hast du auf jeden Fall richtig übersetzt.

GPIO3 und GPIO34 sind an meinem ESP32-Mini nicht belegt - habe gerade nochmal nachgesehen: oben hatte ich **_irq: 34_** geschrieben, weil so aus der json entnommen, aber das ist unrichtig. 
Fehlt da noch eine Leitung? (auf der ich gerade stehe)
Hmm: die cmt hat doch gar kein irq...

lumapu, 2023-12-08:
ja da fehlt dann scheinbar eine Leitung. Ahoy benötigt neben den SPI Pins auch eine Interrupt-Leitung (IRQ). Dafür muss vom CMT der Pin `GPIO3` mit einem ESP Pin (nach Wahl) verbunden werden.

jogybaer0815, 2023-12-08:
Ich finde leider keinen Anschluss an der CMT2300A mit GPIO3.

Habe hier den Link mit Pinout, wie ich es verschaltet habe: [Github](https://github.com/tbnobody/OpenDTU/discussions/844)

blueline13, 2023-12-08:
@jogybaer0815 
Schau dir bitte mal das Bild an. Findest du in deinem verlinkten Post ein paar Einträge tiefer.

Dort ist der GPIO3 vom Modul zu sehen. Im Gegensatz zu OpenDTU braucht ahoy diesen Pin für die Kommunikation mit dem CMT Modul. Verbinde bitte den GPIO3 vom CMT2300 mit einem noch freien GPIO vom ESP vebindest und trage den GPIO vom ESP in ahoy ein.

[image]

redscho, 2023-12-08:
> ja da fehlt dann scheinbar eine Leitung. Ahoy benötigt neben den SPI Pins auch eine Interrupt-Leitung (IRQ). Dafür muss vom CMT der Pin `GPIO3` mit einem ESP Pin (nach Wahl) verbunden werden.

Ohh, das könnte die Lösung sein, da Wahrscheinlich auf dem PCB der GPIO3 vom CMT2300 nicht auf den ESP verbunden ist. Ich werde morgen mal eine feine Litze auf dem CMT zum Pin7(GPIO3) auf mein ESP-WROOM-32 auf den PIN33 legen und nochmals testen. 

Viele Dank für eure Unterstützung.

[image]
[image]
[image]
[image]
[image]
[image]

jogybaer0815, 2023-12-08:
Ja prima: hier dauert es leider etwas länger, da mit angelöteter GPI03 <-> GPI03 ließ sich die 0.814 an 2 Rechnern nicht flashen - Kabel ab und es klappt mit dem flashen

jogybaer0815, 2023-12-08:
> dann machen wir einen Feature-Request daraus, um den `GPIO3` in Zukunft nicht mehr zu brauchen

oder einfach die Infos auffrischen. Auf diesen Anschluss (GPIO3 vom CMT2300A) wäre ich nicht mehr gekommen. Dachte Ahoy und OpenDTU sind Geschwisterchen :)

```19:28:50 I: (#0) RX  82ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 01 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 c0 
19:28:50 I: (#0) RX 132ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 02 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 4b 
19:28:50 I: (#0) RX 182ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 03 00 00 01 35 00 00 00 00 00 30 00 00 00 00 00 00 c7 
19:28:50 I: (#0) RX 232ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 04 00 00 09 25 13 87 01 25 00 00 00 0c 03 e8 00 b1 0e 
19:28:50 I: (#0) RX 273ms | 15-11dBm | 95  *  *  *  *  *  *  *  * 85 00 03 55 5d 4e 
19:28:50 I: (#0) Payload (66): 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 00 00 01 35 00 00 00 00 00 30 00 00 00 00 00 00 00 00 09 25 13 87 01 25 00 00 00 0c 03 e8 00 b1 00 03 
19:29:05 I: (#0) RX  83ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 01 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 c0 
19:29:05 I: (#0) RX 132ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 02 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 4b 
19:29:05 I: (#0) RX 182ms | 27-12dBm | 95  *  *  *  *  *  *  *  * 03 00 00 01 33 00 00 00 00 00 30 00 00 00 00 00 00 c1 
19:29:05 I: (#0) RX 232ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 04 00 00 09 26 13 86 01 24 00 00 00 0c 03 e7 00 b1 02 
19:29:05 I: (#0) RX 273ms | 15-12dBm | 95  *  *  *  *  *  *  *  * 85 00 03 2b 2c 41 
19:29:05 I: (#0) Payload (66): 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 00 00 01 33 00 00 00 00 00 30 00 00 00 00 00 00 00 00 09 26 13 86 01 24 00 00 00 0c 03 e7 00 b1 00 03 
```
So wie es ausschaut, klappt jetzt alles. Habe den Inverter am Netz/Netzteil, um zu testen.

Vielen lieben Dank an alle für die Unterstützung und nun kann auch das Projekt unterstützt werden und eigentlich hier auch geschlossen werden (bug ist es ja nicht mehr wirklich)

redscho, 2023-12-08:
> > dann machen wir einen Feature-Request daraus, um den `GPIO3` in Zukunft nicht mehr zu brauchen
> 
> oder einfach die Infos auffrischen. Auf diesen Anschluss (GPIO3 vom CMT2300A) wäre ich nicht mehr gekommen. Dachte Ahoy und OpenDTU sind Geschwisterchen :)
> 
> ```
> 19:28:50 I: (#0) RX 132ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 02 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 4b 
> 19:28:50 I: (#0) RX 182ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 03 00 00 01 35 00 00 00 00 00 30 00 00 00 00 00 00 c7 
> 19:28:50 I: (#0) RX 232ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 04 00 00 09 25 13 87 01 25 00 00 00 0c 03 e8 00 b1 0e 
> 19:28:50 I: (#0) RX 273ms | 15-11dBm | 95  *  *  *  *  *  *  *  * 85 00 03 55 5d 4e 
> 19:28:50 I: (#0) Payload (66): 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 00 00 01 35 00 00 00 00 00 30 00 00 00 00 00 00 00 00 09 25 13 87 01 25 00 00 00 0c 03 e8 00 b1 00 03 
> 19:29:05 I: (#0) RX  83ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 01 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 c0 
> 19:29:05 I: (#0) RX 132ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 02 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 4b 
> 19:29:05 I: (#0) RX 182ms | 27-12dBm | 95  *  *  *  *  *  *  *  * 03 00 00 01 33 00 00 00 00 00 30 00 00 00 00 00 00 c1 
> 19:29:05 I: (#0) RX 232ms | 27-11dBm | 95  *  *  *  *  *  *  *  * 04 00 00 09 26 13 86 01 24 00 00 00 0c 03 e7 00 b1 02 
> 19:29:05 I: (#0) RX 273ms | 15-12dBm | 95  *  *  *  *  *  *  *  * 85 00 03 2b 2c 41 
> 19:29:05 I: (#0) Payload (66): 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 2e 00 00 00 fc 00 00 00 00 01 39 00 01 00 62 00 00 01 33 00 00 00 00 00 30 00 00 00 00 00 00 00 00 09 26 13 86 01 24 00 00 00 0c 03 e7 00 b1 00 03 
> ```
> 
> So wie es ausschaut, klappt jetzt alles. Habe den Inverter am Netz/Netzteil, um zu testen.
> 
> Vielen lieben Dank an alle für die Unterstützung und nun kann auch das Projekt unterstützt werden und eigentlich hier auch geschlossen werden (bug ist es ja nicht mehr wirklich)

Habe es davor schon unterstützt 😉. Dem GPIO3 habe och auch keine Beachtung geschenkt, da er mir nicht untergekommen ist in der Doku was ich gelesen habe. Vielen Dank auch von mir an alle Helfer!

lumapu, 2023-12-08:
vielen Dank fürs Feedback - ich ändere die Labels auf gefixt 😀 Danke für eure Unterstützung. Ich bräuchte irgendwie noch die CMT Komponente für Fritzing, dann kann ich die Doku auf https://ahoydtu.de/getting_started/ vervollständigen.
Ich packe in unsere Wishilst #1199 noch den Punkt, CMT ohne `GPIO3`.

jogybaer0815, 2023-12-09:
Das CMT-Bauteil zu zeichnen ist nicht so aufwändig... bis dahin hänge ich hier eine Verdrahtung an:
[image]

lumapu, 2023-12-10:
> Das CMT-Bauteil zu zeichnen ist nicht so aufwändig... 

dann muss ich mir die Zeit dafür mal nehmen, oder kannst du es mir zukommen lassen?

jogybaer0815, 2023-12-11:
ich muss mich da erst reinlesen

jogybaer0815, 2023-12-11:
Nicht besonders präzise, aber zur Illustration sollte es reichen:
[CMT2300A.fzpz.zip](https://github.com/lumapu/ahoy/files/13637532/CMT2300A.fzpz.zip)

lumapu, 2023-12-12:
danke ich probiere es aus

SmashDde, 2023-12-29:
DANKE für den Hinweis CMT2300A GPIO3 > RX0 GPIO3! 
Schon mehrfach kurz davor gewesen den ganzen Krempel gegen die Wand zu werfen! 😄 Jetzt klappt es mit der AhoyDTU [GIT SHA: 7c08d93 :: 0.8.31](https://github.com/lumapu/ahoy/commits/7c08d93)

Auch, wenn ich mich jetzt hier so ins Gespräch rein mogel... wollte eben ein "Danke!" für die Info da lassen. Hab viel länger rumgestochert und rumprobiert wie nötig.

lumapu, 2023-12-29:
ja die Doku hinkt gewaltig hinterher 🙄

stefan123t, 2024-01-05:
@jogybaer0815 hier ist das Fritzing Bauteil für das CMT2300A Breakout Board von @tbnobody & Blinky Parts.
Das hatte ich irgendwann im Sommer 2023 mal gebastelt, nachdem wir da immer noch kein Fritzing davon hatten :stuck_out_tongue_winking_eye: 
https://discord.com/channels/984173303147155506/993500210271629393/1141477467987640331

[EBYTE E49 900M20S CMT2300A_breakout.fzpz.gz](https://github.com/lumapu/ahoy/files/13847345/EBYTE.E49.900M20S.CMT2300A_breakout.fzpz.gz)

Und hier noch mal alle meine Fritzing Schematics für OpenDTU, bzw. AhoyDTU inkl. dem o.a. Fritzing Part. 

Laut @Knickohr fehlen da glaube noch die Kondensatoren und die Display Varianten.
https://discord.com/channels/984173303147155506/1029761098381017098/1093154731876159588

Ich bin natürlich immer noch nicht dazu gekommen das reinzubasteln :shrug:  
[CMT2300A.zip](https://github.com/lumapu/ahoy/files/13847351/CMT2300A.zip)

@lumapu abgesehen von der Doku scheint es ja das selbe Problem beim CMT2300A wie beim NRF24L01+ zu sein, daß AhoyDTU/OpenDTU nicht erkennen ob der GPIO3 bzw. IRQ vom RF Modul connected ist.
Könnte man beim CMT2300A wie beim NRF24L01+ ebenfalls einen TX IRQ auf GPIO3 setzen um das ggf. abzuprüfen ?

stefan123t, 2024-01-05:
[image]
[image]
[image]
[image]
[image]
[image]

stefan123t, 2024-01-05:
Und hier die Fritzings von @Knickohr für die Displays:

[image]
[image]
[image]
[image]
[image]
[image]

lumapu, 2024-01-05:
@stefan123t seit letzer Woche habe ich sowohl Schaltbilder als auch Fritzings auf https://ahoydtu.de/getting_started in der Rubrik `Zusammenbau HMS / HMT DTU`. Ich habe das `CMT2300A` selbst gezeichnet und eingefügt. Einzig eine Steckplatine kann man mit dem Board nicht sinnvoll machen - was ja auch der Wirklichkeit entspricht 😉.

Ja man kann das CMT2300A so konfigurieren, dass man einen TX Interrupt auf dem GPIO3 ausgibt - oder man kann den GPIO evtl. auch manuell auslösen. Habe hier die Möglichkeiten noch nicht ausgelotet, hier ist das RFPDK unser Freund.

stefan123t, 2024-01-05:
@lumapu doch mit dem Breakout Board von Blinky Parts ist das CMT2300A Modul eigentlich sogar pin-kompatibel mit den üblichen NRF24L01+ Pinouts. Deswegen hatte ich das verwendet. 
Evtl. stimmt bei meinem Schematic Symbol irgendwas mit der Skalierung oder den beiden Pins GND / VCC noch nicht, die Verbindungen dazu werden immer etwas schief auf dem Rastermaß. 
Schau Dir ggf. mal die oben **versteckten Kommentare** an, ich wollte nur den Thread / Issue nicht zu unübersichtlich werden lassen :wink:

lumapu, 2024-01-05:
schon gesehen, aber wenn sich einer ein CMT2300A kauft, hat er nicht gleich auch das Breakout-Board da es mWn. noch kein Standard ist.

knickohr, 2024-01-06:
@stefan123t 
Du mußt die Verbindung mal von der anderen Seite routen. Ist ein alt bekanntes Problem vom Fritzing 😞

closed 2024-02-17 by lumapu