lumapu/ahoy #1242: [Suggestion] add date an release No to download file

rmayergfx, 2023-11-27:
If possibel on GIT,  add  date and Relese to download name. Actually it is always the same name ahoydtu_dev. Would be much easier not to rename ich file at download to manage the releases locally. 
e.g. 
ahoydtu_dev_2023.11.20_v0.8.12.zip

rmayergfx, 2023-11-28:
@lumapu 
is this an official link? Never seen this. 
Prefer to download from git with the changes suggested.

Peertje-62, 2023-11-28:
Hi, 
found it on Discord, (search fw.ahoydtu.de)
Because apparently several users are looking for the various versions of the ahoy-dtu.
bye

Peertje-62, 2023-11-28:
actually you should post your request here:
https://github.com/lumapu/ahoy/issues/1199
bye

lumapu, 2023-11-28:
the reason behind the identical name was the direct link from https://ahoydtu.de page. In future it can be changed, meanwhile I suggest to use this download listing:
https://fw.ahoydtu.de

lumapu, 2024-02-29:
now the dev builds include a version number, should be fixed

DanielR92, 2024-06-29:
Then we can close this issue here. :)

closed 2024-06-29 by DanielR92