lumapu/ahoy #1243: [Bug] Integration über P vs Yield

ThomasWallner, 2023-11-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.58

### Github Hash

d0c834f

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Interval = 15s
Yield Efficiency (should be between 0.95 and 0.96) = 0.952

### Error description

Ich nutze einen HM-600 mit 2 Panels.

In den Einstellungen ist bei mir Yield Efficiency=0,952 eingestellt, d.h. es werden 1/0,952=5,0% vom Ertrag abgezogen.

Ich hole mir über die REST API alle 10s die Werte, logge sie mit und erzeuge so eine CSV-Datei pro Tag.

Jetzt vergleiche ich gerade in meinen Daten die Integration der Leistungswerte (P_AC / 0_P_DC, 1_P_DC) über die Zeit mit dem letzten Wert des aufsummierten Ertrags (Yield) pro Tag. Das sollte ja eigentlich weniger als 1 Prozent voneinander abweichen?

Dabei stelle ich fest, dass Integral (P_AC) dt immer etwa 5% höher als Yield ist.
Integral (0_P_DC) dt ist jeden Tag etwa 10% höher als 0_YieldDay, genauso 1_P_DC vs. 1_YieldDay.

Ich habe den Verdacht, dass mein Wechselrichter in den Ertrag (Yield) die Effizienz schon einrechnet und der zweite 5% Abschlag von Ahoy eigentlich unnötig ist.
Übersehe ich das was?

mellow5, 2023-11-28:
Ich habe vor längerer Zeit das Gleiche gemacht und bin auch zu diesem Ergebniss gekommen. Die ohne Faktoren aufintegrierte AC Leistung ergab einen Wert der sehr sehr nah am "Yield Day" lag, wobei bei "Yield Efficiency" 1 eingetragen war. Was ich noch nicht kontrolliert habe, ist ob die AC Leistung wirklich der Angabe entspricht.

ThomasWallner, 2023-11-28:
Es werden max. 612W AC-Leistung angezeigt, das bestätigt ein Zwischensteckermessgerät (trotz fehlendem Vorzeichen).

technics42, 2023-11-28:
Honestly I also don't get the point about this "yield efficiency" factor. What is it used for? The values are directly taken from inverter, or?

(I also think the value for "irradation" is quite senseless. It seems to be just calculated by current power divided by max. power, which should be entered per module. But this doesn't make sense for any module which is limited by inverter. E.g. having 400Wp module at inverter with max. 300W per chanel, you will never reach 100% "irradation"-value...)

lumapu, 2023-11-28:
Up for now we are not sure if the values of the inverter need a efficiency factor. If this factor is `1` nothing is changed at all. Thank you for your investigations, maybe we come to a conclusion that this value isn't needed - I don't know.

Solarteur, 2023-11-29:
@technics42
I agree with you - the Irradiation value is pretty worthless for the reasons you describe.

@All
From my point of view, the instantaneous values for the power (AC power in the green fields) are AC values.
The comparison of "YieldDay" and "YieldTotal" with the values of a Shelly EM (current measurement via current transformer) shows in my case that the "YieldEfficiency" factor makes sense.
If you have a different opinion, you can set it to 1.

technics42, 2023-11-29:
Sorry, but I don't understand for which calculation this yield-efficiency is used.
Is it used for calculating a yield value? I thought these values are directly taken from inverter. Or are the values from inverter multiplied with this factor, but which values are changed? All yield-values?

If it is used for any AC-values, then "yield-efficiency" might be the wrong naming - and I still wouldn't understand the sense of it.

mellow5, 2023-11-29:
I can performe a test next week. I can measure the AC power with a self-build power meter (Arduino and ADC with high resolution 12 bit). The meter gives pretty accurate readings for power compared to my installed smart meter (ED300L). Less than 1 % differenc in power and energy. It samples the current and voltage over three periods and calculates the power. I get a reading approx. twice per seconde. I can log the data against the AC power of my HM1500. And build the integrals of both powers vs. Yield.

technics42, 2023-11-29:
I am really lookinh forward for the test-results!
But in any case, I would propose to only use the values of the inverter itself and not doing any calculations with some static factors.
The values displayed should reflect the values stored and transferred by the inverter itself.
(You are relying on all values from inverter, but questioning only yield-values)
Any "efficiency" is not really static, but would depend on some operating point, which is depending on multiple factors. But you could only use some statistical efficiency-factor, which you would need to get out first. But without unmodified data, this would even be more hard to do so...

To sum up, I would vote for removing "yield-efficiency"-factor completely in any case, or at least set default to 1 (including information what this factor is used for)
And I would remove senseless/wrong "irradiation"-values completely.

mellow5, 2023-12-07:
I logged AC power and  Yield Total of my HM1500 every second into a database.

[image]

I then calculated the integral over time of the AC power and plotted it over Yield Total. I subtracted the start value of yield total to compare the rise of yield total and the AC power integral in one plot.

[image]

The error between both lines is plotted in the last plot.

[image]

I have three modules at the HM1500 therefore the error typically goes up to 4 Wh because the AC power integral is more or less continuous, while the yield total counts up in full Wh steps because the resolution is only 1 Wh.

Conclusion:

The AC power integral is the same as the Yield total or yield day. And as mentioned the "yield-efficiency" was set to 1.

Solarteur, 2023-12-08:
Three more comments on this topic:
1. The **instantaneous** values are correctly assigned to DC or AC.
2. If in the overview (consisting of two screenshots) with “Yield Efficiency = 1” the value for Yield Total (in the olive-green field) is the **AC** value, the values for the individual PV modules (in the blue fields) must be also AC values  - only then the added are values correct. But that wouldn't really be plausible in my opinion.

[image]

3. Measured with a Shelly EM over 18 months, I had an AC feed-in of 1363 kWh, compared with 1408 kWh from AhoyDTU. In my opinion, using a “Vield Efficiency” factor makes perfect sense.

mellow5, 2023-12-10:
> In my opinion all the yield values are measured/calculated by Hoymiles using output-values (AC).
> 
> Your difference on comparing Shelly with direct DTU-inverter values are related to measurement tolerances, which is quite usual. (approx. 3%)
> Please keep in mind inverter and also external measurement-equipments do have tolerances.
> But you cannot compensate those tolerances by using a static correction factor.
> So I would vote for using direct values from inverter and remove this "yield efficiency" factor.
> 
> I guess these values are also used directly in OpenDTU and also in Hoymiles-app/cloud.
> 
> 
> 

This is why I wanted to compare the AC power with a power measurement over time. We don't know where the 3 % diffent come from. It might be the inverter effiency or the measurement accuracy of the Shelly or both. 

The quality of the Shelly's measurement needs to be assumed questionable hence It's just a mass market product designed to cost.

I have to admit that it's hard to find the truth with "home measurement equipment".

Over the holidays I can give it a try with my own powermeter. But realistically spaleaking the accuracy might not be good enough to get closer to the truth.

Another test could be to build the integral of the DC power and compare it to the yield per module. If the yield per module is already converted into AC values as you suggested. The difference might show here.

ThomasWallner, 2023-12-11:
> 2. If in the overview (consisting of two screenshots) with “Yield Efficiency = 1” the value for Yield Total (in the olive-green field) is the **AC** value, the values for the individual PV modules [...]

It is plausible in my opinion. The inverter takes care of each module separately.  This is why a power limitation is applied to each half individually instead of just limiting the total power, as would actually make more sense for me.

I understand that the inverter displays the AC yield per module, as this is a more important value for the operator of the system than the DC yield.

This also fits in with our study: If we integrate all AC power over time, we get the AC yield fairly accurately.

I think it is very unrealistic that AC-Yield and AC-Power both need to be corrected and do not correspond to reality.

mellow5, 2023-12-11:
I compared the integral of the DC power with the YIELD DAY of one module. I can see a clear drift. The DC power integral is approx. 5 % bigger than the YIELD DAY value per module. Therefore I would assume the YIELD DAY already includes the efficiency of the inverter and is actually already converted to AC value. From my point of view this should be sufficient proof to get rid of the "yield-efficiency".

closed 2024-08-16 by lumapu