lumapu/ahoy #1244: [Bug] millis()/micros() overflow handling in hmRadio.h

You69Man, 2023-11-28:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.12

### Github Hash

53186973

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Die Überlauflogik im hmradio.h sieht für mich nicht korrekt aus, ich glaube da 'verschluckt' er sich (ca. ab Zeile 129):
Die Abfrage muss immer (((unsigned)millis() - (unsigned)starttime) < (unsigned)difftime) lauten, sonst funktioniert so etwas nicht.
```
            uint32_t startMicros = micros() + 5110;
            uint32_t loopMillis = millis() + 400;
            while (millis() < loopMillis) {
                while (micros() < startMicros) {  // listen (4088us or?) 5110us to each channel
```

Hab in der git history nachgeschaut, das war früher schon mal anders (d.h. richtig):
```
            uint32_t startMillis = millis(); // + 400;
            uint32_t startMicros = micros(); // + 5110;
            while (millis() - startMillis < 400) {
                while ((unsigned long)(micros() - startMicros) < 5110) {  // listen (4088us or?) 5110us to each channel
```

Das ist mit commit bf772756 "improved set limit and radio statistics" reingekommen.

lumapu, 2023-11-28:
ok, der Überlauf wird aus meiner Sicht in beiden Fällen vernachlässigt. Ich weiß nicht, ob der cast auf `unsigned long` hier noch was auswirken kann.
Danke für den Hinweis. Besonders kritisch ist der `micros` Teil, da dieser Zähler mWn. schon nach 11 Tagen überläuft.

You69Man, 2023-11-29:
Stimmt, der Cast ist bei unsigned Long nicht mehr notwendig (bei unsigned short wäre er es schon), es ging mir bei meinem abstrakte Beispiel jedoch um auf den 'unsigned' Aspekt hinzuweisen, wegen dem das Overflow Handlung ja funktioniert, wenn man es macht.

haraldhotzbehofsits, 2023-12-01:
Die 5110 als auch die 400 sind magic numbers, die 5110 wird 2 mal verwendet, eigentlich 2 mal mit dem gleichen statement:

startMicros = micros() + 5110;

You69Man, 2023-12-01:
> Die 5110 als auch die 400 sind magic numbers, die 5110 wird 2 mal verwendet, eigentlich 2 mal mit dem gleichen statement:
> 
> startMicros = micros() + 5110;

Ja, das ist mir auch schon aufgefallen. Das Problem kommt leider mehrfach vor, nicht nur bei diesen 'magic numbers', sondern ist auch an anderen Stellen systematisch falsch (sorry @lumapu, das ist nicht böse gemeint, aber leider wahr). Ich würde gerne helfen, das zu bereinigen, aber ich verstehe teilweise leider einfach nicht die kommunikationstechnischen Hintergründe und Zeiterwartungen der Inverter, und ich würde da wahrscheinlich vieles nur verschlimmbessern. Ich könnte maximal nur noch ein paar Stellen aufzeigen, die mir verdächtig erscheinen, aber ich will auch nicht nerven...

You69Man, 2023-12-06:
Es war ja wie gesagt schon Mal richtig (siehe oben), und müssre daher einfach nur wieder so gemacht werden, wie es war ...

lumapu, 2023-12-07:
> Ja, das ist mir auch schon aufgefallen. Das Problem kommt leider mehrfach vor, nicht nur bei diesen 'magic numbers', sondern ist auch an anderen Stellen systematisch falsch (sorry @lumapu, das ist [...]

ja bitte schreib mir wo die Fehler sind. An den Stellen wollte ich einfach die Operations in den loops reduzieren, indem die Summe nur einmal gemacht werden muss und in der Schleife lediglich verglichen wird - war gut gemeint aber ist wohl nach hinten losgegangen.
Danke, dass ihr euch den Code antut - genau deswegen liebe ich Open-Source 🎉

You69Man, 2023-12-28:
fixes in #1298

closed 2023-12-29 by lumapu