lumapu/ahoy #1246: AhoyDTU zero export using DDSU666

Wind06, 2023-11-30:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ x] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

Hi,

I would like to know if there is some plan to implement and support the zero export with AhoyDTU using DDSU666 smart meter. I am very please to use AhoyDTU ‘Good job Guys’.

thanks

knickohr, 2023-11-30:
Maybe in future versions with the Fusion PCB. Fusion V2.x has an RS485 connection. But if you run a RasPi, why do you not implement it with it by yourself ?

Wind06, 2023-11-30:
Thanks for your answer.
To answer your question, coding is not something I like to do, anyway way not :-)

lumapu, 2023-11-30:
This project is driven from people who implement / test new features. As the maintainer I only implement features which I need or like, but merge almost everything people provided via pull request.
Tell me a reason why I should implement this special power meter? Your last response sounded a bit snooty too me - I don't like that.
Does this power meter use some common Modbus addresses / registers? Do you have a datasheet?

Wind06, 2023-11-30:
@lumapu, whoa my answer was absolutly not snooty, it was more I don't know how to implement. Sorry if my answer was not appropriate.

I am looking for such implementation because I do have such meter already in place, today I have Hoymiles DTU which use this for Zero injection. but I am not happy with this private and closed solution. as far as I know yes this meter use common addresses/register. Mine is DDSU666 Single-phase I have a paper datsheet, below is a link of this meter. the communication with this meter is done using rs485.

In case this project will be supported, I will be very pleased to help and test it under my environment.

https://www.google.fr/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjA2OrJ6-yCAxX_VqQEHX5vAzQQFnoECBUQAQ&url=https%3A%2F%2Fwww.njoy.global%2Fapi%2Fdownload%2FcHJvZHVjdHMvMTY3ODIwMzcyNlFHczNnZy5wZGY%3D%2FDDSU666_User%2520Manual%2520EN.pdf%2F&usg=AOvVaw2I6t-QuD0aoDERbeWZeNqO&opi=89978449

lumapu, 2023-12-07:
Ok sorry for getting you wrong. 

If your meter does use common adresses I assume it can be supported once we implemented the RS485 feature for the Fusion board which @knickohr mentioned. Currently I'm implementing its ethernet shield which also has connectors for RS485 😀

Wind06, 2023-12-07:
That’s a very good news, many thanks for your answer and support, very much appreciated 🤩 Hopefully I do have 2 DDSU666, one from Hoymiles and second one which is more standard.
If any help is required for testing let me know I will be happy to help.
Take care guys, you are doing a very good job.

Wind06, 2024-05-03:
Hi, Could it be possible to get an update of this feature ? 
If you need beta tester, I will be very happy to help at least with DDSU666 smart meters.

If it can help, I build a .sh script to be integrated in the reserve85 project. The script read powermeter values from a DDSU666 Monophase using USB 485 connected to a raspberry pi 4 to perform HoymilesZeroExport.

stefan123t, 2024-10-29:
@Wind06 did you check the OpenDTU-OnBattery (OBOD) project ?

Please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/hoylabs/OpenDTU-OnBattery/).
I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in issue #272.

Actually there is a Pull Request for adding Modbus RS485 support to OpenDTU / OBOD too.

@lumapu *afaik* the Chint DDSU666 is one of the Power Meter's that are supported by the original Hoymiles DTU Pro.

Wind06, 2024-11-02:
@stefan123t no I did not because I do have an Ahoy DTU.

Anyway I will check mentioned threads and most probably build an OpenDTU.

Thanks for pointers and informations.

stefan123t, 2024-11-02:
@Wind06 see here for the latest PR from b0661 for the Modbus feature: https://github.com/tbnobody/OpenDTU/pull/1893 This may not be exactly what you are looking for, because it is a Server which exports the OpenDTU details via Modbus, it is a good start if you want to query another Power Meter yourself though.