lumapu/ahoy #1247: Freitag, die 0.8.13te - neue Oberfläche

kiu77, 2023-12-01:
## Hardware
  * [ ] ESP32

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller

kiu77, 2023-12-01:
Ich war überrascht, als ich die neue Oberfläche auf dem OLED-Bildschirm der DTU sah, nachdem ich ein Update auf 0.8.13 vorgenommen hatte.
1. Mir gefällt sie. Sie sieht informativer aus und optisch gelungen.
2. Nicht alle Symbole in der Anzeige sind mir von der Bedeutung her klar. Gibt es dafür eine Beschreibung?

Viele Grüße!

[image]

knickohr, 2023-12-01:
Erklärung der Symbolik der 128x64 bzw. 84x48 Display Layouts:
* Balken links (dynamisch): Qualität der Funkverbindungen, abgeleitet von der Heuristik. Es wird der niedrigste Wert aller konfigurierten Inverter genommen (weil man will  ja schnell sehen, wenn irgendwo was nicht stimmt). Balken ganz leer: Qualitätszahl=-6, Balken ganz voll: Qualität=+4
* Balken rechts (dynamisch): Qualität der Wlan Verbindung, abgeleitet vom RSSI Wert des Wlan Moduls
* Antennensymbol links unten (dynamisch): Status des Funk-Moduls. Entspricht beim NRF24 Modul dem Status von der Webseite. Falls nicht gefunden oder deaktiviert, wird es "durchgestrichen". Für CMT kenne ich noch keine Möglichkeit den Modulstatus abzufragen, falls jemand weiß, wie das geht, kann man das natürlich zum Funkmodul-Status dazu verknüpfen.
* WLAN Symbol rechts unten (dynamisch): Wlan Status oder MQTT-Status. MQTT Symbol hat Vorrang, weil wenn MQTT Verbindung aufrecht ist, ist auch (W)LAN Verbindung aufrecht.
* Mond-Symbol (dynamisch): Alle konfigurierten Inverter schlafen (oder nicht erreichbar)
* Sonne-Symbol (dynamisch): Alle konfigurierten Inverter aktiv
* Mond- und Sonne Symbol, mit Zahl daneben: Anzahl aktivierter und inaktiver Inverter
* Kalender-Symbol (mit 1, statisch): Tagesertrag
* Summen-Symbol (statisch): Gesamtertrag

You69Man, 2023-12-01:
Mit PR #1248 sollte dann auch der CMT Status mit dem Antennen-Statussymbol verknüft sein:
Wenn eines der konfigurierten Funkmodule (NRF, CMT, oder beide) nicht gefunden wird, wird die Antenne 'ausgepixelt', um zu ziegen, dass was mit dem/den Funkmodul(en) nicht stimmt.
Die 'Bakteriophage' vulgo 'Space Invader' gibt' dann nicht mehr ;).

kiu77, 2023-12-06:
Danke knickohr - jetzt ist mir alles klar.
Interessant ist, dass die Verbindung zu den Invertern trotzgrößeren Abstands besser ist als die WLAN-Anbindung.
Die DTU steht nur 3m vom Router entfernt. Ergo ist die WLAN-Anbindung der ESP32 mit aufgedruckter Antenne nicht so optimal. Das nächste mal werde ich ein Exemplar mit externer Antenne verwenden.

closed 2023-12-29 by lumapu