lumapu/ahoy #1251: Negative Temperaturen vom Inverter sind mit drei Nachkommastellen dargestellt.

lille2000, 2023-12-03:
Das ist kein Fehler, eher „unhübsche“ Darstellung
Bei negativen Temperaturen  sind die Temperaturen auf 3 Nachkommestellen angegeben. 
[image]

BastyESP, 2023-12-03:
Hatte ich gestern auch bei einem Inverter -0.699, aber gestern waren sie auch schnell wieder alle vormittags 11.20-11.40uhr mit AhoyDTU Offline, mit Open noch Online hatte dazu auch was in Discord geschrieben, scheint aber kein anderer das Problem zu haben.

Hab aktuell mit OpenDTU 5w auf 2 Invertern (werden nur 2/3 damit abgefragt), und 0 Erzeugung laut AhoyDTU die 3 Abfragt...da scheint was komplett intern abgeschmiert zu sein, hatte ich noch gar nicht in den ganzen Monaten, das es Abgeschmiert aber die Web Version problemlos läuft. 0.8.13 ist es bei mir übrigens...

LeSpocky, 2023-12-06:
> Hatte ich gestern auch bei einem Inverter -0.699, aber gestern waren sie auch schnell wieder alle vormittags 11.20-11.40uhr mit AhoyDTU Offline, mit Open noch Online hatte dazu auch was in Discord geschrieben, scheint aber kein anderer das Problem zu haben.

Das kann ich so nicht unterschreiben.  Bei mir werden negative Temperaturen auch mit mehreren Nachkommastellen dargestellt, development version 0.7.57 aktuell.

lille2000, 2023-12-17:
Heute war es endlich mal wieder kalt genug.  Die Darstellung ist nun schöner. 
[image]

lille2000, 2023-12-17:
Ticket wir geschlossen

closed 2023-12-17 by lille2000