lumapu/ahoy #1256: AhoyDTU Hoymiles HM800 - Reactive Power 0

Wind06, 2023-12-06:
Hi,

I have an installation with 2 HM1500 which is working ok.
I installed a new Inverter HM800, which is also working ok apparently.
but on AhoyDTU the HM800 do have a reactive power reported to 0

is it normal ?

Your help will be very much appreciated.

[image]

You69Man, 2023-12-06:
On Onduleur_3 the reactive power of 0var is consistent with a power factor of 1.
More strange are Onduleur_1 and _2, where the reactive power is 18.2var while the power factor is shown as 1.

Wind06, 2023-12-06:
Thanks for your answer, is there something I can do ? 
Or just let the installation as it is ?
apparently the Ac Power seems to be closed to the actual weather condition.

Wind06, 2023-12-06:
Currently using hoymiles-DTU with zero export set. Sometimes the % is very strange.
When I disable the zero export from hoymilesDTU, AhoyDTU report 484% ….

That’s one of the reason why I want to use AhoyDTU with zero export but I have a DDSU666 as a smart meter.

Wind06, 2023-12-14:
Hello, After a factory reset and also changed Pins connections according to ESP8266 recommandation the power limit is back at 100% which is more accurate :-)

No change for the reactive power.

closed 2024-02-17 by lumapu