lumapu/ahoy #1257: Reset YieldDay At Midnight not working

Jumperm, 2023-12-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.12

### Github Hash

Edit later

### Build & Flash Method

ESP Tools (flash)

### Setup

Mit 0.7.46 keinen Probleme

### Error description

Keine Nullung der YieldDay um Mitternacht. Bei erstem verbinden der DTU Übernahme des Wertes des Vortages

knickohr, 2023-12-08:
Doch, es nullt, aber irgendwoher kommen wieder alte Werte 😲

Jumperm, 2023-12-08:
Ich lese das per API aus. ich Frage mich halt, woher der Wert des Vortages kommt!

Jumperm, 2023-12-08:
> Wahrscheinlich noch in irgendeiner Variable vorhanden. Das hatten wir schon mal.

Aus dem WR/DTUnoder im Homeassistant?

knickohr, 2023-12-09:
WR kann man ausschließen, DTU wäre möglich, und wenn es aus HA kommt, dann ist der Issue hiermit geclosed (andere Baustelle) 😉

Jumperm, 2023-12-09:
Da ich weder an Homeassistant noch am WR etwas geändert habe gehe ich von einem Bug in der DTU aus. ich hab im Ursprungscode nochmal meine Rest API Anfrage des Homeassisant gepostet.

Jumperm, 2023-12-09:
Ohne mein zutun wurde heute Morgen wieder bei 0 angefangen. Ich beobachte das mal weiter.

Jumperm, 2023-12-15:
Es ist ein Ahoy Problem. Nach dem flashen auf die aktuelle 0.8.23 wird der korrekte Wert aus den WR abgerufen. Oder im Hintergrund wurde es gefixed.

strepwo, 2023-12-28:
@Jumperm kannst du mal 0.8.29 testen? Da ist es möglicherweise repariert

Jumperm, 2023-12-30:
> @Jumperm kannst du mal 0.8.29 testen? Da ist es möglicherweise repariert

Mach ich

closed 2024-01-01 by Jumperm