lumapu/ahoy #1259: [Bug] Heuristic: Quality level jumps

You69Man, 2023-12-09:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.15

### Github Hash

379654fa51a3c8012051d35f834a35ae4a702cb9

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-

### Debug Serial Log output

```shell
14:49:41 I: (#0) TX 27 CH23 | 15  *  *  *  * 81 76 73 82 80 0b 00 65 74 70 75 00 00 00 00 00 00 00 00 0e ca 8a 
14:49:41 I: (#0) RX  51ms | 27 CH75  95  *  *  *  *  *  *  *  * 01 00 01 01 a3 00 21 00 8b 01 a1 00 20 00 84 00 03 9a 
14:49:41 W: (#0) frame 2 missing: request retransmit (3 attempts left)
14:49:41 I: (#0) TX 11 CH23 | 15  *  *  *  * 81 76 73 82 82 53 
14:49:41 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 9, s: 8, f: 0, n: 1 | p: 255
14:49:41 I: (#0) TX 27 CH23 | 15  *  *  *  * 81 76 73 82 80 0b 00 65 74 70 75 00 00 00 00 00 00 00 00 0e ca 8a 
14:49:42 I: (#0) request timeout: 500ms
14:49:46 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 10, s: 8, f: 0, n: 2 | p: 255
14:49:46 I: heuristic test mode
```

### Error description

Das sollte glaube ich nicht so sein: Der Qualitätswert springt wegen eines einzigen Aussetzers von 4 auf -6.

lumapu, 2023-12-09:
heißt das, dass das NRF bei dir mit der `0.8.15` (null-acht-fuchzehn) 😂 läuft?

You69Man, 2023-12-09:
Ja klar, schon seit dem frühen morgen, auf meinen beiden ESP32 und ESP8266 wemos boards mit NRF. Ich bin auch gar nicht so unzufrieden, auf jeden Fall läuft die Kommunikation stabiler als bisher. So Nullachtfünfzehn ist die gar nicht, wenn sie nicht gerade auf einem ETH Fusion laufen soll 😉. Ein paar komische Sachen beobachte ich halt noch. Die tx-rx Statistik dürfte noch nicht ganz korrekt berechnet zz werden, die ergibt in Summe keine 100% und dann ist mir eben dieser seltsame Sprung des Qualitätswertes aufgefallen. Ich vermute das hängt irgendwie mit diesem zwischengespeicherten storedIdx zusammen, der wahrscheinlich eher ein storedQuality sein soll (oder umgekehrt?? ).

knickohr, 2023-12-09:
Das wissen wir bereits. Das ist es auch was bei mir die Verbindung meiner 16 Inverter so verhagelt. Irgendetwas resettet die komplette Heuristik 😢

Was, wo ?   Offenbar funktioniert ein 8266 mir der 0.8.15 7nd NRF 😉

You69Man, 2023-12-09:
Auch der Esp32 funktioniert. Wahrscheinlich hat nur der ESP32S3 ein NRF SPI Peoblem.

You69Man, 2023-12-09:
> Das wissen wir bereits. Das ist es auch was bei mir die Verbindung meiner 16 Inverter so verhagelt. Irgendetwas resettet die komplette Heuristik 😢

Zwischen der 0.8.14 und 0.8.15  haben sich schon ein paar wesentliche dinfe bei der heurisik getan. Aber ein paar Dinge gehören halt noch gefixt. Auf den Fall sind wir auf dem Weg der Besserung 🙂

You69Man, 2023-12-09:
> Du liest schon im https://discord.com/channels/984173303147155506/992030956313657384 mit ?
>
Nicht nur das, ich schreibe dort sogar mit, allerdings (aus historischen Gründen) unter dem Alter Ego Platypus 😂. Sorry für die Verwirrung!

lumapu, 2023-12-09:
> Also scheint es mal wieder ein generic Fusion Problem zu sein, ich habe mit PoE ja auch nichts am Hut.

noch nicht @knickohr 

Bei mir ist die Verbindung schlecht, ich muss unbedingt den Ahoy näher zum Wechelrichter bringen

knickohr, 2023-12-09:
Neinein !  Nicht unbedingt. Das habe ich bei der neuen Heuristik auch bemerkt, das diese sehr empfindlich auf schlechte Empfangsbedingungen ist. Mit einem Inverter habe ich so gut wie gar keine Kommunikation mehr zurande gebracht. Bei denen mit guten Empfangsbedingungen war es offenbar kein Problem.

You69Man, 2023-12-09:
Offenbar hat mit der 0.8.16 ein Paradigmenwechsel in der Heuristik stattgefunden. Der explizite Testmode ist verschwunden!
Neue Chance neues Glück! Die 0.8.16 wartet gespannt auf das nächste Tageslicht :)

knickohr, 2023-12-09:
Ich glaube nicht das er weg ist, er kommt nur wesentlich seltener. Wenn ich das richtig verstanden habe, dann wurde wieder die alte Logik eingebaut und er testet erst nach 50 Antworten erneut. Und offenbar springt bzw. resettet sich die Heuristik nicht nach dem ersten Fehler komplett.

Ich kann’s momentan leider nicht testen da die letzten beiden Versionen mit dem Fusion-Board Probleme machen (NRF wird nicht erkannt).

You69Man, 2023-12-10:
0.8.16 werkelt auf ESP32 und ESP8266 brav vor sich hin. Habe bisher auch keine völlig sinnlosen Sprünge durch die Heurisik mehr festgestellt, d.h. die Heuristik macht zumindest keinen Unsinn mehr!

You69Man, 2023-12-10:
Hoppla, da ist ja doch ein kleiner Bug:
heuristic.h line 44:
`                    if(ih->testChId **=**= ih->txRfChId)`

mtet2, 2023-12-11:
Unter meinen Empfangsbedingungen funktioniert die 0.8.18 auf dem ESP8266 immer noch nicht zufriedenstellend.
Die 0.8.10. liefert Werte aus den Invertern.

Gubi2023, 2023-12-11:
Datenübertragung mit 8.18 :

mtet2, 2023-12-12:
Die 0.8.20 auf dem ESP8266 fühlt sich am Batteriespeicher schon gut an ;-)
Mal sehen wie es morgen bei Sonne weiter geht
[image]

closed 2023-12-29 by lumapu