lumapu/ahoy #1260: 0.8.13 hängt sich auf und verbindet sich nicht mehr[Bug]

BastyESP, 2023-12-09:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.13

### Github Hash

cfef2ce

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP32 (30PIN)
NRF Modul 
2.42" OLED
Elko 

### Error description

Hallo, seit ich die 0.8.13 drauf habe. Ist es nun schon 2* passiert das es sich komplett aufhängt, aber die webUi problemlos weiter läuft, es verbindet sich einfach nicht mehr mit den Invertern...

Erster Screenshot ist von der AhoyDTU, zweiter ist von OpenDTU die die ersten beiden (im Ahoy Screenshot ebenfalls abfragt), kann aber nicht das alleinige Problem sein, denn der HM-300 der nur über die Ahoy läuft ging ja auch nicht mehr...

[image]
[image]

Warum das passiert kann ich nicht sagen, heute früh war ich definitiv nicht im Ui unterwegs. Also muss irgendwas sein, was von alleine Auftritt...

Gruß Bastian

P.s. 30 PIN ESP32, NRF Modul, 2.42" Display dran, und auch ein Elko. Lief seit Juni aber auch ohne die Probleme, muss was mit der 0.8.13 zutun haben, denn erst danach kam es 2* vor.

BastyESP, 2023-12-10:
Ist GitHub unübersichtlich...wie kann man einen Kommentar abgeben? Ohne es zu schließen?

Ergänzend: Gewartet ob es sich heute früh doch verbindet...

[image]

Wie man sieht sind zwischen den beiden Screenshots knapp über eine Minute vergangen, da wurde neu gestartet und 3 Inverter angefragt...alle Aktiv....

You69Man, 2023-12-10:
Schon mal 0.8.16 oder höher probiert?

Dirk-Ahoy, 2023-12-10:
oder 0.8.10 wenn es da geht ist es das gleiche Problem wie bei mir.

BastyESP, 2023-12-10:
Nein bisher nichts gemacht, da ich von den anderen Versionen darüber nicht durchsehe, in Discord, geht es nun nicht wegen dem LAN Modul oder allgemein nicht richtig...vorher ging es ja auch ohne die Aufhänger...

Ist 0.8.16 für NRF+ESP32? Geeignet oder ist das die nächste Baustelle, die ich draufspielen würde?

Dirk-Ahoy, 2023-12-10:
Versuch mal die 8.10  und danach gerne mal die 8.17 - wir sind gerade im Discord am suchen irgendwo ist da ein Fehler reinkommen beim esp32.

lumapu, 2023-12-29:
die aktuelle Version `0.8.31` sollte auf jeden Fall besser gehen

BastyESP, 2023-12-29:
Ich hab die 0.8.30 gestern Abend drauf gemacht, ich mach dann mal zu.

closed 2023-12-29 by lumapu