lumapu/ahoy #1265: MQTT :  umbenennen eines Topics und „mit Leben“ füllen

knickohr, 2023-12-12:
Mir ist seit der ersten 0.8.x aufgefallen das es den alten Topic <topic>/comm_disabled nicht mehr gibt und durch den neuen Topic <topic>/dis_night_comm ersetzt wurde.  Diesen gibt es jetzt sowohl für jeden Inverter einzeln als auch global.

Vorschlag : 
- Umbenennen des globalen Topics nach <topic>/night_behaviour
- Mit Leben füllen :  true bei Nacht und false bei Tag (wie vorher beim alten <topic>/comm_disabled
- Die anderen inverterspezifischen Topics können so bleiben

So wäre die Abwärtskompatibilität wieder hergestellt 😉

knickohr, 2023-12-13:
Kein Enhancement, sonder die alte Funktion vom Topic wieder herstellen 😉

lumapu, 2023-12-13:
dann doch ein bug

knickohr, 2023-12-14:
Neun, einfach nur vergessen worden 😅

knickohr, 2023-12-28:
Funzt wieder 👍

closed 2023-12-29 by lumapu