lumapu/ahoy #1267: Ahoy 0.8.21 mit Nulleinspeisung sehr instabil

Ollipop030, 2023-12-13:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.21
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich verwende das Script von @reserve85 für die Nulleinspeisung. Bei mir wird alle 13 Sekunden das Limit auf beide WR geschrieben. 

Zum Einen ist Ahoy mit der Version sehr instabil, ich habe ca. alle 2 Minuten Neustarts. Wenn ich das Script beende läuft Ahoy ohne Neustart. Das war unter den 0.7.xx Versionen aber nicht so, die liefen Wochen lang ohne Neustart. Ich weiß nicht, wer hier noch eine Anlage mit Nulleinspeisung betreibt, vielleicht können ja ein paar Leute was berichten.

Zum Anderen habe ich noch ein etwas komisches Verhalten beobachtet, dazu zwei Logs:

```
11:14:05.038 I: (#1)  has accepted power limit set point 90 with PowerLimitControl 0
11:14:05.049 -----
11:14:05.049 -----
11:14:05.050 I: (#1) sendControlPacket cmd: 0x0b
11:14:06.551 -----
11:14:06.551 I: (#0) sendControlPacket cmd: 0x0b
11:14:06.053 I: (#0) RX  34ms | 17 CH03 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:06.055 I: (#0)  has accepted power limit set point 451 with PowerLimitControl 0
11:14:06.056 -----
11:14:06.066 -----
11:14:06.066 I: (#0) sendControlPacket cmd: 0x0b
11:14:07.568 I: (#0) RX  34ms | 17 CH03 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:07.570 I: (#0)  has accepted power limit set point 430 with PowerLimitControl 0
11:14:07.571 -----
11:14:07.581 -----
11:14:07.581 I: (#0) sendControlPacket cmd: 0x0b
11:14:07.083 I: (#0) RX  35ms | 17 CH03 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:07.085 I: (#0)  has accepted power limit set point 430 with PowerLimitControl 0
11:14:07.086 -----
11:14:07.096 -----
11:14:08.101 I: (#0) sendControlPacket cmd: 0x0b
11:14:08.609 -----
11:14:08.610 I: (#0) sendControlPacket cmd: 0x0b
11:14:09.112 -----
11:14:09.614 I: (#0) RX  56ms | 27 CH23 95  *  *  *  *  *  *  *  * 81 00 01 01 1e 00 00 03 e8 ff ff ff ff 01 68 31 1c a5 
11:14:09.617 I: (#0) Payload (14): 00 01 01 1e 00 00 03 e8 ff ff ff ff 01 68 
11:14:09.620 -----
11:14:10.121 I: (#1) RX  40ms | 27 CH23 95  *  *  *  *  *  *  *  * 81 00 01 01 2c 00 00 03 e8 ff ff ff ff 01 68 89 e8 db 
11:14:10.124 I: (#1) Payload (14): 00 01 01 2c 00 00 03 e8 ff ff ff ff 01 68 
11:14:10.126 -----
11:14:10.628 -----
11:14:11.130 I: (#1) RX  55ms | 27 CH40 95  *  *  *  *  *  *  *  * 01 00 01 01 2c 00 38 00 a8 00 04 b0 dc 00 0f 09 01 47 
11:14:11.133 I: (#1) RX 102ms | 27 CH23 95  *  *  *  *  *  *  *  * 82 13 88 00 a1 00 00 00 07 03 e8 00 91 00 01 84 b7 62 
11:14:11.146 I: (#1) Payload (30): 00 01 01 2c 00 38 00 a8 00 04 b0 dc 00 0f 09 01 13 88 00 a1 00 00 00 07 03 e8 00 91 00 01 
11:14:11.149 -----
11:14:11.150 I: (#1) sendControlPacket cmd: 0x0b
11:14:11.662 I: (#1) RX  34ms | 17 CH23 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:11.664 I: (#1)  has accepted power limit set point 90 with PowerLimitControl 0
11:14:11.665 -----
11:14:11.665 -----
11:14:11.676 I: (#1) sendControlPacket cmd: 0x0b
11:14:12.177 I: (#1) RX  35ms | 17 CH23 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:12.180 I: (#1)  has accepted power limit set point 90 with PowerLimitControl 0
11:14:12.180 -----
11:14:12.191 -----
11:14:12.191 I: (#1) sendControlPacket cmd: 0x0b
11:14:12.693 I: (#1) RX  35ms | 17 CH23 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:12.695 I: (#1)  has accepted power limit set point 90 with PowerLimitControl 0
11:14:12.695 -----
11:14:12.706 -----
11:14:13.209 I: (#0) RX  59ms | 27 CH03 95  *  *  *  *  *  *  *  * 81 00 01 01 1e 00 00 03 e8 ff ff ff ff 01 68 31 1c a5 
11:14:13.212 I: (#0) Payload (14): 00 01 01 1e 00 00 03 e8 ff ff ff ff 01 68 
11:14:13.214 -----
11:14:13.716 I: (#1) RX  46ms | 27 CH23 95  *  *  *  *  *  *  *  * 81 00 01 01 2c 00 00 03 e8 ff ff ff ff 01 68 89 e8 db 
11:14:13.719 I: (#1) Payload (14): 00 01 01 2c 00 00 03 e8 ff ff ff ff 01 68 
11:14:13.721 -----
11:14:14.223 -----
11:14:14.224 I: (#1) sendControlPacket cmd: 0x0b
11:14:14.726 I: (#1) RX  49ms | 17 CH40 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
11:14:14.728 I: (#1)  has accepted power limit set point 86 with PowerLimitControl 0
```
Das ist von Ahoy, hier wurde das Limit mehrmals kurz hintereinander akzeptiert. 
Außerdem: Limitänderung auf 451 Watt um 11:14:06, das Limit wurde aber gesendet um 11:13:48 (beide Uhren laufen synchron, das habe ich überprüft):

```
Dec 13 11:13:48 Pi-Hole python3[1599]: 2023-12-13 11:13:48 INFO     Ahoy: Inverter "HM-1500": setting new limit from 450 Watt to 451 Watt
Dec 13 11:13:51 Pi-Hole python3[1599]: 2023-12-13 11:13:51 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Dec 13 11:13:51 Pi-Hole python3[1599]: 2023-12-13 11:13:51 INFO     Ahoy: Inverter "HM-300": setting new limit from 90 Watt to 90 Watt
Dec 13 11:13:56 Pi-Hole python3[1599]: 2023-12-13 11:13:56 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
```

Also, Limitänderung auf 451 Watt um 11:14:06, und nur 1,5 Sekunden später wieder auf 430, das war schon der nächste Durchlauf des Scriptes. Der zweite WR akzeptiert in einer Sekunde 3x das neue Limit, obwohl nur einmal gesetzt. Ich habe den Eindruck, dass Ahoy mit den Limitänderungen nicht nachkommt, und es deshalb zu den Neustarts kommt. 
Ach ja, ist ein frisch geflashter ESP, mit vorigem Erase.

Bin auf Antworten gespannt.

lumapu, 2023-12-13:
das hat vor dir noch keiner getestet, danke fürs Feedback.
Ich denke ich muss unter der Voraussetzung für Statemachine nochmal betrachten.

Ollipop030, 2023-12-14:
Alter Falter, das nenne ich mal eine schnelle Reaktion. Die 0.8.22 läuft übrigens sehr viel besser als ihr Vorgänger. Vielen Dank dafür.

Hier mal ein Auszug aus dem Log mit 0.8.22.

```
09:08:35.594 -----
09:08:35.605 I: (#0) sendControlPacket cmd: 0x0b
09:08:36.106 I: (#0) RX  35ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:36.108 W: Inverter serial does not match, got: 0x74953798, expected: 0x83924915
09:08:36.108 I: (#0) RX  35ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:36.120 I: (#0)  has accepted power limit set point 405 with PowerLimitControl 0
09:08:36.131 -----
09:08:41.591 I: (#0) RX  35ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:41.593 I: (#0)  has accepted power limit set point 405 with PowerLimitControl 0
09:08:41.594 -----
09:08:42.095 I: (#1) RX  35ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:42.097 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
09:08:42.097 I: (#1) RX  45ms | 27 CH75 95  *  *  *  *  *  *  *  * 81 00 01 01 0e 00 00 03 e8 ff ff ff ff 01 68 f1 48 21 
09:08:42.110 I: (#1) Payload (14): 00 01 01 0e 00 00 03 e8 ff ff ff ff 01 68 
09:08:42.122 -----
09:08:47.090 I: (#1) sendControlPacket cmd: 0x0b
09:08:47.592 I: (#1) RX  36ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:47.594 I: (#1)  has accepted power limit set point 81 with PowerLimitControl 0
09:08:47.594 -----
09:08:47.605 I: (#0) sendControlPacket cmd: 0x0b
09:08:48.106 I: (#0) RX  36ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:48.108 W: Inverter serial does not match, got: 0x74953798, expected: 0x83924915
09:08:48.108 I: (#0) RX  37ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:48.120 I: (#0)  has accepted power limit set point 407 with PowerLimitControl 0
09:08:48.131 -----
09:08:53.591 I: (#0) RX  37ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:53.593 I: (#0)  has accepted power limit set point 407 with PowerLimitControl 0
09:08:53.594 -----
09:08:54.095 I: (#1) RX  37ms | 17 CH75 d1  *  *  *  *  *  *  *  * 81 00 00 0b 00 14 07 48 
09:08:54.097 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
09:08:54.097 I: (#1) RX  40ms | 27 CH75 95  *  *  *  *  *  *  *  * 81 00 01 01 0f 00 00 03 e8 ff ff ff ff 01 68 0d 4c d8 
09:08:54.110 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
09:08:54.121 I: (#1) RX  40ms | 27 CH75 95  *  *  *  *  *  *  *  * 81 00 01 01 0e 00 00 03 e8 ff ff ff ff 01 68 f1 48 21 
09:08:54.124 I: (#1) Payload (14): 00 01 01 0e 00 00 03 e8 ff ff ff ff 01 68 
09:08:54.136 -----
```

Der erste Inverter nimmt das Limit noch 2x an, das passiert aber nur sporadisch.

Und `Inverter serial does not match, got: 0x83924915, expected: 0x74953798`, immer im Wechsel. Scheinbar wird hier irgendwas vertauscht, dieser Fehler zieht sich aber durch das gesamte Log.

reserve85, 2023-12-15:
Also bei mir gehts nicht so gut, kurz nach einem reboot wird 1x das Limit gesetzt, danach erscheint im Log nur noch das:

`08:44:19.597 I: (#0) RX  38ms | 17 CH03 | 81
08:44:19.599 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
08:44:19.599 -----
08:44:19.600 I: com loop duration: 505ms
08:44:19.600 -----
08:44:25.598 I: (#0) RX  38ms | 17 CH03 | 81
08:44:25.599 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
08:44:25.600 -----
08:44:25.600 I: com loop duration: 505ms
08:44:25.601 -----
08:44:31.597 I: (#0) RX  38ms | 17 CH03 | 81
08:44:31.598 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
08:44:31.599 -----
08:44:31.599 I: com loop duration: 504ms
08:44:31.600 -----`

Quasi endlos, ZeroExportScript kann ich auch ausschalten, es ändert nix. Kommen keine Daten in der Live-View an.

lumapu, 2023-12-15:
@reserve85 hast du mal geschaut wie das Intervall bei dir steht? In der `0.8.23` wurde der default wieder auf 15s zurückgesetzt durch eine Änderung.

reserve85, 2023-12-15:
@lumapu ja, das habe ich wieder auf 6s gestellt, wie bei der 0.7.60 (die übrigens bei mir absolut spitze läuft)

lumapu, 2023-12-15:
ne passt, habs in der Zwischenzeit auch gesehen, gerade bei einem Inverter ist das sehr großzügig. Du bist auf einem ESP32?

reserve85, 2023-12-15:
korrekt, esp32 mit Kondensator und ext. Antenne. Der normale Standard :-)

lumapu, 2023-12-15:
kannst du selbst kompilieren? falls ja dann ersetze doch bitte mal das in der `/src/hm/CommQueue.h:33`:

```cpp
uint8_t getFillState(void) {
    DPRINTLN(DBG_INFO, "wr: " + String(mWrPtr) + ", rd: " + String(mRdPtr));
    return abs(mRdPtr - mWrPtr);
}
```

reserve85, 2023-12-15:
Leider nicht vom Firmen-Rechner aus. Evtl schaffe ich es heute nachmittag von zu Hause

reserve85, 2023-12-15:
doch noch geschafft: leider das gleiche verhalten.

```
10:48:39.162 I: wr: 1, rd: 1
10:48:39.665 I: (#0) RX  49ms | 17 CH03 | 81
10:48:39.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:48:39.667 -----
10:48:39.668 I: com loop duration: 505ms
10:48:39.668 -----
10:48:54.162 I: wr: 2, rd: 2
10:48:54.665 I: (#0) RX  49ms | 17 CH03 | 81
10:48:54.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:48:54.666 -----
10:48:54.667 I: com loop duration: 504ms
10:48:54.667 -----
10:49:09.162 I: wr: 3, rd: 3
10:49:09.666 I: (#0) RX  49ms | 17 CH03 | 81
10:49:09.667 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:49:09.667 -----
10:49:09.668 I: com loop duration: 504ms
10:49:09.668 -----
10:49:24.162 I: wr: 4, rd: 4
10:49:24.665 I: (#0) RX  49ms | 17 CH03 | 81
10:49:24.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:49:24.666 -----
10:49:24.667 I: com loop duration: 504ms
10:49:24.667 -----
10:49:39.162 I: wr: 5, rd: 5
10:49:39.665 I: (#0) RX  49ms | 17 CH03 | 81
10:49:39.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:49:39.666 -----
10:49:39.667 I: com loop duration: 504ms
10:49:39.667 -----
10:49:54.162 I: wr: 6, rd: 6
10:49:54.665 I: (#0) RX  49ms | 17 CH03 | 81
10:49:54.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:49:54.666 -----
10:49:54.667 I: com loop duration: 504ms
10:49:54.667 -----
10:50:09.162 I: wr: 7, rd: 7
10:50:09.665 I: (#0) RX  49ms | 17 CH03 | 81
10:50:09.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:50:09.666 -----
10:50:09.667 I: com loop duration: 504ms
10:50:09.667 -----
10:50:24.162 I: wr: 8, rd: 8
10:50:24.665 I: (#0) RX  49ms | 17 CH03 | 81
10:50:24.666 I: (#0)  has accepted power limit set point 1500 with PowerLimitControl 0
10:50:24.666 -----
10:50:24.667 I: com loop duration: 504ms
10:50:24.667 -----

```

Ollipop030, 2023-12-19:
Ich habe jetzt die 0.8.27 drauf, diese Version scheint komplett fehlerfrei zu laufen.

```
14:25:37.581 -----
14:25:38.583 -----
14:25:38.583 I: com loop duration: 1505ms
14:25:38.584 -----
14:25:47.078 I: (#1) sendControlPacket cmd: 0x0b
14:25:47.580 I: (#1) RX  36ms | 17 CH61 | 81
14:25:47.581 I: (#1)  has accepted power limit set point 95 with PowerLimitControl 0
14:25:47.581 -----
14:25:48.083 I: (#0) sendControlPacket cmd: 0x0b
14:25:48.585 I: (#0) RX  39ms | 17 CH75 | 81
14:25:48.585 I: (#0)  has accepted power limit set point 475 with PowerLimitControl 0
14:25:48.586 -----
14:25:48.586 I: com loop duration: 1508ms
14:25:48.586 -----
14:25:57.079 I: (#1) sendControlPacket cmd: 0x0b
14:25:57.581 I: (#1) RX  57ms | 17 CH61 | 81
14:25:57.581 I: (#1)  has accepted power limit set point 97 with PowerLimitControl 0
14:25:57.582 -----
14:25:58.082 I: (#0) sendControlPacket cmd: 0x0b
14:25:58.584 I: (#0) RX  41ms | 17 CH75 | 81
14:25:58.584 I: (#0)  has accepted power limit set point 485 with PowerLimitControl 0
14:25:58.585 -----
14:25:58.585 I: com loop duration: 1507ms
14:25:58.585 -----
14:26:07.579 -----
14:26:08.583 I: (#1) RX  46ms | 27 CH75 | 81
14:26:08.584 I: (#1) Payload (14)
14:26:08.585 -----
14:26:08.585 I: com loop duration: 1507ms
14:26:08.585 -----
14:26:17.078 I: (#1) sendControlPacket cmd: 0x0b
14:26:17.580 I: (#1) RX  37ms | 17 CH75 | 81
14:26:17.581 I: (#1)  has accepted power limit set point 96 with PowerLimitControl 0
14:26:17.581 -----
14:26:18.082 I: (#0) sendControlPacket cmd: 0x0b
14:26:18.584 I: (#0) RX  36ms | 17 CH75 | 81
14:26:18.584 I: (#0)  has accepted power limit set point 484 with PowerLimitControl 0
14:26:18.585 -----
14:26:18.585 I: com loop duration: 1507ms
14:26:18.585 -----

```

Ist nur ein Auszug, habe das Log Fenster aber länger offen gelassen und mal durchgescrollt. Auch das beim ZeroExport kommt das ACK vom Inverter direkt, keine Timeouts mehr. Und die Fails bei den Invertern sind sind <1%. Saubere Arbeit.

Einzig `Inverter #1: HM-300 is available but no data was received until now` taucht hier länger auf nach einem Neustart, obwohl er Limits annimmt lt. Log.

reserve85, 2023-12-19:
Hab die .27 ebenfalls seit heute Morgen laufen und bin auch zufrieden. Saubere Arbeit!

mtet2, 2023-12-20:
0.8.27 auf ESP8266 mit 4 Invertern (HM1500, HM600, HM600, HM600)
Inverter 0 an PV
Inverter 1 an Batterie
Inverter 2 deaktiviert
Inverter 3 deaktiviert

Interval 8 S
Inverter Gap 200 ms 
(An der Antwort zur Gapfrage bin ich auch interessiert)

Das Auslesen und Anzeigen der aktiven Inverter funktioniert prima.

Fehler:
Sende ich an Inverter1 einen "TURN OFF" Befehl, dann wird dieser Befehl beliebig oft wiederholt:

[Ahoy_log_0_8_27.txt](https://github.com/lumapu/ahoy/files/13725676/Ahoy_log_0_8_27.txt)

lumapu, 2023-12-27:
die Gap ist aktuell ein Versuch, man sollte den Wert auf ~500ms stellen. Es kommt uU. bei der Kommunikation zu problemen, wenn die Kommandos zu schnell nacheinander gesendet werden. Diese Gap schafft eine Lücke in der Kommunkation um Überschneidungen zu verhindern. In naher Zukunft denke ich, wird dieser Parameter wieder verschwinden.

closed 2023-12-29 by lumapu