lumapu/ahoy #1268: Feedback Versionen und Funktion (0.8.12 - 0.8.21)

kiu77, 2023-12-13:
## Hardware
ESP32

## nRF24L01+ Module
nRF24L01+ you verified this is a Plus model capable of the required 256kBit/s mode

## Antenna:
external antenna (SMA)

## Power Stabilization:
100uF Electrolytic Capacitor

## Build & Flash Method:
AhoyDTU Webinstaller

## Inverter
2 x HM-800

Kurzes Feedback:
0.8.13 lief gut.
Nach Update auf 0.8.19 war das Setup irgendwie zerschossen.
Im Log stand: "Inverter serial does not match". In den Settings stand aber der richtige Wert.
Also neu geflachst und nochmal auf 0.8.19 upgedatet. Jetzt lief es, aber der Balken links
für die Feldstärke der Inverter war von 4 auf 3 Striche runter.
Nach Update auf 0.8.21 dauert es ein paar Minuten, bis die Werte von beiden invertiert
kamen (schnell der eine, langsam der andere) und es läuft, aber der linke Balken hat nur noch
einen einzigen Strich, was ja nicht so gut ist.

lumapu, 2023-12-13:
der Test war aber nicht von bis, sondern wohl nur .12, .13, .19 und .21 wenn ich ich richtig verstehe. Bitte nochmal mit der .22 verifizieren (ist ganz neu)

kiu77, 2023-12-13:
Ja richtig, die Zwischenstufen fehlen.

kiu77, 2023-12-13:
Nach zwei Stunden war der Balken links bei 0.8.21 wieder da mit 4 Strichen.

kiu77, 2023-12-14:
0.8.22 läuft gut mit Balken links auf 4 Strichen.

Edibo63, 2023-12-23:
Ich mach hier mal weiter:
ESP 32  ext. Antenne.  HM600, HM 400, HM 350
Vers. 0.8.23 funktioniert problemlos.
Vers. 0.8.26 bekommt Daten vom HM 600.  Beim HM 350 und HM 400 kommt nur available und keine Daten.
Vers. 0.8.27 bekommt Daten vom HM 350 und HM 400. Der HM 600 nur available.

Edibo63, 2023-12-27:
0.8.28 läuft jetzt auch problemlos.
Habe mal das Powerlimit auf 80% reduziert. Der Inverter hat dann zwar 80% angezeigt aber nichts mehr produziert.
DTU rebootet, Inverter keine Leistung.  Nach ca. 5 Minuten hat der Inverter wieder produziert, wurde aber anscheinen rebootet da Yield Day wieder von 0 begonnen hat.

closed 2024-01-01 by lumapu

kiu77, 2024-05-28:
@aliander11 
Wenn Du schon den Thread kaperst, sollst Du auch eine Antwort haben ;-)
Sie lautet: nein.
Mit einfachen Mitteln wie dieser DTU geht das nicht, weil mehr als 100% nicht drin sind.
Selbst wenn die Leistungshalbleiter das an Mehrleistung vertragen würden, kommst
Du nicht an den Code des Inverters, um den zu ändern oder zu patchen, bzw.
wäre das so aufwendig, dass Verkauf und Neukauf eine genial günstige Lösung ist.