lumapu/ahoy #1269: Heuristic again

You69Man, 2023-12-13:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.21

### Github Hash

c06705d

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Die heuristic update nicht korrekt. Grund vermutlich:

heuristic.h Zeile 106
 if(RF_TX_TEST_CHAN_1ST_USE == ih->txRfChId) {
                        // switch back to original quality
                        ih->txRfQuality[ih->txRfChId] = ih->saveOldTestQuality;
                    }
RF_TX_TEST_CHAN_1ST_USE  ist 0xFF, das txRfQuality[] beherbergt aber nur 5 Kanäle, also Indexüberlauf...

lumapu, 2023-12-13:
sehe den Fehler nicht, ich hoffe du hast gesehen, dass es zwei verschiedene Variablen sind:
* `ih->testChId`
* `ih->txRfChId`

dadurch sehe ich keinen Indexüberlauf. Danke für deinen Hinweis.

You69Man, 2023-12-13:
Das sollte wahrscheinlich so sein, ist es aber nicht:
[image]

lumapu, 2023-12-14:
ah Zeile `106`, ich habe in `78` geschaut, sorry. Das ist der Nachteil von doppeltem Code 😉

lumapu, 2023-12-14:
fix ist nicht so schwerwiegend wie angenommen, der Code in der `if` Condition wird nie aufgerufen, hier der fix:

```patch
diff --git a/src/hm/Heuristic.h b/src/hm/Heuristic.h
index 4b3705d..c3ce5cc 100644
--- a/src/hm/Heuristic.h
+++ b/src/hm/Heuristic.h
@@ -103,7 +103,7 @@ class Heuristic {
                     // graceful evaluation for big inverters that have to send 4 answer packets
                     updateQuality(ih, RF_TX_CHAN_QUALITY_OK);
                 } else if((rxFragments - ih->lastRxFragments) < 2) {
-                    if(RF_TX_TEST_CHAN_1ST_USE == ih->txRfChId) {
+                    if(RF_TX_TEST_CHAN_1ST_USE == ih->testChId) {
                         // switch back to original quality
                         ih->txRfQuality[ih->txRfChId] = ih->saveOldTestQuality;
                     }

```

You69Man, 2023-12-14:
PR #1270
Du warst schneller :)

You69Man, 2023-12-14:
Danke für den Hinweis auf den Originalcode von Oberfritze!
Damit hab ich jetzt auch noch den anderen Heuristik-Bug, wodurch die verlorenen Pakete nicht mehr zu einer Abwertung der Qualitätszahlen führen gefunden!

commit 222bf0e5 (2023-12-14): 0.8.23
* heuristics fix #1269 #1270
* moved `sendInterval` in settings, **important:** *will be reseted to 15s after update to this version*
* try to prevent access to radio classes if they are not activated
* fixed millis in serial log
* changed 'print whole trace' = `false` as default
* added communication loop duration in [ms] to serial console
* don't print Hex-Payload if 'print whole trace' == `false`
closed 2023-12-29 by lumapu