lumapu/ahoy #1271: [HMS-1000] data the inverter failed to parse and the data didn't come out.

aditiyagio, 2023-12-14:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

Development 03. Version 0.8.18

### Github Hash

-

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-

### Debug Serial Log output

```shell
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA 
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 7, s: 0, f: 0, n: 7 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 9B 00 00 00 00 00 00 00 00 9E F5 8D 
I: (#0) RX  59ms | 14 -86dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 8, s: 0, f: 0, n: 7 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 9B 00 00 00 00 00 00 00 00 9E F5 8D 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA 
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 9, s: 0, f: 0, n: 8 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 B9 00 00 00 00 00 00 00 00 3F 75 8E 
I: (#0) RX  59ms | 14 -84dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 10, s: 0, f: 0, n: 8 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 B9 00 00 00 00 00 00 00 00 3F 75 8E 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 11, s: 0, f: 0, n: 9 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 D7 00 00 00 00 00 00 00 00 5D 91 66 
I: (#0) RX  58ms | 14 -84dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 12, s: 0, f: 0, n: 9 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 D7 00 00 00 00 00 00 00 00 5D 91 66 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA 
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 13, s: 0, f: 0, n: 10 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A B8 F5 00 00 00 00 00 00 00 00 FC 11 65 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
[... 106 lines cut ...]
I: (#0) RX  59ms | 14 -82dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 40, s: 0, f: 0, n: 25 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A BA B7 00 00 00 00 00 00 00 00 95 98 C5 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA 
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 41, s: 0, f: 0, n: 26 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A BA D5 00 00 00 00 00 00 00 00 F7 29 74
I: (#0) RX  59ms | 14 -80dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 42, s: 0, f: 0, n: 26 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A BA D5 00 00 00 00 00 00 00 00 F7 29 74 
I: (#0) request timeout : 500ms from freq : 868000Hz
W: Desired frequency is out of EU legal range! (863 - 870MHz)
I: (#0) TX 915.00Mhz | 56  *  *  *  * 81 00 17 65 02 15 21 20 14 ED 4D BA 
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 43, s: 0, f: 0, n: 27 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A BA F3 00 00 00 00 00 00 00 00 96 9B 81
I: (#0) RX  59ms | 14 -81dBm | D6  *  *  *  *  *  *  *  * 02 01 17 3C B0 
 Tes Data SuccesI: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 44, s: 0, f: 0, n: 27 | p: 30
I: (#0) TX 868.00Mhz | 15  *  *  *  * 81 00 17 65 80 01 00 65 7A BA F3 00 00 00 00 00 00 00 00 96 9B 81 
I: (#0) request timeout : 500ms from freq : 868000Hz
```

### Error description

Here I use a HMS microinverter, and for DTU I use the E49 900M20S. I'm trying to read the data from the HMS inverter, but the problem is that the RX is right, but for the data of the inverter failed to parse and the data didn't come out. Maybe if anyone can help me ask for help.

aditiyagio, 2023-12-14:
### Error description
Here I use a HMS microinverter, and for DTU I use the E49 900M20S. I'm trying to read the data from the HMS inverter, but the problem is that the RX is right, but for the data of the inverter failed to parse and the data didn't come out. Maybe if anyone can help me ask for help. @lumapu

lumapu, 2023-12-14:
you only get packet number 2, but never number 1 and the last one.
Which inverter do you have?

aditiyagio, 2023-12-14:
i have inverter HMS-1000 Series. @lumapu

aditiyagio, 2023-12-18:
is there anything that can we check or test to find the root cause of my problem? @lumapu

lumapu, 2023-12-18:
have you tried a newer version since then?
You are the first one who has such a problem. How far is the distance to your inverter?
Do you think there is much other traffic in the 868MHz band?

aditiyagio, 2023-12-21:
Yeah, I'm using the latest firmware development, and I've tried it in a radius of 4 meters, but it's not working. And actually, the RX on my serial logs does not appear if the HOY_BOT_FREQ_KHZ settings are not changed to 915 MHz. and after I change the HOY_BOT_FREQ_KHZ, the Rx on the log serial appears. @lumapu

aditiyagio, 2023-12-21:
[image]

@lumapu

lumapu, 2023-12-21:
are you located in the US? Sorry didn't know that. Can you try to be more clear? 
What do you mean by "on my serial logs does not appear"? What appears once you change the BOOT_FREQ?

lumapu, 2024-01-01:
are there any news?

aditiyagio, 2024-01-01:
My location is in Southeast Asia, specifically in Indonesia. My problem remains the same, my device still cannot read data from the HMS-1000 inverter @lumapu

aditiyagio, 2024-01-01:
Do you have any suggestions regarding my problem? What steps should I take to be able to access data from the inverter? @lumapu

lumapu, 2024-01-02:
you don't have to change the `HOY_BOT_FREQ_KHZ` because you already got data from your inverter on `868MHz` (as I can see in your log above).
Can you have another try with the latest release (`0.8.36)` and post another log with this version?

aditiyagio, 2024-01-02:
okay, i will try first @lumapu

aditiyagio, 2024-01-02:
[image]

aditiyagio, 2024-01-02:
here's the log using firmware version 0.8.36 @lumapu

aditiyagio, 2024-01-02:
[image]

aditiyagio, 2024-01-11:
@lumapu please insight

lumapu, 2024-01-11:
can't see any RX in your Log. Do you know if your inverter sends in `868MHz` oder `915MHz` band?
Do you have a SDR (software defined radio) with you?

nazmibojan, 2024-01-20:
Hi @lumapu , I'm Nazmi and currently helping @aditiyagio in this project. In Indonesia I think we use the different frequency for sub-1GHz. We are using this DTU from Hoymiles: https://www.hoymiles.com/product/dtu/lite-s/ and it's worked. I think in Indonesia we use 915 or 923 MHz frequency. So if we use different frequency with Europe, in which part of the code that we need to change?

lumapu, 2024-01-20:
it's not that easy because of several points where you need to adjust.
Maybe I can try to add an option to select different frequency for the range you mentioned.
Do you know on which frequency the inverter starts? Do you have a SDR with you?

nazmibojan, 2024-01-20:
I see. Currently we dont have SDR but we consider to purchase it. Is RTL-SDR enough to detect it? Or we need Hack-RF in this case @lumapu ?

lumapu, 2024-01-20:
Yes RTL SDR should be enough. You only need to measure 915MHz for HMS / HMT series.

nazmibojan, 2024-01-20:
Oke @lumapu . We will get back to you soon. Thanks a lot

nazmibojan, 2024-01-24:
Hi @lumapu, we already check the RF signal using RTL-SDR. We think this inverter is using 915MHz frequency. You can check the picture belom.

[image]

We also record the video when running DTU device to communicate with HMS-1000: https://drive.google.com/file/d/11lKiyDBvDaRQ-CVo3md8meCGIAihnjdl/view?usp=sharing

Wdyt @lumapu ?

lumapu, 2024-01-24:
wow really good, I'll check it.

aditiyagio, 2024-01-28:
hello @lumapu , have you done to check for this?  I will wait for the good news ✌️

lumapu, 2024-01-28:
sorry had no time so far. I think I can build a special version for you, which ESP are you using?

lumapu, 2024-01-28:
please check `0.8.66` (https;//fw.ahoydtu.de). I've added a region selector to `/system` as well as the compared frequencies for this region.
Please reboot your DTU once you changend the region to have the ability to set the correct frequency.

As you have a original Hoymiles DTU with you be sure to turn it off once you try to communicate using AhoyDTU. It isn't possible to have two DTUs talking to the same inverter.

nazmibojan, 2024-01-29:
Thanks @lumapu !
We will try it tomorrow and get back to you about the results

lumapu, 2024-01-29:
please try the latest development Version `0.8.68` https://fw.ahoydtu.de

aditiyagio, 2024-01-30:
@lumapu wow this works, I'm using development3 firmware 0.8.67
[image]

aditiyagio, 2024-01-30:
[image]

@lumapu

lumapu, 2024-01-30:
thank you for the response. That is good news.

lumapu, 2024-01-31:
@nazmibojan how is your DTU performing with the new version?

nazmibojan, 2024-02-01:
Sorry @lumapu , maybe we can check this next week. Because our DTU device is currently using by another team until next week. We will get back to you soon Lukas

closed 2024-02-17 by lumapu