lumapu/ahoy #1272: dev 0.8.22 - 2023-12-13 takes up to 5 min before data is visible

rmayergfx, 2023-12-14:
ESP32 with NRF24L01.

Since update to V8 from the first connecttion to the first data in web interface tooks very long time,
the setup didn´t change, interval for inverter is still 30sec and in my case really fast enough, but it takes up to 5 minutes until all data is displayed in the web interface, equal a reboot or cold boot is done. This was in the past much, much faster.

how can this be fixed, only for the time after reboot?

blueline13, 2023-12-14:
Please try the following. Under "System Config", check the "Print whole traces in Log" box. Sounds really strange. I also have an ESP32(0.8.22) + NRF + CMT and have an interval of 10 seconds and have only received the data every 30 seconds to 10 minutes, rather in the range of 3-5 minutes with the 0.8.x versions on my NRF Chip. The CMT runs perfectly. After setting the checkmark and a restart, the data comes back at 10 second intervals from the NRF Chip. I have checked this several times. Here it takes about 2 minutes after the reset until the data comes from the NRF.

Maybe that will help you too.  I know it's not exactly your error, but it sounds pretty similar. Maybe it will help.

rmayergfx, 2023-12-15:
after update to 0.8.23 with the default setting sendinterval to 15 https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md the webinterface shows the data much faster, after changing back to 30 sec, it tooks again very long time until the values from the inverter are displayed.

> 11:21:54.095 **** Display off ****
11:22:02.094 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 2, s: 0, f: 0, n: 2 | p: 2
11:22:02.097 I: (#0) TX 27 CH61 | 80
11:22:02.599 I: (#0) request timeout: 500ms
11:22:02.599 -----
11:22:02.600 I: com loop duration: 506ms
11:22:02.600 -----
11:22:32.095 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3, s: 0, f: 0, n: 3 | p: 2
11:22:32.097 I: (#0) TX 27 CH75 | 80
11:22:32.599 I: (#0) RX  41ms | 27 CH40 | 81
11:22:32.600 I: (#0) Payload (14)
11:22:32.602 -----
11:22:32.602 I: com loop duration: 508ms
11:22:32.603 -----
11:23:02.094 I: (#0) Radio infos: -6 -6 -6 -6 -4 | t: 4, s: 1, f: 0, n: 3 | p: 2
11:23:02.098 I: (#0) TX 27 CH75 | 80
11:23:02.600 I: (#0) RX  39ms | 27 CH40 | 81
11:23:02.601 I: (#0) Payload (14)
11:23:02.603 -----
11:23:02.603 I: com loop duration: 509ms
11:23:02.603 -----
11:23:32.095 I: (#0) Radio infos: -6 -6 -6 -6 -2 | t: 5, s: 2, f: 0, n: 3 | p: 2
11:23:32.097 I: (#0) TX 27 CH75 | 80
11:23:32.599 I: (#0) RX  61ms | 27 CH23 | 81
11:23:32.600 I: (#0) Payload (14)
11:23:32.602 -----
11:23:32.602 I: com loop duration: 508ms
11:23:32.602 -----
11:24:02.095 I: (#0) Radio infos: -6 -6 -6 -6 0 | t: 6, s: 3, f: 0, n: 3 | p: 2
11:24:02.098 I: (#0) TX 27 CH75 | 80
11:24:02.600 I: (#0) RX  45ms | 27 CH40 | 01
11:24:02.601 I: (#0) RX  92ms | 27 CH23 | 02
11:24:02.602 I: (#0) RX 142ms | 23 CH23 | 83
11:24:02.603 I: (#0) Payload (42)
11:24:02.604 I: alarm ID incremented to 1
11:24:02.605 -----
11:24:02.605 I: com loop duration: 511ms
11:24:02.605 -----
11:24:04.095 **** Display on ****
11:24:32.094 I: (#0) Radio infos: -6 -6 -6 -6 0 | t: 7, s: 4, f: 0, n: 3 | p: 2
11:24:32.097 I: (#0) TX 27 CH75 | 80
11:24:32.599 I: (#0) RX  45ms | 27 CH40 | 81
11:24:32.600 I: (#0) Payload (14)
11:24:32.602 -----
11:24:32.602 I: com loop duration: 508ms
11:24:32.602 -----
11:25:02.095 I: (#0) Radio infos: -6 -6 -6 -6 0 | t: 8, s: 5, f: 0, n: 3 | p: 2
11:25:02.097 I: (#0) TX 27 CH75 | 80
11:25:02.600 I: (#0) RX  60ms | 27 CH23 | 01
11:25:02.601 I: (#0) RX  87ms | 27 CH23 | 02
11:25:02.602 I: (#0) RX 134ms | 23 CH03 | 83
11:25:02.603 I: (#0) Payload (42)
11:25:02.604 -----
11:25:02.605 I: com loop duration: 511ms
11:25:02.605 -----

The Values comes together with Display on, and i don´t know why it tooks so long to get the data into the web interface, because the inverter is still online and has all data, it should be possible to read all data stored in the inverter with the first connect.

rejoe2, 2023-12-15:
First few requests to the inverter are only for hardware data, alarms etc.. 

Esp. in case you have several inverters and poor rf conditions, this can lead to relatively late enqueueing of the production data requests.

To cope around that, in some 0.7 versions a feature called internally "fastNext" had been implemented. Let's see, if we could reimplement that in the (completely restructured) 0.8 code.

rmayergfx, 2023-12-15:
i have only one inverter and this one is about 3m away from AhoyDTU. Would be great to speed up this to get a faster result.

Henrik-89, 2024-01-03:
Hi all,
Hi @lumapu 
I am also using ESP32 with NRF24L01 and a HM-800. I am using a very slow inverter update intervall of 30s.
I think I had a quite fast startup time with data polling and showing the results on the web interface on v.0.7.xx DEV.

Now, I switched to 0.8.35 as well as 0.8.36. It looks like the data shows after roughly 5min after DTU reboot. I tried it several times with both versions.

--DTU RESTART
--Connection after 1:45min
--Data visible in web interface after roughly 04:45min

Here are the Webserial logs. Data visible ~12:43:50
>Webserial
12:39:49.579 I: (#0) RX  39ms | 27 CH23 | 95 81
12:39:49.580 I: (#0) Payload (14)
12:39:49.582 -----
12:39:49.582 I: com loop duration: 505ms
12:39:49.582 -----
12:40:19.579 -----
12:40:19.580 I: com loop duration: 503ms
12:40:19.580 -----
12:40:49.579 -----
12:40:49.579 I: com loop duration: 503ms
12:40:49.580 -----
12:41:19.579 -----
12:41:19.579 I: com loop duration: 503ms
12:41:19.580 -----
12:41:49.579 I: (#0) RX  40ms | 27 CH61 | 95 81
12:41:49.580 I: (#0) Payload (14)
12:41:49.582 -----
12:41:49.582 I: com loop duration: 505ms
12:41:49.582 -----
12:42:19.579 I: (#0) RX  38ms | 27 CH61 | 95 81
12:42:19.580 I: (#0) Payload (14)
12:42:19.582 -----
12:42:19.582 I: com loop duration: 505ms
12:42:19.582 -----
12:42:49.829 I: (#0) RX  70ms | 27 CH61 | 95 02
12:42:49.830 I: (#0) RX 133ms | 27 CH61 | 95 03
12:42:49.831 I: (#0) RX 174ms | 27 CH61 | 95 04
12:42:49.832 I: (#0) RX 243ms | 27 CH40 | 95 05
12:42:49.832 I: (#0) RX 288ms | 27 CH40 | 95 06
12:42:49.843 I: (#0) RX 329ms | 27 CH40 | 95 07
12:42:49.844 I: (#0) RX 374ms | 27 CH23 | 95 08
12:42:49.845 I: (#0) RX 415ms | 23 CH23 | 95 89
12:42:50.978 I: (#0) RX  31ms | 27 CH61 | 95 01
12:42:50.979 I: (#0) Payload (138)
12:42:50.980 -----
12:42:50.980 I: com loop duration: 903ms
12:42:50.980 -----
12:43:19.579 I: (#0) RX  38ms | 17 CH61 | 95 81
12:43:19.580 I: (#0) Payload (4)
12:43:19.581 -----
12:43:19.581 I: com loop duration: 504ms
12:43:19.582 -----
12:43:49.580 I: (#0) RX 132ms | 23 CH61 | 95 83
12:43:49.714 I: (#0) RX  33ms | 27 CH61 | 95 01
12:43:50.978 I: (#0) RX  32ms | 27 CH61 | 95 02
12:43:50.980 I: (#0) Payload (42)
12:43:50.980 I: alarm ID incremented to 1
12:43:50.981 -----
12:43:50.981 I: com loop duration: 904ms
12:43:50.982 -----

rmayergfx, 2024-01-03:
> Do you tested new versions?
yes. i am on 0,8,42.
Data is displayed on Live with 
15:39:55.208 I: [NTP]: 2024-01-03 14:39:55 UTC
.
.
.
15:45:24.093 **** Display on ****

At this point index pages shows me uptime 00:05:51
so it takes very long until the data is displayed on the web interface. And i don´t know why. It is faster when i set interval from 30s to 10s, but i only need the speedup at first connect.

lumapu, 2024-01-04:
I don't understand this behaviour. What happens if you have this version installed already in the morning when the inverter wakes up und reboot the DTU then?
For me the data is visible for 3 inverters within a minute - but yes I'm using a 5s intervall.

Hmm.. during writing I had a thought: Your idea is not to read all the version and alarm stuff in the beginning - you wan't AhoyDTU to read the live data first.

rejoe2, 2024-01-05:
> Hmm.. during writing I had a thought: Your idea is not to read all the version and alarm stuff in the beginning - you wan't AhoyDTU to read the live data first.

Yes, people are more interested in the live data... But in general, I really like the idea of having sth like the the old "fastNext" stuff. Technically, we could add sth like a variant of the "addimportant()" function to enqueue a live data request any time any command was executed other than live data. This would also lead to faster execution of the next live data request in cases like "GetLossRate".

rmayergfx, 2024-01-05:
> I don't understand this behaviour. What happens if you have this version installed already in the morning when the inverter wakes up und reboot the DTU then? For me the data is visible for 3 [...]

What did you not understand?
In version 0.7.x with the same settings (30s) the data was shown after 1 min.
with version 0.8.x it takes up to 5 min to show live data. so something has changed in the boot sequence

It has nothing to do with the time, e.g. in the morning. Right now the inverter is online and producing, if i reboot the system, it takes up to long time > 4 min to see any data in web on the live view. data is coming step by step, first i can see "Active Power Control" and after that when i see in Webserial **** Display on **** the rest of the data is available.

it seems to me that the settings 30s is blocking at boot to get the data faster. if i set it to 5s (not useful for me) it will change much faster. So i think the 30s should be ignored at boot to get the view as fast as possible.

rmayergfx, 2024-01-07:
WOW, with 0.8.45 the live is up within a few seconds, faster than the webserial is answering with 30s settings !

Perfekt to me.

Henrik-89, 2024-01-08:
I hereby confirm, I share the opinion of @rmayergfx with my own setup: ESP32 with NRF24L01 and a HM-800 with 30s update interval.

With the now installed version 0.8.44 the /live data shows up immediatly 5-10s after the reboot of the DTU.

Great solution! Thanks! Feel free to close this issue :-)

closed 2024-02-17 by lumapu