lumapu/ahoy #1275: Nur 4 Inverter möglich?

walter2323, 2023-12-16:
Leider kann ich nicht mehr als 4 Wechselrichter hinzufügen.
Lässt sich dies beheben?

## Hardware
  * [x] ESP8266

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2023-12-16:
Korrekt !

Mit einem ESP8266 sind ressourcenbedingt nur 4 Inverter möglich.

Mit einem ESP32 sind bis zu 16 Inverter möglich.

teletapi, 2023-12-16:
Ja hol dir einen ESP32 dann gehn 16 WR

walter2323, 2023-12-16:
Da liegt also das Problem!
Vielen Dank euch für die schnelle Lösung!!!!

dtuuser, 2023-12-20:
Was ist denn das für ein Display Menü?

walter2323, 2023-12-21:
> Was ist denn das für eine Display Anzeige mit der Kurve?

Lediglich die neueste Version von OpenDTU...

dtuuser, 2023-12-21:
Danke für die Rückmeldung. Dachte es wär ein verstecktes Feature von Ahoy

closed 2023-12-29 by lumapu