lumapu/ahoy #1276: Problem mit senden von Power Limits per "Live Ansicht" und MQTT

blueline13, 2023-12-16:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.24

### Github Hash

78b603e37331b0ba6c8632fb9ce5d0f4ed6fd398

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP32 & NRF & CMT2300
Wecheslrichter: TSUN M1600(#0), 2x HMS2000(#2 & #3) 1xHMS1000(#1)

### Error description

Sobald ein Limit über die Weboberfläche an den WR gesendet wird Verhält sich die DTU sehr merkwürdig.

**1) Wird das Limit über den NRF Chip an den WR(#0) gesendet (egal ob per Live Ansicht oder MQTT) wiederholt die DTU das LIMIT immer wieder.**
12:44:00.793 -----
12:44:01.084 I: wr: 47, rd: 25
12:44:02.293 I: (#0) sendControlPacket cmd: 0x0b
12:44:02.796 I: (#0) RX  45ms | 17 CH40 | 81
12:44:02.797 I: (#0)  has accepted power limit set point 100 with PowerLimitControl 1
12:44:02.798 -----
12:44:04.801 I: (#2) RX  84ms | 27 -40dBm | | 01
12:44:04.802 I: (#2) RX 134ms | 27 -40dBm | | 02
12:44:04.803 I: (#2) RX 183ms | 27 -40dBm | | 03
12:44:04.804 I: (#2) RX 233ms | 27 -40dBm | | 04
12:44:04.804 I: (#2) RX 274ms | 15 -40dBm | | 85
12:44:04.816 I: (#2) Payload (66)
12:44:04.818 -----
12:44:06.820 I: (#3) RX  84ms | 27 -55dBm | | 01
12:44:06.821 I: (#3) RX 133ms | 27 -55dBm | | 02
12:44:06.822 I: (#3) RX 183ms | 27 -55dBm | | 03
12:44:06.823 I: (#3) RX 233ms | 27 -54dBm | | 04
12:44:06.823 I: (#3) RX 274ms | 15 -55dBm | | 85
12:44:06.835 I: (#3) Payload (66)
12:44:06.837 -----
12:44:08.080 I: wr: 50, rd: 27
12:44:08.837 I: (#0) RX  45ms | 17 CH40 | 81
12:44:08.839 I: (#0)  has accepted power limit set point 100 with PowerLimitControl 1
12:44:08.840 -----
12:44:10.843 I: (#1) RX  81ms | 27 -52dBm | | 01
12:44:10.844 I: (#1) RX 130ms | 27 -52dBm | | 02
12:44:10.845 I: (#1) RX 177ms | 23 -52dBm | | 83
12:44:10.846 I: (#1) Payload (42)
12:44:10.848 -----
12:44:12.850 I: (#2) RX  80ms | 27 -40dBm | | 01
12:44:12.851 I: (#2) RX 130ms | 27 -40dBm | | 02
12:44:12.852 I: (#2) RX 180ms | 27 -40dBm | | 03
12:44:12.852 I: (#2) RX 230ms | 27 -40dBm | | 04
12:44:12.853 I: (#2) RX 270ms | 15 -40dBm | | 85
12:44:12.865 I: (#2) Payload (66)
12:44:12.867 -----
12:44:14.869 I: (#3) RX  80ms | 27 -55dBm | | 01
12:44:14.870 I: (#3) RX 130ms | 27 -55dBm | | 02
12:44:14.871 I: (#3) RX 179ms | 27 -55dBm | | 03
12:44:14.872 I: (#3) RX 230ms | 27 -55dBm | | 04
12:44:14.872 I: (#3) RX 270ms | 15 -55dBm | | 85
12:44:14.884 I: (#3) Payload (66)
12:44:14.886 -----
12:44:15.083 I: wr: 54, rd: 31
12:44:16.886 I: (#0) RX  45ms | 17 CH40 | 81
12:44:16.887 I: (#0)  has accepted power limit set point 100 with PowerLimitControl 1
12:44:16.888 -----
12:44:18.891 I: (#1) RX  83ms | 27 -53dBm | | 01
12:44:18.892 I: (#1) RX 132ms | 27 -53dBm | | 02
12:44:18.893 I: (#1) RX 179ms | 23 -52dBm | | 83
12:44:18.894 I: (#1) Payload (42)

**2) Irgendwann kommt dann diese Meldung(Zeitstempel von vorherigem Versuch):**
12:42:43.985 -----
12:42:44.080 I: wr: 3, rd: 87
12:42:44.080 W: send queue almost full, consider to increase interval, 84 of 100entries used

**3) Wird das Limit über das CMT and den WR(#1 oder #2 oder #3) gesendet wird sieht das so aus aber wird nicht wiederholt gesendet.**
14:07:59.228 -----
14:08:01.080 I: wr: 40, rd: 95
14:08:01.231 I: sendControlPacket cmd: 0x0b
14:08:01.735 I: (#2) RX  81ms | 27 -55dBm | | 01
14:08:01.736 W: Inverter serial does not match, got: 0x80148504, expected: 0x80148xxx
14:08:01.737 I: (#2) RX 130ms | 27 -55dBm | | 02
14:08:01.737 W: Inverter serial does not match, got: 0x80148504, expected: 0x80148xxx
14:08:01.748 I: (#2) RX 180ms | 27 -55dBm | | 03
14:08:01.749 W: Inverter serial does not match, got: 0x80148504, expected: 0x80148xxx
14:08:01.760 I: (#2) RX 230ms | 27 -55dBm | | 04
14:08:01.761 W: Inverter serial does not match, got: 0x80148504, expected: 0x80148xxx
14:08:01.772 I: (#2) RX 271ms | 15 -55dBm | | 85
14:08:01.773 W: Inverter serial does not match, got: 0x80148504, expected: 0x80148xxx
14:08:01.773 I: (#2) RX  69ms | 17 -40dBm | | 81
14:08:01.784 I: (#2)  has accepted power limit set point 5 with PowerLimitControl 1
14:08:01.785 -----
14:08:03.788 I: (#3) RX  69ms | 17 -40dBm | | 81
14:08:03.789 W: Inverter serial does not match, got: 0x80148783, expected: 0x80148504
14:08:03.790 I: (#3) RX  84ms | 27 -55dBm | | 01
14:08:03.791 I: (#3) RX 133ms | 27 -55dBm | | 02
14:08:03.801 I: (#3) RX 183ms | 27 -55dBm | | 03
14:08:03.802 I: (#3) RX 233ms | 27 -55dBm | | 04
14:08:03.803 I: (#3) RX 274ms | 15 -55dBm | | 85
14:08:03.804 I: (#3) Payload (66)
14:08:03.806 -----
14:08:05.807 I: (#0) RX  46ms | 27 CH23 | 01
14:08:05.808 I: (#0) RX  92ms | 27 CH03 | 02
14:08:05.809 I: (#0) RX 144ms | 27 CH03 | 03
14:08:05.810 I: (#0) RX 194ms | 27 CH03 | 84
14:08:05.811 I: (#0) Payload (62)
14:08:05.813 -----
14:08:07.816 I: (#1) RX  82ms | 27 -52dBm | | 01
14:08:07.817 I: (#1) RX 131ms | 27 -52dBm | | 02
14:08:07.818 I: (#1) RX 178ms | 23 -52dBm | | 83
14:08:07.819 I: (#1) Payload (42)
14:08:07.821 -----
14:08:08.085 I: wr: 43, rd: 98
14:08:09.824 I: (#2) RX  83ms | 27 -40dBm | | 01
14:08:09.825 I: (#2) RX 132ms | 27 -40dBm | | 02
14:08:09.826 I: (#2) RX 183ms | 27 -40dBm | | 03
14:08:09.827 I: (#2) RX 233ms | 27 -40dBm | | 04
14:08:09.827 I: (#2) RX 273ms | 15 -40dBm | | 85
14:08:09.839 I: (#2) Payload (66)
14:08:09.841 -----

**4) Danach sieht es in der Weboberfläche auch immer so aus:**
[image]

lumapu, 2023-12-16:
danke für das Feedback, muss ich untersuchen woran es liegt. Eigenartig ist der Unterschied zwischen HM und HMS

blueline13, 2023-12-17:
Moin.

Ich habe nochmal mit anderen FW getestet und habe unterschiedliche Ergebnisse bekommen:

V 0.8.10:
NRF: 
Setzen des Limit ist OK. Keine Auffälligkeiten im LOG. "Live Ansicht" des Limits beim Wechselrichter wird normal aktualisiert(1-2 Abfragezyklen)

CMT: 
Setzen des Limit ist OK. Keine Auffälligkeiten im LOG. "Live Ansicht" des Limits beim Wechselrichter wird normal aktualisiert(1-2 Abfragezyklen)

V 0.8.20:
NRF: 
Setzen des Limit ist OK. Keine Auffälligkeiten im LOG. "Live Ansicht" des Limits beim Wechselrichter wird erst nach grob 30-60 Sekunde aktualisiert.

CMT & HMS1000: 
Setzen des Limit ist OK. Keine Auffälligkeiten im LOG. "Live Ansicht" des Limits beim Wechselrichter wird erst nach grob 30-60 Sekunde aktualisiert.

CMT & HMS2000: 
Setzen des Limit ist OK. Aber wieder der Hinweis im Log mit falscher SN.  Live Ansicht des Limits beim Wechselrichter wird erst nach grob 30-60 Sekunde aktualisiert.

Ich werde dann erst mal mit der 0.8.10 weiter machen. Wenn du noch mehr Infos brauchst., oder noch etwas getestet werden kann .... Ich versuche gerne zu helfen ;-)

Ollipop030, 2023-12-17:
> Wird das Limit über den NRF Chip an den WR(#0) gesendet (egal ob per Live Ansicht oder MQTT) wiederholt die DTU das LIMIT immer wieder.

Ähnliches habe ich auch beobachtet, siehe #1267.

lumapu, 2023-12-17:
setzt bitte mal die Inverter Gap auf 400ms (nicht wie Standard auf 2000ms).

Wenn die Zeile `I: wr: 27, rd: 27` mit unterschiedlichen Werten auftritt passen die Einstellungen nicht zusammen, entweder das Intervall vergrößern oder eben die Gap-Time verringern. Wenn es von Interesse ist hätte ich auch mein Log dazu:

<details>
<summary>Log (2x HMS-2000-4T), Zuerst wird das Limit auf 50% gesetzt, dann wieder auf 100%</summary>

```
13:11:21.177 I: wr: 16, rd: 16
13:11:21.178 I: (#0) Radio infos: t: 108, s: 108, f: 0, n: 0 | p: 20
13:11:21.179 I: (#0) TX 863.00Mhz | 15 80 42 38 10 81 00 17 65 80 0b 00 65 7e e5 69 00 00 00 00 00 00 00 00 7b 65 0e 
13:11:21.686 I: (#0) RX  84ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 01 00 01 01 32 01 34 01 3e 01 3c 03 cd 03 cd 00 05 8d 
13:11:21.689 I: (#0) RX 133ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 02 31 bc 00 05 22 fa 01 b5 01 b5 01 33 01 35 01 33 ea 
13:11:21.702 I: (#0) RX 183ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 03 01 38 03 b0 03 c5 00 05 17 77 00 05 5d 5d 01 a3 01 
13:11:21.705 I: (#0) RX 233ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 04 01 a6 09 53 13 87 0e 51 ff ff 00 99 00 01 00 a6 80 
13:11:21.718 I: (#0) RX 274ms | 15 -66dBm | 95 80 42 38 10 81 00 17 65 85 00 03 01 20 2b 
13:11:21.731 I: (#0) Payload (66): 00 01 01 32 01 34 01 3e 01 3c 03 cd 03 cd 00 05 31 bc 00 05 22 fa 01 b5 01 b5 01 33 01 35 01 33 01 38 03 b0 03 c5 00 05 17 77 00 05 5d 5d 01 a3 01 a6 09 53 13 87 0e 51 ff ff 00 99 00 01 00 a6 00 03 
13:11:21.748 -----
13:11:21.148 I: (#1) Radio infos: t: 108, s: 107, f: 0, n: 1 | p: 20
13:11:21.149 I: (#1) TX 865.00Mhz | 15 81 31 36 66 81 00 17 65 80 0b 00 65 7e e5 69 00 00 00 00 00 00 00 00 7b 65 04 
13:11:22.656 I: (#1) RX  83ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 01 00 01 01 3d 01 3d 01 4e 01 4b 04 24 04 1b 00 03 bf 
13:11:22.659 I: (#1) RX 134ms | 27 -66dBm | 95 81 31 36 66 81 00 17 65 02 40 e1 00 03 2d d1 00 a4 00 a2 01 3c 01 3d 01 4a 96 
13:11:22.672 I: (#1) RX 182ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 03 01 4e 04 15 04 25 00 03 48 42 00 03 3a 78 00 a2 10 
13:11:22.675 I: (#1) RX 236ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 04 00 a4 09 43 13 87 0f a7 ff ff 00 a8 00 01 00 9f 66 
13:11:22.688 I: (#1) RX 272ms | 15 -65dBm | 95 81 31 36 66 81 00 17 65 85 00 01 cb 37 fe 
13:11:22.701 I: (#1) Payload (66): 00 01 01 3d 01 3d 01 4e 01 4b 04 24 04 1b 00 03 40 e1 00 03 2d d1 00 a4 00 a2 01 3c 01 3d 01 4a 01 4e 04 15 04 25 00 03 48 42 00 03 3a 78 00 a2 00 a4 09 43 13 87 0f a7 ff ff 00 a8 00 01 00 9f 00 01 
13:11:22.718 -----
13:11:22.719 I: com loop duration: 1541ms
13:11:22.719 -----
13:11:26.177 I: wr: 18, rd: 18
13:11:26.179 I: (#0) Radio infos: t: 109, s: 109, f: 0, n: 0 | p: 20
13:11:26.180 I: (#0) TX 863.00Mhz | 15 80 42 38 10 81 00 17 65 80 0b 00 65 7e e5 6e 00 00 00 00 00 00 00 00 4b 43 1f 
13:11:26.686 I: (#0) RX  84ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 01 00 01 01 32 01 33 01 3d 01 3b 03 cb 03 cb 00 05 8e 
13:11:26.689 I: (#0) RX 133ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 02 31 bc 00 05 22 fa 01 b5 01 b5 01 33 01 35 01 32 eb 
13:11:26.702 I: (#0) RX 183ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 03 01 37 03 ae 03 c3 00 05 17 77 00 05 5d 5e 01 a3 15 
13:11:26.705 I: (#0) RX 234ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 04 01 a7 09 4f 13 88 0e 49 ff ff 00 99 00 01 00 a6 8a 
13:11:26.718 I: (#0) RX 274ms | 15 -66dBm | 95 80 42 38 10 81 00 17 65 85 00 03 c0 0d c7 
13:11:26.730 I: (#0) Payload (66): 00 01 01 32 01 33 01 3d 01 3b 03 cb 03 cb 00 05 31 bc 00 05 22 fa 01 b5 01 b5 01 33 01 35 01 32 01 37 03 ae 03 c3 00 05 17 77 00 05 5d 5e 01 a3 01 a7 09 4f 13 88 0e 49 ff ff 00 99 00 01 00 a6 00 03 
13:11:26.748 -----
13:11:26.148 I: (#1) Radio infos: t: 109, s: 108, f: 0, n: 1 | p: 20
13:11:26.149 I: (#1) TX 865.00Mhz | 15 81 31 36 66 81 00 17 65 80 0b 00 65 7e e5 6e 00 00 00 00 00 00 00 00 4b 43 15 
13:11:27.656 I: (#1) RX  85ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 01 00 01 01 3d 01 3d 01 4d 01 4a 04 22 04 19 00 03 b9 
13:11:27.659 I: (#1) RX 135ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 02 40 e1 00 03 2d d1 00 a4 00 a2 01 3b 01 3d 01 4a 91 
13:11:27.672 I: (#1) RX 185ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 03 01 4e 04 14 04 23 00 03 48 42 00 03 3a 78 00 a2 17 
13:11:27.675 I: (#1) RX 234ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 04 00 a4 09 44 13 88 0f a1 ff ff 00 a8 00 01 00 9f 68 
13:11:27.688 I: (#1) RX 272ms | 15 -65dBm | 95 81 31 36 66 81 00 17 65 85 00 01 1c 55 4b 
13:11:27.701 I: (#1) Payload (66): 00 01 01 3d 01 3d 01 4d 01 4a 04 22 04 19 00 03 40 e1 00 03 2d d1 00 a4 00 a2 01 3b 01 3d 01 4a 01 4e 04 14 04 23 00 03 48 42 00 03 3a 78 00 a2 00 a4 09 44 13 88 0f a1 ff ff 00 a8 00 01 00 9f 00 01 
13:11:27.718 -----
13:11:27.719 I: com loop duration: 1541ms
13:11:27.719 -----
13:11:31.177 I: wr: 20, rd: 20
13:11:31.179 I: (#0) Radio infos: t: 110, s: 110, f: 0, n: 0 | p: 20
13:11:31.180 I: (#0) TX 863.00Mhz | 15 80 42 38 10 81 00 17 65 80 0b 00 65 7e e5 73 00 00 00 00 00 00 00 00 1b d6 c7 
13:11:31.686 I: (#0) RX  84ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 01 00 01 01 31 01 33 01 3d 01 3b 03 ca 03 ca 00 05 8d 
13:11:31.689 I: (#0) RX 134ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 02 31 bc 00 05 22 fa 01 b5 01 b5 01 33 01 35 01 32 eb 
13:11:31.702 I: (#0) RX 184ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 03 01 37 03 ad 03 c2 00 05 17 77 00 05 5d 5e 01 a3 17 
13:11:31.705 I: (#0) RX 234ms | 27 -66dBm | 95 80 42 38 10 81 00 17 65 04 01 a7 09 50 13 88 0e 43 ff ff 00 99 00 01 00 a6 9f 
[... cut ...]
13:12:07.618 I: (#1) RX 231ms | 27 -65dBm | 95 81 31 36 66 81 00 17 65 04 00 a5 09 45 13 87 0f d4 ff ff 00 aa 00 01 00 a0 2f 
13:12:07.631 I: (#1) RX 271ms | 15 -65dBm | 95 81 31 36 66 81 00 17 65 85 00 01 8f a0 2d 
13:12:07.644 I: (#1) Payload (66): 00 01 01 3d 01 3d 01 51 01 4e 04 30 04 26 00 03 40 e2 00 03 2d d2 00 a5 00 a3 01 3c 01 3d 01 4e 01 52 04 21 04 31 00 03 48 43 00 03 3a 79 00 a3 00 a5 09 45 13 87 0f d4 ff ff 00 aa 00 01 00 a0 00 01 
13:12:07.661 -----
13:12:07.662 I: com loop duration: 1484ms
13:12:07.662 -----

```
</details>

lumapu, 2023-12-17:
korrigiere - beim log durchschauen sind mir diese `Inverter serial does not match ...` aufgefallen.
Daher lieber 500ms (bisher habe ich es bei mir nicht erneut gesehen)

Ollipop030, 2023-12-17:
Gap auf 500, Intervall 6:

```
14:59:15.075 I: wr: 28, rd: 28
14:59:15.851 I: (#0) RX  45ms | 27 CH40 | 01
14:59:15.857 I: (#0) RX  91ms | 27 CH23 | 02
14:59:15.858 I: (#0) RX 148ms | 27 CH23 | 03
14:59:15.859 I: (#0) RX 177ms | 27 CH23 | 84
14:59:15.860 I: (#0) Payload (62)
14:59:15.863 -----
14:59:16.863 I: (#1) RX  45ms | 27 CH40 | 01
14:59:16.864 I: (#1) RX  92ms | 27 CH23 | 82
14:59:16.865 I: (#1) Payload (30)
14:59:16.866 -----
14:59:16.866 I: com loop duration: 1790ms
14:59:16.867 -----
14:59:21.075 I: wr: 30, rd: 30
14:59:21.076 I: (#1) sendControlPacket cmd: 0x0b
14:59:21.578 I: (#1) RX  34ms | 17 CH23 | 81
14:59:21.579 I: (#1)  has accepted power limit set point 30 with PowerLimitControl 0
14:59:21.579 -----
14:59:22.083 I: (#0) sendControlPacket cmd: 0x0b
14:59:22.585 I: (#0) RX  34ms | 17 CH23 | 81
14:59:22.585 W: Inverter serial does not match, got: 0x74953798, expected: 0x83924915
14:59:22.586 I: (#0) RX  35ms | 17 CH23 | 81
14:59:22.587 I: (#0)  has accepted power limit set point 150 with PowerLimitControl 0
14:59:22.597 -----
14:59:22.598 I: com loop duration: 1522ms
14:59:22.598 -----
14:59:27.075 I: wr: 30, rd: 30
14:59:27.577 I: (#0) RX  35ms | 17 CH23 | 81
14:59:27.578 I: (#0)  has accepted power limit set point 150 with PowerLimitControl 0
14:59:27.578 -----
14:59:28.584 I: (#1) RX  35ms | 17 CH23 | 81
14:59:28.585 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
14:59:28.586 I: (#1) RX  45ms | 27 CH40 | 81
14:59:28.586 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
14:59:28.597 I: (#1) RX  40ms | 27 CH40 | 81
14:59:28.598 I: (#1) Payload (14)
14:59:28.599 -----
14:59:28.599 I: com loop duration: 1523ms
14:59:28.610 -----
14:59:33.075 I: wr: 32, rd: 32
14:59:33.076 I: (#1) sendControlPacket cmd: 0x0b
14:59:33.578 -----
14:59:34.083 I: (#0) sendControlPacket cmd: 0x0b
14:59:34.585 I: (#0) RX  34ms | 17 CH23 | 81
14:59:34.585 I: (#0)  has accepted power limit set point 244 with PowerLimitControl 0
14:59:34.586 -----
14:59:34.586 I: com loop duration: 1510ms
14:59:34.587 -----
14:59:39.075 I: wr: 32, rd: 32
14:59:39.577 I: (#0) RX  34ms | 17 CH23 | 81
14:59:39.578 I: (#0)  has accepted power limit set point 244 with PowerLimitControl 0
14:59:39.578 -----
```

Gap auf 500, Intervall 10:

```
15:09:50.087 -----
15:09:50.087 I: com loop duration: 2007ms
15:09:50.087 -----
15:09:58.079 I: wr: 17, rd: 17
15:09:58.080 I: (#1) sendControlPacket cmd: 0x0b
15:09:58.582 -----
15:09:59.084 I: (#0) sendControlPacket cmd: 0x0b
15:09:59.586 I: (#0) RX  36ms | 17 CH23 | 81
15:09:59.586 I: (#0)  has accepted power limit set point 310 with PowerLimitControl 0
15:09:59.587 -----
[... 75 lines cut ...]
15:11:09.585 I: (#0)  has accepted power limit set point 295 with PowerLimitControl 0
15:11:09.586 -----
15:11:09.586 I: com loop duration: 1506ms
15:11:09.587 -----
15:11:18.079 I: wr: 23, rd: 23
15:11:18.080 I: (#1) sendControlPacket cmd: 0x0b
15:11:18.582 I: (#1) RX  56ms | 17 CH03 | 81
15:11:18.583 W: Inverter serial does not match, got: 0x83924915, expected: 0x74953798
15:11:18.650 I: (#1) RX  41ms | 11 CH61 | 81
15:11:18.650 -----
15:11:19.150 I: (#0) sendControlPacket cmd: 0x0b
15:11:19.652 I: (#0) RX  41ms | 11 CH61 | 81
15:11:19.653 W: Inverter serial does not match, got: 0x74953798, expected: 0x83924915
15:11:19.653 I: (#0) RX  50ms | 17 CH23 | 81
15:11:19.654 I: (#0)  has accepted power limit set point 335 with PowerLimitControl 0
15:11:19.665 -----
15:11:19.665 I: com loop duration: 1585ms
15:11:19.666 -----

```

Klappt schonmal besser, Limitänderung wird nur noch einmal quittiert. Nur `Inverter serial does not match` kommt noch ständig.

Frage an @lumapu und @reserve85:
Das Intervall dürfte doch auf Limitänderungen keine Auswirkungen haben, oder anders gefragt, das Limit wird doch direkt gesendet? Hätte kein Problem mit einem großen Intervall, solange es die Nulleinspeisung nicht verlangsamt.

blueline13, 2023-12-17:
Ich habe jetzt die neuen Werte mal eingetragen:

FW 0.8.25

Interval: 10s
Gap:    500ms

Leider kein Unterschied bei mir .
Bei NRF hat sich der TSUN mit der Kommunikation irgendwie verschluckt, das die DTU ihn als Off anzeigt aber er proudziert noch. Im Log kommt weiterhin bei jedem Zyklus obwohl er als OFF auf der Startseite angezeigt wird. Diese Meldung verschwindet erst nach einem Neustart der DTU. Trotzdem ist er nach kurzer Zeit OFF wieder als OFF angezeigt. Kann aber bis dahin den WR mit "TURN ON" und "TURN OFF" die Produktion starten und stoppen.

15:02:27.070 I: wr: 47, rd: 47
15:02:27.573 I: (#0) RX  56ms | 17 CH03 | 81
15:02:27.574 I: (#0)  has accepted power limit set point 100 with PowerLimitControl 1

Nach dem Wechsel zurück auf die 0.8.10 konnte die DTU wieder ohne Probleme mit dem TSUN kommunizieren und hat ihn nach der üblich Zeit wieder als Online auf der Statusseite angezeigt und die aktuellen Werte dazu.

Bei CMT kommt nach wie vor diese Meldung mit Serial aber nur noch einmal. Die laufen weiter und liefern auch Daten und lassen sich mit neuen Limit versorgen und jedes mal eine SN Meldung dazu. Am Ende des Testen und mehrfachen Setzen von Limits hörten nun die SN Meldungen auch nicht mehr auf.

War also leider nicht erfolgreich bei mir. Wieder zurück auf 0.8.10, die läuft gerade wieder ohne die Probleme.

Ollipop030, 2023-12-17:
Kurzer Nachtrag:

Im Log der Nulleinspeisung kommt jetzt sehr oft Timeout.

```
Dec 17 15:34:25 Pi-Hole python3[757]: 2023-12-17 15:34:25 INFO     Ahoy: Inverter "HM-1500": setting new limit from 375 Watt to 362 Watt
Dec 17 15:34:32 Pi-Hole python3[757]: 2023-12-17 15:34:32 INFO     Ahoy: Inverter "HM-1500": Limit timeout!
Dec 17 15:34:32 Pi-Hole python3[757]: 2023-12-17 15:34:32 INFO     Ahoy: Inverter "HM-300": setting new limit from 75 Watt to 72 Watt
Dec 17 15:34:33 Pi-Hole python3[757]: 2023-12-17 15:34:33 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     intermediate meter EMLOG: 41  Watt
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     Cut limit to 221 Watt, limit was higher than 10 percent of live-production
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     Ahoy: Inverter "HM-1500" reachable: True
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     Ahoy: Inverter "HM-300" reachable: True
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     powermeter EMLOG: 189  Watt
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     setting new limit to 465 Watt
Dec 17 15:34:45 Pi-Hole python3[757]: 2023-12-17 15:34:45 INFO     Ahoy: Inverter "HM-1500": setting new limit from 362 Watt to 387 Watt
Dec 17 15:34:52 Pi-Hole python3[757]: 2023-12-17 15:34:52 INFO     Ahoy: Inverter "HM-1500": Limit timeout!
Dec 17 15:34:52 Pi-Hole python3[757]: 2023-12-17 15:34:52 INFO     Ahoy: Inverter "HM-300": setting new limit from 72 Watt to 77 Watt
Dec 17 15:34:53 Pi-Hole python3[757]: 2023-12-17 15:34:53 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     intermediate meter EMLOG: 38  Watt
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     Cut limit to 218 Watt, limit was higher than 10 percent of live-production
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     Ahoy: Inverter "HM-1500" reachable: True
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     Ahoy: Inverter "HM-300" reachable: True
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     powermeter EMLOG: 181  Watt
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     setting new limit to 454 Watt
Dec 17 15:35:05 Pi-Hole python3[757]: 2023-12-17 15:35:05 INFO     Ahoy: Inverter "HM-1500": setting new limit from 387 Watt to 378 Watt
Dec 17 15:35:12 Pi-Hole python3[757]: 2023-12-17 15:35:12 INFO     Ahoy: Inverter "HM-1500": Limit timeout!
Dec 17 15:35:12 Pi-Hole python3[757]: 2023-12-17 15:35:12 INFO     Ahoy: Inverter "HM-300": setting new limit from 77 Watt to 75 Watt
Dec 17 15:35:13 Pi-Hole python3[757]: 2023-12-17 15:35:13 INFO     Ahoy: Inverter "HM-300": Limit acknowledged

```

Liegt das am höheren Abfrageintervall?

jogybaer0815, 2023-12-17:
Hatte beim Rumspielen mit Active Power Limit in der LiveAnsicht sogar die Verbindung verloren: Offline und Anzeige aus.

mqtt: active_PowerLimit sollte sich doch auch aktualisieren. Änderungen in der LiveAnsicht aktualisieren das Objekt nicht und umgekehrt auch leider nicht

lumapu, 2023-12-17:
ok, mein Test war mit einem HMS Wechselrichter, morgen kann ich mal mit einem NRF testen

blueline13, 2023-12-18:
@lumapu 

@lumapu

Ich habe eben auch mal das ganze mit einem ESP8266 & NRF & TSUN M1600 probiert. Kein Unterschied zu dem ESP32 mit 4 Invertern. Es kommt wieder im LOG.

Interval: 5 Sekunden
Gap: 700ms

08:31:38.624 -----
08:31:38.625 I: com loop duration: 509ms
08:31:38.625 -----
08:31:43.114 I: wr: 23, rd: 23
08:31:43.618 I: (#0) RX  44ms | 27 CH23 | 01
08:31:43.619 I: (#0) RX  90ms | 27 CH03 | 02
08:31:43.620 I: (#0) RX 143ms | 27 CH03 | 03
08:31:43.687 I: (#0) RX  51ms | 27 CH03 | 84
08:31:43.688 I: (#0) Payload (62)
08:31:43.690 -----
08:31:43.691 I: com loop duration: 575ms
08:31:43.691 -----
08:31:48.114 I: wr: 24, rd: 24
08:31:48.116 I: (#0) sendControlPacket cmd: 0x0b
08:31:48.619 I: (#0) RX  36ms | 17 CH03 | 81
08:31:48.620 I: (#0)  has accepted power limit set point 75 with PowerLimitControl 1
08:31:48.621 -----
08:31:48.621 I: com loop duration: 505ms
08:31:48.622 -----
08:31:53.114 I: wr: 24, rd: 24
08:31:53.618 I: (#0) RX  36ms | 17 CH03 | 81
08:31:53.619 I: (#0)  has accepted power limit set point 75 with PowerLimitControl 1
08:31:53.620 -----
08:31:53.620 I: com loop duration: 505ms
08:31:53.621 -----
08:31:58.114 I: wr: 25, rd: 25
08:31:58.618 I: (#0) RX  36ms | 17 CH03 | 81
08:31:58.619 I: (#0)  has accepted power limit set point 75 with PowerLimitControl 1
08:31:58.620 -----
08:31:58.620 I: com loop duration: 505ms
08:31:58.621 -----
08:32:03.114 I: wr: 26, rd: 26
08:32:03.618 I: (#0) RX  36ms | 17 CH03 | 81
08:32:03.619 I: (#0)  has accepted power limit set point 75 with PowerLimitControl 1
08:32:03.620 -----
08:32:03.620 I: com loop duration: 505ms
08:32:03.621 -----
08:32:08.114 I: wr: 27, rd: 27
08:32:08.618 I: (#0) RX  36ms | 17 CH03 | 81
08:32:08.619 I: (#0)  has accepted power limit set point 75 with PowerLimitControl 1
08:32:08.620 -----
08:32:08.620 I: com loop duration: 504ms
08:32:08.621 -----

jogybaer0815, 2023-12-18:
> ok, mein Test war mit einem HMS Wechselrichter, morgen kann ich mal mit einem NRF testen

ist ja interessant. ich hatte mich wegen der selben Symptome hier angehängt. Nutze ja auch HMS1600

blueline13, 2023-12-18:
@lumapu  
Nach dem Update auf 0.8.27, kann ich bestätigen das es wieder geht mit ESP32 und ESP8266.

Wie immer eine Geschwindekit die kaum zu glauben ist wie hier neue Versionen entstehen. DANKE!!!!

closed 2023-12-18 by blueline13

jogybaer0815, 2023-12-19:
mqtt: active_PowerLimit wird aber leider noch nicht aktualisiert bzw. kann nicht über mqtt gesetzt werden