lumapu/ahoy #1278: Serialnumber / Seriennummer nicht eintragbar

WeWantCode, 2023-12-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.56

### Github Hash

3b4365a697d6dba1b8f50950405dc9a26df2b8db

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Currently in the beginning and still setting up my DTU Ahoy. 

Ich hatte zuerst die 0.7.36 drauf, allerdings wohl etwas auf meinem CMT Board falsch verbunden, da mir aus der Doku nicht klar war wo die zwei anderen Pins hin müssen und das IRQ = GPIO3 ist. 
Nach dem Upgrade auf 0.7.56 kann ich alle Ports einstellen und jetzt ist laut Ansicht mein CMT "aktiv" (siehe Screenshot)

[image]

Ich habe einen STM32, mit SH1106 OLED angeschlossen und ein CMT2300A. Am CMT und am STM32 hängen je ein Kondensator.
Das Display funktioniert. Das CMT bisher noch nicht.

### Debug Serial Log output

```shell
Unklar wo ich das finden kann, liefer ich nach, wenn ich mehr Infos kriege wie ich da dran komm.
```

### Error description

System Config meldet das Inverter-Serienummer nicht erlaubt ist. 

Mein Inverter hat Seriennummer 141 xxx xxx xxx. 
In der System Config wird das Feld der Inverternummer rot gefärbt. Wenn ich mit der Maus drüber geh kommt eine Hilfsmeldung "Bitte wählen Sie einen Wert, der nicht größer ist als 138999999999."

Kann ich eine neuere Version verwenden, damit meine Seriennummer geht? Kann ich bei den Settings was ändern, damit meine Seriennummer erlaubt wird?
Per Handy kann ich mich direkt auf die DTU des Inverters verbinden. Dort sehe ich auch das der Inverter aktiv ist und Leistung liefert.

WeWantCode, 2023-12-18:
Update : 
Mit 0.8.22 lässt sich die Seriennummer wohl eintragen. Kommunikation kommt aber trotzdem nicht zustande. Da mein Modul aber noch nie gefunkt hat würde ich vermuten das der Fehler irgendwo in meinem Aufbau liegt. 

Frage zum WLAN: 
Trage ich dort mein WLAN ein, das ich in der Interverter-DTU eingetragen habe (mein Haus-WLAN) oder trage ich dort das WLAN der Inverter DTU ein?

WeWantCode, 2023-12-18:
Alles klar, danke für die Hilfe. 

Seriennummer, erste 6 Stellen: 141292xx xxxx. Ja, ich habe eine Wifi-Version des Wechselrichters. Einen HMS-800W-2T

Die DTU des Inverters ist im Hoymiles Inverter verbaut. Wie man die abschalten kann ist mir nicht klar. Aber ich denke dann wäre der Wechselrichter auch nicht mehr per App erreichbar. Die DTU im Wechselrichter spannt ein WLAN auf und kann dort angesprochen werden. Weshalb diese die Kommunikation mit der Ahoy stören sollte versteh ich grad nicht. Ich dachte die Ahoy snifft mit was der Wechselrichter sendet?

Kann ich irgendwie helfen, wie die HMS-xxxW-2T unterstützt werden können. Bin selber Programmierer, wenn auch nicht Hardwarenah normalerweise, aber wenn mir jemand sagt was hilft oder welchen Trace man sich schnappen kann mach ich gern mit.

WeWantCode, 2023-12-18:
Ok, aus der Ahoy Website war mir nicht klar das die HMS-2T Modelle anders sind als die HMS Serie. Hab jetzt vermutlich die falsche Hardware gekauft, finde ich schade. Bitte ergänzen auf der Ahoy Site wenn möglich, damit der nächste nicht nochmal in die Falle tappt!

Für alle die mehr wissen will, kann man vermutlich hier: https://github.com/lumapu/ahoy/issues/1189 weiterlesen.

knickohr, 2023-12-18:
Nicht HMS-2T, die funktionieren 😉

HMS-**W**-2T

DanielR92, 2024-06-29:
The comment is already written on the main site: 
⚠️ Warning: HMS-XXXXW-2T WiFi inverters are not supported. They have a 'W' in their name and a DTU serial number on its sticker

Can be closed here.

stefan123t, 2024-07-11:
Wir sollten die Eintragung von HMS-xxx**W** (WiFi Modelle mit built-in DTU-BI) nach Möglichkeit mit einem Warnhinweis versehen.

Die wöchentliche DTU und Inverter Serialnumber sind übrigens identisch:

DTU-SN: 4143 923 12345
Mikro-SN: 1412 923 12345

Loetnase, 2024-07-11:
@stefan123t 
Das finde ich super von dir.
Vieleicht kann der Fehler von vorne herein vermieden werden.
Danke

lumapu, 2024-07-11:
> Wir sollten die Eintragung von HMS-xxx**W** (WiFi Modelle mit built-in DTU-BI) nach Möglichkeit mit einem Warnhinweis versehen.
> 
> Die wöchentliche DTU und Inverter Serialnumber sind übrigens identisch:
> 
> DTU-SN: 4143 923 12345 Mikro-SN: 1412 923 12345

hast du eine Liste der Wifi-Modell Seriennnummern?

stefan123t, 2024-07-14:
@lumapu nein leider noch nicht!
Aber wir haben bei OpenDTU mal angefangen mit einer Liste, aber ich habe bisher nur HMS-800W-2T nach dem o.g. Schema.

DTU-SN: 4143 923 12345
Mikro-SN: 1412 923 12345

* https://github.com/tbnobody/OpenDTU/issues/714
* https://github.com/tbnobody/OpenDTU/issues/1324
* https://github.com/tbnobody/OpenDTU/issues/2061

* https://github.com/tbnobody/OpenDTU/issues/519
  hier geht es zwar speziell um HERF Inverter, die verwenden das für OEMs gedachte `sn92long()` Serial Number Format mit einem anderen Charset:
  * https://github.com/tbnobody/OpenDTU/issues/1751

stefan123t, 2024-07-14:
> @stefan123t Das finde ich super von dir. Vieleicht kann der Fehler von vorne herein vermieden werden. Danke

War doch Deine Idee :stuck_out_tongue_winking_eye: wenn mich nicht alles täuscht!

Wir müssten mal sehen ob die Serial IDs der HMS-xxx**W** wirklich unterschiedlich sind von den HMS-XXX ohne Built-In WiFi DTUBI ?

Könnte man die alte Tabelle mal mit den HMS-800 und HMT-800 Modell Nummern aktualisieren ?
Ich habe weder die HMS/HMT-Modelle noch ein HMS-xxx**W** Modell.

Seriennummern
https://github.com/lumapu/ahoy/blob/5ebfe5a8fb5bcc833b64a0dd083686031808cec8/doc/hoymiles-format-description.md#seriennummern

closed 2024-08-16 by lumapu