lumapu/ahoy #1279: [Bug]

lambda58tel, 2023-12-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.8

### Github Hash

972b033

### Build & Flash Method

ESP Tools (flash)

### Setup

[image]
[image]
[image]

### Debug Serial Log output

```shell
16:11:09 I: hoymileshm800/ch1/U_DC: 29.400 V
16:11:09 I: hoymileshm800/ch1/I_DC: 0.020 A
16:11:09 I: hoymileshm800/ch1/P_DC: 0.700 W
16:11:09 I: hoymileshm800/ch1/YieldDay: 301.000 Wh
16:11:09 I: hoymileshm800/ch1/YieldTotal: 337.997 kWh
16:11:09 I: hoymileshm800/ch1/Irradiation: 0.171 %
16:11:09 I: hoymileshm800/ch1/MaxPower: 1.600 W
16:11:09 I: hoymileshm800/ch2/U_DC: 29.400 V
16:11:09 I: hoymileshm800/ch2/I_DC: 0.030 A
16:11:09 I: hoymileshm800/ch2/P_DC: 0.800 W
16:11:09 I: hoymileshm800/ch2/YieldDay: 293.000 Wh
16:11:09 I: hoymileshm800/ch2/YieldTotal: 339.944 kWh
16:11:09 I: hoymileshm800/ch2/Irradiation: 0.195 %
16:11:09 I: hoymileshm800/ch2/MaxPower: 1.700 W
16:11:09 I: hoymileshm800/ch0/U_AC: 242.700 V
16:11:09 I: hoymileshm800/ch0/F_AC: 50.010 Hz
16:11:09 I: hoymileshm800/ch0/Temp: 10.900 °C
16:11:09 I: hoymileshm800/ch0/ALARM_MES_ID: 1.000 
16:11:09 I: hoymileshm800/ch0/YieldDay: 594.000 Wh
16:11:09 I: hoymileshm800/ch0/YieldTotal: 677.941 kWh
16:11:09 I: hoymileshm800/ch0/P_DC: 1.500 W
16:11:09 I: hoymileshm800/ch0/MaxPower: 3.200 W
16:11:09 I: 
16:11:09 I: (#0) CH qualities: -2 -2 -4 -4 2 | t: 24, s: 10, f: 12, n: 2
16:11:09 I: heuristic test mode
16:11:09 I: (#0) TX 27 CH3 | 15 84 61 18 11 82 92 63 00 80 0b 00 65 80 61 0d 00 00 00 00 00 00 00 00 6f 8e 69 
16:11:09 I: (#0) RX  76ms 27 CH40, -75dBm | 95 84 61 18 11 84 61 18 11 02 28 4d 00 05 2f e8 01 2d 01 25 09 7b 13 89 00 00 d0 
16:11:09 I: (#0) RX 127ms 23 CH40, -75dBm | 95 84 61 18 11 84 61 18 11 83 00 00 00 00 00 00 00 6e 00 01 56 42 6d 
16:11:09 W: (#0) frame 1 missing: request retransmit (3 attempts left)
16:11:09 I: (#0) TX 11 CH3 | 15 84 61 18 11 82 92 63 00 81 0b 
16:11:10 I: (#0) RX  34ms 27 CH40, -75dBm | 95 84 61 18 11 84 61 18 11 01 00 01 01 25 00 02 00 07 01 25 00 03 00 07 00 05 91 
16:11:10 I: (#0) Payload (42): 00 01 01 25 00 02 00 07 01 25 00 03 00 07 00 05 28 4d 00 05 2f e8 01 2d 01 25 09 7b 13 89 00 00 00 00 00 00 00 00 00 6e 00 01 
16:11:10 I: (#1) CH qualities: -6 -6 -6 -6 -6 | t: 35, s: 0, f: 11, n: 24
16:11:10 I: (#1) TX 865.00Mhz | 15 91 62 38 84 81 00 17 65 80 01 00 65 80 61 0e 00 00 00 00 00 00 00 00 95 90 a7 
16:11:10 I: (#1) RX  30ms 0, -57dBm | 
16:11:10 W: Inverter serial does not match
16:11:11 I: (#1) CH qualities: -6 -6 -6 -6 -6 | t: 36, s: 0, f: 12, n: 24
16:11:11 I: (#1) TX 865.00Mhz | 15 91 62 38 84 81 00 17 65 80 01 00 65 80 61 0f 00 00 00 00 00 00 00 00 05 9d 3b 
16:11:11 I: (#1) request timeout: 500ms
16:11:11 I: (#1) TX 868.00Mhz | 56 91 62 38 84 81 00 17 65 02 15 21 14 14 2d 5b aa 
16:11:14 I: Iv: 0
```

### Error description

Ich habe dieses Board noch nicht lange und mit den neuen HMS WRs noch keine Erfahrung. Mein HM800 läuft nach wie vor top, nur den neuen HMS bekomme ich nicht zum reden. Laut Log stimmt wohl meine SN nicht, diese habe ich aber bereits mehrfach kontrolliert, sowie Inverter entfernt, reboot und wieder neu eingetragen (ohne Erfolg). Beide Funkmodule werden als "connected" deklariert. Ich hatte vorher die v. 0.7.36 aber mit dieser Version habe ich das Pinout nicht zum laufen gebracht. Ich würde mich über jede Hilfe sehr freuen. VG Lambda

blueline13, 2023-12-18:
@lambda58tel  Schau dir bitte mal diesen Eintrag ab der Stelle https://github.com/lumapu/ahoy/issues/1237#issuecomment-1847481126  an. Dein GPIO3 ist nicht belegt im Setup. Könnte schon das Problem sein.

knickohr, 2023-12-18:
Genau das isses !  Ahoy benötigt diesen Pin.

lambda58tel, 2023-12-18:
Dankeschön, genau das ist das Problem.