lumapu/ahoy #1280: Yield Correction Nachkommastellen "rotes Fenster"

Dirk-Ahoy, 2023-12-18:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.27

### Github Hash

d31f392

### Build & Flash Method

was already installed

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Yield Correction mit Nachkommastelle macht einen roten Kasten, speichern geht  zwar - aber ob es genau genommen wird kann ich nicht sagen ;)

knickohr, 2023-12-18:
Der rote Kasten sagt das etwas nicht stimmt. Eben die Nachkommastellen 😉

lumapu, 2024-01-01:
ist im Release `0.8.36` schon gefixt, daher mache ich hier gleich zu 😉

closed 2024-01-01 by lumapu