lumapu/ahoy #1281: mqtt: active_PowerLimit [Bug]

jogybaer0815, 2023-12-19:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.27

### Github Hash

d31f39286ee570b2f373a03934115455dad7b63b

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP32 & CMT2300
HMS1600

### Debug Serial Log output

```shell
10:33:00.094 I: (#0) Radio infos: t: 8, s: 8, f: 0, n: 0 | p: 10
10:33:00.095 I: sendControlPacket cmd: 0x0b
10:33:00.096 I: (#0) TX 863.00Mhz | 81
10:33:00.599 I: (#0) RX  70ms | 17 -37dBm | | 81
10:33:00.600 I: (#0)  has accepted power limit set point 100 with PowerLimitControl 1
10:33:00.600 -----
10:33:00.601 I: com loop duration: 508ms
10:33:00.601 -----
10:33:15.093 I: (#0) Radio infos: t: 9, s: 9, f: 0, n: 0 | p: 10
10:33:15.095 I: (#0) TX 863.00Mhz | 80
10:33:15.599 I: (#0) RX  83ms | 27 -37dBm | | 81
10:33:15.600 I: (#0) Payload (14)
10:33:15.601 -----
10:33:15.602 I: com loop duration: 509ms
10:33:15.602 -----
10:33:30.094 I: (#0) Radio infos: t: 10, s: 10, f: 0, n: 0 | p: 10
10:33:30.095 I: sendControlPacket cmd: 0x0b
10:33:30.096 I: (#0) TX 863.00Mhz | 81
10:33:30.599 I: (#0) RX  66ms | 17 -37dBm | | 81
10:33:30.600 I: (#0)  has accepted power limit set point 30 with PowerLimitControl 1
10:33:30.600 -----
10:33:30.601 I: com loop duration: 508ms
10:33:30.601 -----
```

### Error description

mqtt: active_PowerLimit wird leider nicht aktualisiert bzw. kann nicht über mqtt gesetzt werden #1276 ist ja als gelöst gelabelt. Setzen in der Live-Ansicht geht, wird jedoch nicht zum mqtt-Broker kommuniziert.

Bei mir bleibt das Objekt `hms1600/Terrasse/ch0/active_PowerLimit` immer auf 100(rot) bzw wird auch nicht nach Wertänderung in der Ahoy-Liveansicht aktualisiert

blueline13, 2023-12-19:
@jogybaer0815 
Ich hatte das gestern alles erst geprüft bevor ich das als gelöst markiert hatte. An welcher Stelle schreibst du neuen den Wert?

Der richtige Punkt wäre unter dem DTU Namen (xxx) und dann XXX/ctrl/limit/0 wenn du nur einen WR hast.

knickohr, 2023-12-19:
Er hat es akzeptiert, steht auch so im Log. Was in der Anzeige kommt und geht, dauert eine Weile, je nachdem wenn es wieder abgefragt wird.

Die richtigen Topics sind :

zum setzen AhoyDTU-Name/ctrl/limit/Inverter-ID
für das Acknowledge AhoyDTU-Name/Inverter-Name/ack_pwr_limit
für den Wert AhoyDTU-Name/Inverter-Name/ch0/active_PowerLimit

AhoyDTU-Name, Inverter-ID und Inverter-Name mußt Du bei Dir entsprechend anpassen. Inverter-ID und Inverter-Name sind nicht das gleiche !

blueline13, 2023-12-19:
Als Ergänzung das Limt wurde über die Weboberfläche gesetzt. Wenn es über MQTT kommt sieht das so aus

11:14:11.646 I: (#2) Payload (66)
11:14:11.648 -----
**11:14:12.287 I: MQTT got topic: DTU/ctrl/limit/0**
11:14:12.451 I: (#3) RX  82ms | 27 -51dBm | | 01
11:14:12.452 I: (#3) RX 132ms | 27 -51dBm | | 02
11:14:12.453 I: (#3) RX 191ms | 27 -51dBm | | 03
11:14:12.454 I: (#3) RX 235ms | 27 -51dBm | | 04
11:14:12.454 I: (#3) RX 277ms | 15 -51dBm | | 85
11:14:12.465 I: (#3) Payload (66)
11:14:12.467 -----
11:14:12.468 I: com loop duration: 3392ms
11:14:12.468 -----
11:14:14.077 I: (#0) sendControlPacket cmd: 0x0b
11:14:14.579 I: (#0) RX  41ms | 17 CH40 | 81
11:14:14.580 I: (#0)  has accepted power limit set point 50 with PowerLimitControl 1
11:14:14.580 -----
11:14:15.383 I: (#1) RX  81ms | 27 -57dBm | | 01

Das Acknowledge vom Inverter wird immer erst im nächsten Abfrage Zyklus geholt. Wenn das Acknowledge da ist, dann wechselt die Liveanzeige beim Limt auf "n.a" und dann einen weiterer Abfragezyklus später wird es dann auch in der Live Ansicht richtig wieder gegeben. Das kann bei dir schon mal 30 Sekunden dauern.

Wenn es schneller gehen soll dann unter den "Settings/Inverter/Intervall" die Zeit auf 5 Sekunden runter setzen.

jogybaer0815, 2023-12-19:
Vielen Dank für die Unterstützung. Ich teste alles jetzt erstmal im IOBroker

jogybaer0815, 2023-12-19:
Nochmals Danke. Es klappt nun alles. Mein Irtum war, dass ich Lesen und Setzen des Wertes über ein Topic machen wollte.
Bitte hier schliessen

closed 2023-12-29 by lumapu