lumapu/ahoy #1282: Multiple alarms when using 2 inverters

technics42, 2023-12-19:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.65
Github Hash: 50ca076

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

When using 2 inverters (HM-600 and HM-300) I get alarms for both of them very often. The alarms are always ID 2 "DTU command failed".
What is the reason for this?
How to fix this issue or just ignore it?

When only using one inverter (HM-600) I never encountered such alarms.
Polling interval is 9 seconds.
Transmit power level is set to min.
RSSI is -64dBm for both inverters.

lille2000, 2023-12-19:
See this item #1202 
Solved with >=0.8.23

technics42, 2023-12-19:
Thank you - I didn't have this problem in the used version,yet, and also reported it to that issue.
It really seems only occuring if one uses multiple inverters.

closed 2023-12-19 by technics42