lumapu/ahoy #1283: Einstellen des PowerLimits

nobe2023, 2023-12-21:
Hallo,
ich haben an der Ahoy-DTU 2 Invertier (HM-800 und HM-1500) aktiviert.
Ich wollte für einen der Inverter (HM-800) das Output Powerlimit auf 50% setzen.
Die Schaltfläche wo ich das tun kann finde ich nicht und wollte nach einem Tipp fragen wie die Einstellung vornehmen kann? 
Lt einer Anleitung gab / gibt ein eine Möglichkeit nach % oder Watt zu reduzieren.

Vielen Dank für die Hilfe.

## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.27
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

zappoXX, 2023-12-21:
Live Ansicht ... Klick auf "Active Power Control"!!

nobe2023, 2023-12-21:
Check – thanks so mutch
Hätte ich nie gefunden.

closed 2023-12-21 by stefan123t