lumapu/ahoy #1285: [Bug]  yield-of-day (YD) zeigt manchmal unnatürlich hohe Werte an

strepwo, 2023-12-21:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.22 (+ some modifications)

### Github Hash

eaafe63 :: 0.8.22

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

ein HM 300
sonst alles default-Werte (Intervall = 30 sec, kein Mqtt, ...)
Sourcecode etwas modifiziert, aber an ganz anderer Stelle

### Error description

Moin,
ich benutze FW 0.8.22 (mit ein paar eigenen Änderungen), und habe einen komischen Effekt bemerkt:
der yield-of-day (YD) zeigt manchmal unnatürlich hohe Werte an. Heute morgen z.B. ein trüber Himmel, und es sollen schon 290 Wh reingegangen sein. Nach einem reboot zeigt er einen sehr realistischen Wert von 2 Wh an (ja, es ist nur ein kleines 300 Watt-Modul.. ). 

Habe etwas herumgesucht, und folgendes gefunden: da gibt es seit Version 0.8.3 eine Korrektur-Logik (Variablen _mOffYD, mLastYD_), die anscheinend den YD korrigiert, falls der WR z.B. tagsüber mal kurz aus- und wieder anschaltet (wegen z.B. einer sehr dunklen Wolke) - der WR fängt wieder bei YD = 0 an --> doof.
Die Logik erkennt diesen Sprung nach unten, und addiert den zuletzt erhaltenen 'guten' YD drauf. Damit das am nächsten Tag weiter funktioniert, muss die Variable _mOffYD_ nachts gelöscht werden; das passiert in _app::tickMidnight() --> resetAlarms() --> mOffYD = 0_.
Ich weiss (durch die eigenen Änderungen), dass heute nacht _app::tickMidnight()_ aufgerufen wurde, und dass gestern der letzte YD = 288 Wh war. Und heute morgen zeigt er eben 290 Wh an.. interessant: 290 = 288 + 2
Das kann eigentlich nur bedeuten, dass _resetAlarms()_ doch nicht aufgerufen wurde, die YD-Korrektur-Logik dadurch heute morgen einen 'pseudo-Sprung' erkannt hat, und die 288 Wh von gestern draufaddiert hat.

Der Aufruf von _resetAlarms()_ wird geklammert von einem _if(InverterStatus::OFF == iv->status)_ --> mein WR wurde offensichtlich nicht als OFF markiert, obwohl es Mitternacht war..
Im Normalbetrieb scheint _Inverter.status = InverterStatus::OFF_ nur in _Inverter.isAvailable()_ gesetzt zu werden - fehlt da vllt. ein Aufruf von _isAvailable()_, vor dem _if(InverterStatus::OFF == iv->status)_ ?

Möglicherweise dasselbe wie in Bug https://github.com/lumapu/ahoy/issues/1257 ?

[image]
[image]

strepwo, 2023-12-22:
beim Durchgucken im Code gibt es noch 2-3 Stellen, wo Inverter::status in ähnlicher Situation (also 'von ausserhalb') abgefragt wird:
 - in pubMqtt.h, processIvStatus(): dort steht allerdings vorher diese Zeile:
                 iv->isProducing(); // recalculate status
 - in pubMqttIvData.h, stateSend(): dort gibt es kein Update in Form von isProducing(); ist wahrscheinlich aber vorher in processIvStatus() gemacht worden --> vermutlich ok
  - RestApi.h, getInverter(): hier fehlt auch ein Update (Aufruf von isProducing()), und könnte ein Problem sein
  
 Da wäre es ingesamt vllt. besser, in hmInverter.h einen kleinen Wrapper zu machen:
         _InverterStatus getStatus(){
            isProducing(); // recalculate status
            return status;
        }_
  und diese Funktion dann an den obigen 3 Stellen zu benutzen, und in app::tickMidnight()

lumapu, 2023-12-29:
hat sich das Verhalten verbessert?

strepwo, 2023-12-30:
ja Problem ist weg, danke  :-)
ich schliesse den Bug

closed 2023-12-30 by strepwo