lumapu/ahoy #1286: Inkonsistenz beim Parameter dis_night_comm

roku133, 2023-12-22:
Konfiguration: Minianlage mit HM-400 und einem 410-W-Solarpanel

In den Inverter-Settings ist "Pause communication during night"  aktiviert, das Verhalten nachts ist auch entsprechend.

Rest API / Index: Parameter dis_night_comm true
MQTT: Parameter dis_night_comm false

Interessant ist in diesem Zusammenhang, dass nach dem Upgrade von v0.7.58 auf v0.8.27 die vorher aktive Nachtabschaltung deaktiviert worden  ist und anschließend wieder aktiviert worden ist.

knickohr, 2023-12-22:
Siehe #1265

roku133, 2024-01-01:
Nach meinem Verständnis soll der Parameter dis_night_comm jeweils die Inverter-Einstellung unter Advanced "Pause communication during night" wiedergeben. 
Offensichtlich wird der Parameter jedoch über Nacht verändert, heute Morgen hat er in v0.8.36 wieder die oben beschriebene Inkonsistenz gezeigt.
Nach erneutem Setzen und Reboot hat der MQTT-Parameter dis_night_comm wieder den korrekten Wert true.

knickohr, 2024-01-01:
Aufpassen !

Es gibt diesen dis_night_comm sowohl für jeden Inverter einzeln, als auch als globalen Status. Die Inverterspezifischen Topics geben die Einstellung wieder, der globale ist in meinen Augen unsinnig. Dafür gibt es einen weiteren globalen Topic comm_disabled, dieser gibt den Status der tatsächlichen Kommunikation wieder.

Wobei der comm_disabled zur Zeit auch nur zur Hälte funktioniert. Er schaltet zwar nachts auf true, wechselt aber morgens nicht mehr auf false.

[image]

Siehe hierzu auch #1309

roku133, 2024-01-01:
@knickohr Das ist bei mir anders:
es gibt nur einen inverterspezifischen Parameter dis_night_comm.
Das mag daran liegen, dass ich nur einen Wechselrichter in Betrieb habe.

mosquitto_sub -v -u xxxx -P yyyy -t HM-inverter/# | grep dis
HM-inverter/comm_disabled true
HM-inverter/HM-400/dis_night_comm false

Seit Beginn der Nachtabschaltung zeigt dis_night_comm wieder den falschen Wert.

knickohr, 2024-01-01:
Es könnte auch sein das dieser globale Topic mittlerweile nicht mehr existiert 😉 Das Chanelog sagt, das es mit der 29 gefixed wurde. Scheint wohl ein Relikt zu sein.

Ach, dann ist der Payload bei den inverterspezifischen Topics bei mir gar nicht falsch. Habe nich schon gewundert 😅

lumapu, 2024-01-01:
zwei Bugs und doch einer, sollte in der näcshten Verison `0.8.39` behoben sein

roku133, 2024-01-02:
Auch mit v0.8.41 wird der Parameter dis_night_comm während der Nachtabschaltung wieder auf den falschen Wert gesetzt.

stefan123t, 2024-10-29:
@roku133 kannst Du das Problem bitte mal mit der v0.8.152 nochmals prüfen und das issue ggf. schließen ?

roku133, 2024-10-30:
@stefan123t: es gibt keine Veränderung, der Bug ist in v0.8.152 weiter vorhanden.

Hier noch einmal eine Zusammenstellung:
mosquitto_sub -v -u xxxx -P yyyy -t HM-inverter/# | grep dis

tagsüber:
HM-inverter/comm_disabled false
HM-inverter/HM-400/dis_night_comm true

nachts:
HM-inverter/comm_disabled true
HM-inverter/HM-400/dis_night_comm false ⭍

REST API zum Vergleich, tagsüber und nachts sind die beiden Parameter identisch:
1. inverter/list/inverter/0
disnightcom:     true
2. index
disNightComm:    true

stefan123t, 2024-10-30:
@knickohr Du hast doch die beiden verwandten Issues #1265 und #1309 geöffnet und als behoben geschloßen.
Kannst Du das Problem auf dem Mosquitto hier nachvollziehen ?

| Topic | Example Value | Remarks | Retained |
|---|---|---|---|
| `dis_night_comm` | true | setting if night communication is disabled | true |

https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/publisher/pubMqtt.h#L175-L185

vs. 

https://github.com/lumapu/ahoy/blob/2d5d62eb0e2b35c743db2b75079fb039c0f9c191/src/app.cpp#L276-L292

@lumapu hier in Zeile 276 wird der Wert auf die Negierung `iv->commEnabled = !iv->config->disNightCom;` gesetzt, 
vielleicht ist das ja die Ursache ?

knickohr, 2024-10-31:
Mein, allerdings nutze ich die API nicht und auch die Einstellung der einzelnen Inverter nicht, nur die globale dis_night_comm