lumapu/ahoy #1287: 0.8.28 Active Power Control for inverter Endlosschleife

mtet2, 2023-12-23:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.28
Github Hash: 5a52677

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

0.8.28 auf ESP8266 mit 4 Invertern (HM1500, HM600, HM600, HM600)
Inverter 0 an PV
Inverter 1 an Batterie
Inverter 2 deaktiviert
Inverter 3 deaktiviert

Interval 8 S
Inverter Gap 200 ms

Das Auslesen und Anzeigen der aktiven Inverter funktioniert prima.

Fehler:
Sende ich an Inverter1 einen "TURN OFF" oder "TURN ON" Befehl in der Weboberfläche, dann wird dieser Befehl beliebig oft wiederholt:

[image]
[image]

[Ahoy_log_0_8_28.txt](https://github.com/lumapu/ahoy/files/13759587/Ahoy_log_0_8_28.txt)

rejoe2, 2023-12-23:
Gegen Ende ist die loop dann durch.

Anscheinend gibt es Probleme, wenn man in der Startphase commands sendet.

Tritt das Problem auch auf, wenn du danach/jetzt commands schickst?

mtet2, 2023-12-23:
Ja, siehe Anhang. 
Ich habe im laufenden Betrieb 1x "TURN ON"  und dann 2 oder 3x "TURN Off" damit die Battrie nicht noch weiter entladen wird. Die schleife hört erst auf, wenn ich die Ahoy-DTU "reboote"

[Ahoy_log_0_8_28_2.Versuch.txt](https://github.com/lumapu/ahoy/files/13759821/Ahoy_log_0_8_28_2.Versuch.txt)

mtet2, 2023-12-23:
Noch ein Nachtrag von 19:53 Uhr.
Der Alarmzähler ist jetzt schon im 2000er Bereich.

[image]

lumapu, 2023-12-23:
ah das liegt bestimmt daran, dass nie `closeRequest` in der `Communication.h` aufgerufen wird

closed 2023-12-29 by lumapu