lumapu/ahoy #1294: [Bug] 0.8.29 Pin settings not working for ESP8266

You69Man, 2023-12-28:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.29

### Github Hash

4e277e20c8b0b1663f365dd1d1c14fd6432311c5

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

Pin Settings 

### Error description

1) esp32pins werden nur unter   /*IF_ESP32*/ angelegt.
Das führt in function parsePinout beim einem ESP8266 dann zu einem "Uncaught ReferenceError: esp32pins is not defined"

2) In function parseDisplay:
if("ESP8266" == type) pinList = esp8266pirpins;        // -> muss  esp8266pins sein!
Die spezielle Einschränkung für den PIR Pin beim ESP8266 auf A0 habe ich dann damals erst weiter unten gemacht und würde dann wieder so funktionieren:
var esp8266pirpins = [[255, "off / default"],
                                      [17,  "A0"]];
                document.getElementById("pirPin").append(
                    ml("div", {class: "row mb-3"}, [
                        ml("div", {class: "col-12 col-sm-3 my-2"}, "PIR sensor"),
//                        ml("div", {class: "col-12 col-sm-9"}, sel("pir_pin", pinList, obj["pir_pin"]))  -> entfernen!
                        ml("div", {class: "col-12 col-sm-9"}, sel("pir_pin", ("ESP8266" == type) ? esp8266pirpins : pinList, obj["pir_pin"]))  // -> hinzufügen
                    ])
                );

lille2000, 2023-12-28:
Es fehlen bei mir alle Einstellmöglichkeiten (Inverter, Pins, Display,…) bei 0.8.29 unter „Settings“.  (Esp8266)

You69Man, 2023-12-28:
Das wäre der fix dazu: #1295

You69Man, 2023-12-28:
> Es fehlen bei mir alle Einstellmöglichkeiten (Inverter, Pins, Display,…) bei 0.8.29 unter „Settings“. (Esp8266)

Das liegt daran, dass die fehlenden esp32pins beim ESP8266 einen javascript Fehler ergibt. Daher wird die Webseite nicht mehr richtig aktualisiert.
Fix #1295 liegt aber schon bereit zum mergen.

closed 2023-12-29 by lumapu