lumapu/ahoy #1296: OLED 120x64 Nachkommastellen

knickohr, 2023-12-28:
Bitte in Display_Mono_128X64.h folgende beiden Zeilen im development03-Branch ändern.

Zeile 124 :
snprintf(mFmtText, DISP_FMT_TEXT_LEN, "%.2f kWh", mDisplayData->totalYieldDay / 1000.0);

Zeile 130:
snprintf(mFmtText, DISP_FMT_TEXT_LEN, "%.2f MWh", mDisplayData->totalYieldTotal / 1000.0);

Ich kann es leider nicht testen und einen PR bekomme ich auch nicht hin. Also bitte kritisch prüfen. Kann auch sei. Das es ab 99,99kWh bzw. 99.99MWh zu einem Platzproblem kommt, wobei letzteres wohl in den nächsten Jahrzehnt ? niemand erreichen wird 😉 Ebenso könnte es auch sein das der Wert nicht mehr zentriert steht, dazu reichen leider meine Kenntnisse in dieser Programmiersprache nicht.

You69Man, 2023-12-28:
Ich übernehm das, nachdem ich das ja auch 'verschlimmbessert' habe (wobei, MWh gab's vorher noch gar nicht, möchte ich auch dazu sage ;))
Und wenn schon, dann natürlich auch für das Nokia!

knickohr, 2023-12-28:
Die MWh laß nur drin. Ich denke das ich spätestens in 3 Jahren den Sprung in der Anzeige mache. Wobei ich ja das ePaper verwende, da ist dann ja alles wieder anders. Habe ich mir noch gar nicht angeschaut. Und bei den OLEDs mit noch weniger Auflösung habe ic/ ebenfalls nicht geschaut. Aber machen wir erst mal das für 128x64.

You69Man, 2023-12-28:
> Die MWh laß nur drin. Ich denke das ich spätestens in 3 bis 5 Jahren den Sprung in der Anzeige mache. Wobei ich ja das ePaper verwende, da ist dann ja alles wieder anders. Habe ich mir noch gar nicht angeschaut. Und bei den OLEDs mit noch weniger Auflösung habe ic/ ebenfalls nicht geschaut. Aber machen wir erst mal das für 128x64.
> 
> Man könnte das natürlich auch fließend machen. Jedesmal wenn vorne (links) eine Stelle dazu kommt, rücken wir das Komma ggfs. um eine Stelle Nach und lassen die letzte Stelle (rechts) raus fallen. Sind dann halt ein paar if-Abfragen mehr drin. Ich denke 4 Digits mit Komma bekommen wir hin.

Das ePaper greife ich nicht an (weil ich keins habe, und somit nichts testen kann).
Die Änderung für die OLED und Nokia ist kein Problem, sieht aber noch etwas anders aus, damit es vom Layout für alle Fälle passt.

knickohr, 2023-12-28:
Das meinte ich mit nicht mehr zentriert und so 😉

Das ePaper laß, da ist offenbar schon „optimiert“ worden. Geht da auch nicht so einfach weil man da die Textbreite in Pixel beachten muß wegen dem Partial Refresh.

closed 2023-12-29 by lumapu