lumapu/ahoy #1300: Version 8.30      added info if grid profile was not found

Gubi2023, 2023-12-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

8.31

### Github Hash

????

### Build & Flash Method

was already installed

### Setup

Im setup Read Grid Profile abgewählt

### Error description

ich habe in den Settings Read Grid Profile abgewählt und erhalte trotzdem die Meldung:   Grid Profile for inverter HM600
Unknown Profile  (Etwas verwirrend...)

knickohr, 2023-12-29:
Vielleicht hat die DTU es noch gar nicht gelesen ?  Es dauert eine Weile bis es ausgelesen wird.

Gubi2023, 2023-12-29:
[image]
ich meine, der "Show" button sollte gare nicht angezeigt werden, wenn in den Settings Read Grid Profile abgewählt wird.

Gubi2023, 2023-12-29:
[image]
auch nicht schlecht, die Lösung.
Respekt, wie ihr immer so schnell reagiert!