lumapu/ahoy #1301: [Bug] Release ZIP ahoy_v0.8.34 is empty

bentolor, 2023-12-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0-7-36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

not relevant

### Error description

Thanks for the new release.

Nota bene: The release ZIP at https://github.com/lumapu/ahoy/releases/tag/ahoy_v0.8.34 only contains the readme, but not release artefact.

lumapu, 2023-12-29:
oh, thank you for reporting

lumapu, 2023-12-29:
fixed manually

closed 2023-12-29 by lumapu