lumapu/ahoy #1302: ESP32 8.34 bin lässt sich bei mir nicht updaten

dtuuser, 2023-12-29:
Die 
231229_ahoy_0.8.34_3178325_esp32.bin
aus 
https://fw.ahoydtu.de/fw/release/ahoy_v0.8.34/ESP32/
lässt sich bei mir nicht einspielen. 
Nach dem Reboot ist die alte Version noch drauf. Habe wohl noch die 31 drauf.

Dirk-Ahoy, 2023-12-29:
mach mal einen reboot über die Weboberfläche und wenn sie dann noch nicht da ist noch mal probieren - hier und da hängt das update schon mal.

knickohr, 2023-12-29:
Updaten, wenn success kommt, sofort den Stecker ziehen. Aber nicht vor dem success !

Und wenn das auch nicht hilft, in den AP Booten und da das Update machen.

dtuuser, 2023-12-29:
Hammer..... hat geklappt.  
Reboot über Weboberfläche und dann die gleiche bin nochmal geupdatet.

Danke für den Tipp und die super Programmierarbeit

closed 2023-12-29 by dtuuser