lumapu/ahoy #1303: Settings für W5500

Edibo63, 2023-12-30:
## Hardware
  * [ ] ESP8266
  * [ x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _0.8_.36__
Github Hash: _______

### Build & Flash Method:
* [ x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

Habe einen ESP32 mit W5500 zum laufen gebracht. Ich finde unter Settings keine Einstellung für Ethernet.
Ist da nichts oder finde ich es blos nicht.
Danke.

lumapu, 2023-12-30:
was genau willst du einstellen? Die statische IP aus dem WLAN sollte auch für ETH funktonieren

lumapu, 2023-12-30:
sorry habe versehentlich auf den falschen Knopf gedrückt, Issue ist natürlich noch nicht geschlossen 😉

Edibo63, 2023-12-30:
Da alles funktioniert will ich nichts mehr verstellen. Ich dachte es gibt ein Menü für die Pinbelegung wie beim NRF24
und ich nicht wußte ob ich die richtigen Pins benutzt habe bzw. eine bestimmte Belegung voreingestellt ist.

lumapu, 2023-12-30:
ist hart codiert aktuell:
https://github.com/lumapu/ahoy/blob/6fb05359398f24e14f4ddc3e7304b55b24fdd53c/src/config/config.h#L41-L59

kann auch wie hier in `platformio.ini` angepasst / überschrieben werden:
https://github.com/lumapu/ahoy/blob/6fb05359398f24e14f4ddc3e7304b55b24fdd53c/src/platformio.ini#L182-L187

Edibo63, 2023-12-31:
An die config.h habe ich nicht gedacht.
Habe die Belegung aus #886 übernommen.
4 war CE vom NRF24 und das wollte ich eigentlich so lassen und fürs W5500 einen anderen nehmen.
CE vom NRF24 ist jetzt 17.
Danke.

closed 2024-02-17 by lumapu