lumapu/ahoy #1304: Unknown Grid Profile for inverter TSUN TSOL-400

yam1yam1, 2023-12-30:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.34

### Github Hash

3178325

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter: 15 sec

### Error description

Grid Profile for inverter TSUN TSOL-400 (wird als HM-350 erkannt)
Unknown Profile

03 01 20 00 00 08 08 FC 07 30 00 01 0A 55 00 01 09 E2 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 07 02 8A 0A 55 07 30 14 1E 12 8E 00 50 00 1E 40 00 07 D0 03 E8 70 00 00 01

stefan123t, 2023-12-30:
Wir haben das selbe Grid Profile bereits einmal auf der OpenDTU Wiki Seite dokumentiert:
https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

`03 01 20 00 00 08 08 fc 07 30 00 01 0a 55 00 01 09 e2 10 00 13 88 12 8e 00 01 14 1e 00 01 20 00 00 01 30 07 02 8a 0a 55 07 30 14 1e 12 8e 00 50 00 1e 40 00 07 d0 03 e8 70 00 00 01 1a 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Ich habe es mal mit der dort angegebenen CRC16 Checksum `1a 45` ins Binärformat konvertiert und mit `./dtu.py` in der S-Miles Cloud hochgeladen um zu sehen was Hoymiles dazu sagt: `DE_VDE4105_2011`

#### Grid Profile

Name | Value | Unit
-- | -- | --
**DE_VDE4105_2011** |   | -
H/LVRT |   | -
H/LFRT |   | -
Islanding Detection (ID) |   | -
Reconnection (RT) |   | -
Ramp Rates (RR) |   | -
Active Power Control (APC) |   | -
Nominal Voltage (NV) | 230 | V
Low Voltage 1 (LV1) | 184 | V
LV1 Maximum Trip Time (MTT) | 0.1 | s
High Voltage 1 (HV1) | 264.5 | V
HV1 Maximum Trip Time (MTT) | 0.1 | s
10mins Average High Voltage (AHV) | 253 | V
Nominal Frequency | 50 | Hz
Low Frequency 1 (LF1) | 47.5 | Hz
LF1 Maximum Trip Time (MTT) | 0.1 | s
High Frequency 1 (HF1) | 51.5 | Hz
HF1 Maximum Trip time (MTT) | 0.1 | s
ID Function Activated | 1 |  
Reconnect Time (RT) | 65 | s
Reconnect High Voltage (RHV) | 264.5 | V
Reconnect Low Voltage (RLV) | 184 | V
Reconnect High Frequency (RHF) | 51.5 | s
Reconnect Low Frequency (RLF) | 47.5 | s
Short Interruption Reconnect Time (SRT) | 8 | s
Short Interruption Time (SIT) | 3 | s
Normal Ramp up Rate (RUR_NM) | 20 | Rated%/s
Soft Start Ramp up Rate (RUR_SS) | 10 | Rated%/s
APC Function Activated | 1 |

stefan123t, 2023-12-30:
I have used this code to convert the Hex Dump into a binary grid_profile.bin for upload via `./dtu.py`.
https://github.com/rovo89/hoymiles_proto/pull/1

Maybe @rovo89 can make use of the script/s `hex2profile.py` and `profile2hex.py` based on the below code in his repo too.

##### hex2profile.py

```python3
#!/bin/python3

import struct

def reverse_bytes(binary_data):
    byte_list = list(binary_data)
    for i in range(0, len(byte_list), 2):
        byte_list[i], byte_list[i+1] = byte_list[i+1], byte_list[i]
    reversed_data = bytes(byte_list)
    return reversed_data

def binary_to_hex_dump(file_path):
    with open(file_path, 'rb') as file:
        data = file.read()
    hex_dump = ''
    for byte in reverse_bytes(data):
        hex_code = format(byte, '02X')
        hex_dump += hex_code + ' '
    return hex_dump.strip()

def hex_dump_to_binary(hex_dump, output_file):
    hex_dump = hex_dump.replace(" ", "")
    hex_bytes = [hex_dump[i:i+2] for i in range(0, len(hex_dump), 2)]
    binary_data = bytes.fromhex("".join(hex_bytes))
    reversed_data = reverse_bytes(binary_data)
    with open(output_file, "wb") as file:
        file.write(reversed_data)

#hex_dump = "03 00 20 00 00 0A 08 FC 07 30 00 1E 0B 3B 00 01 04 0B 00 1E 09 E2 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 03 02 58 0A C8 07 A3 13 92 12 8E 40 00 07 D0 03 E8 50 08 00 01 13 9C 01 90 00 10 01 F6 13 74 70 02 00 01 27 10 80 00 00 00 08 5B 01 2C 08 B7 09 41 09 9D 01 2C 90 00 00 00 FF A1 B0 00 00 00 01 F4 00 5F A0 02 00 00 00 00"
hex_dump = "03 01 20 00 00 08 08 fc 07 30 00 01 0a 55 00 01 09 e2 10 00 13 88 12 8e 00 01 14 1e 00 01 20 00 00 01 30 07 02 8a 0a 55 07 30 14 1e 12 8e 00 50 00 1e 40 00 07 d0 03 e8 70 00 00 01 1a 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"

#file_path = 'grid_profile_original.bin'
#hex_dump = binary_to_hex_dump(file_path)
#print(hex_dump)

file_path = 'grid_profile.bin'
hex_dump_to_binary(hex_dump, file_path)

hex_dump = binary_to_hex_dump(file_path)
print(hex_dump)
```

tbnobody, 2023-12-30:
Should already get parsed correctly on OpenDTU

lille2000, 2023-12-31:
Seit dieser Änderung (0.8.37) schließt „Show Grid“ sofort das Fenster der Inverterdetails bei mir (esp8266).

lumapu, 2023-12-31:
mein Fehler, JSON hat einen Syntaxfehler, wird mit der nächsten Version repariert

yam1yam1, 2023-12-31:
OK, Grid Profile TSOL400/HM-350 wird mit Version 0.8.38 richtig angezeigt.

closed 2024-02-17 by lumapu