lumapu/ahoy #1307: [Bug]  0.8.36 läuft nicht auf ESP8266, 0.8.34 schon / 0.8.36 won't work on ESP8266, 0.8.34 does

cptechnik, 2024-01-01:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Set 
SSID,-Wifi password, inverter sn, 

### Error description

Online flash tool installiert 0.7.36
von 0.7.36 oder von 0.8.34 aus (Aktualisierung auf 0.8.34 kein Problem)
aktualisiere ich übers WebIF auf 0.8.36 oder 0.8.38 
Nach dem boot, zeigt Daten im Display an, WebIF ist aber nicht oder nur kurz zu erreichen, bricht Datenübertragung ab, und ist dann nicht mehr zu erreichen.
Serielle Konsole zeigt immer
`I: com loop duration: 1503ms`
Flashen mit dem Online-Flash-Tool auf 0.7.36 klappt immer wieder

cptechnik, 2024-01-01:
ok one step ahead...
ich habe die wlan einstellungen weggelassen und er läuft stabil... 
(sofern man das ohne Sonne sagen kann)
... ergo... WLAN

cptechnik, 2024-01-01:
ähhhh... ch13?

... ich habe mal mein WLAN auf Kanal 9 geändert,.... war vorher auf Kanal 13... könnte das ein Problem sein... in der neuen Firmware (Generation)?

I changed my wifi channel from 13 to 9 and i have no problems... until now... 
check-check... waiting for errors...

Jumperm, 2024-01-01:
Ähnliches Problem bei mir. Habe Release und aktuelle Dev 0.8.38 frisch geflashed.
Bin auf 0.8.30 zurück und die läuft gerade.
Im Log stand etwas von:
I: comm ...

cptechnik, 2024-01-01:
> Ähnliches Problem bei meinem ESP8266.
Also auch das mit Kanal 13, lässt sich das mit dem Verstellen im Router beheben?

Jumperm, 2024-01-02:
In meinem Mesh ist auf dem nahem Accessppoint Kanal 13 aktiv. Trotzdem hatte ich extreme Übertragungsfehler durch die träge DTU.

Jumperm, 2024-01-02:
> > Ähnliches Problem bei meinem ESP8266.
> 
> Also auch das mit Kanal 13, lässt sich das mit dem Verstellen im Router beheben?
> 
> ...also ich kann nämlich sagen, durch das verstellen des WLAN Kanals im Router/Accesspoint funktioniert die Ahoy wie sie soll...

Sehr Spannend.

Ich habe den Kanal 1 fixiert und nun läuft die DTU butterweich...

Jumperm, 2024-01-02:
Scheint doch ein Problem des ESP zu sein. Nach einem Kaltstart schlägt die Verbindung fehl.
https://olimex.wordpress.com/2021/12/10/avoid-wifi-channel-12-13-14-when-working-with-esp-devices/

closed 2024-02-17 by lumapu