lumapu/ahoy #1308: Sunrise & Sunset

zappoXX, 2024-01-01:
Moin,

habe mich heute einmal mit der Option "Sunrise & Sunset" beschäftigt. Ich kann dort den Offset (pre sunrise, post sunset) einstellen, also vor Sonnenaufgang, nach Sonnenuntergang, allerdings nur zusammen. Nun ist es aber so das es bei Sonnenaufgang noch ziemlich düster ist, die Radio Statistics schon ordentlich "No answer" zählen und dadurch die Statistik verfälscht ist. Besser wäre es die beiden Punkte "pre sunrise, post sunset separat zu behandeln, damit man nur z.B. 20 min 
nach Sonnenaufgang einstellen kann.

Andy-Op, 2024-01-01:
Mal Danke dass das hier Thema ist. Ich suche das Leistungsmerkmale in Verbindung mit einer PIN-Ausgabe am ESP um eine Nachteinspeisung zu steuern. Ich würde dann von OpenDTU hier her wechseln. Hat sich da schon jemand dran versucht?

[image]

DanielR92, 2024-01-01:
Guten Morgen zusammen,

@zappoXX hmm da könnte ich dir Recht geben. Zwecks Statistik kannst du jedoch (je nachdem was du einsetzt) selbst dafür sorgen das die Daten "richtig" aufgenommen werden...

@Andy-Op Verstehe ich dich richtig, du möchtest ein Pin am Ahoy so Einstellen das sobald "produziert" wird, etwas am Ausgang High/Low schaltet?

Andy-Op, 2024-01-01:
@DanielR92 ich möchte tagsüber einen 100 Ah 9S LiFePO4 laden (Tag, Zeit zwischen Sunrise und Sunset low) und Nachts als Grundlastkompensation einspeisen (Nacht, Zeit zwischen Sunset und Sunrise high). Wenn ihr Sunrise und Sunset schon habt muss es doch ein Leichtes sein den Zeitabschnitt zu bestimmen, alle Minute zu prüfen, ein Flag zu setzen und an einem PIN des ESP auszugeben.

knickohr, 2024-01-01:
Ihr könnt solche Wünsche in die Wunschliste #1199 eintragen.

Bezüglich unterschiedlichen Settings wäre es relativ einfach das zu implementieren.

Bezüglich Ein-/Ausgabe über Pins halten wir uns etwas zurück. Denn ansonsten wurden wir da eine Lawine ins Rollen bringen. Außerdem ist es beim ESP8266 eh schon „aus die Maus“.  Ja, beim ESP32 gibt es genügend.

knickohr, 2024-01-02:
Na schaut mal an !  So schnell kann’s gehen 😉

Dirk-Ahoy, 2024-01-02:
klappt aber noch nicht auch nach speichern kommt unter System Fehlermeldung "Invalid Date"   bei Start/Stop. - siehe Screenshot im Discord

Ghost30121978, 2024-01-02:
Hallo
Hier ein Screenshot 
[image]

roku133, 2024-01-02:
Es ist ein reines Anzeigeproblem auf der Weboberfläche.
MQTT-Parameter comm_start und comm_stop bzw. REST-API ts_offsSr und ts_offsSs sind korrekt.

knickohr, 2024-01-02:
Gut zu wissen 👍

zappoXX, 2024-01-02:
Super wie schnell neue Features eingefügt werden... wenn jetzt noch bei Sunrise & Sunset für jeden Zeitpunkt, also Sonnenaufgang und Sonnenuntergang positiver wie negativer Offset, z.B. Sonnenaufgang 08:38 Uhr +20 min. (um 08:38 Uhr ist es nämlich noch nicht wirklich hell und der Wechselrichter sendet auch noch keine Daten, bei mir heute Morgen um 08:59 Uhr) dann ist diese Funktion perfekt.

lumapu, 2024-01-02:
ok, leider ist grad die `0.8.40` draußen 😉 - in der nächsten Verison

roku133, 2024-01-02:
Anzeige funktioniert jetzt in v0.8.41 👍

knickohr, 2024-01-02:
Heute fliegen die Versionen wieder 😂

Andy-Op, 2024-01-02:
Macht mir doch bitte noch die Tag/Nacht Pin-Ausgabe. Dann teste ich ob das so zu gebrauchen ist.

knickohr, 2024-01-02:
Eine LED die den Status von comm_disabled anzeigt 😉

lumapu, 2024-01-02:
@knickohr das ist eine geniale Idee, wie es dann doch wieder für alle passt 😉

knickohr, 2024-01-02:
Das ist doch genau das was @Andy-Op will. Machen wir’s doch so wie bei den anderen LEDs 😉

zappoXX, 2024-01-03:
Die Offset Funktion funktioniert einwandfrei, ich habe 15 min. nach Sonnenaufgang eingestellt und genau dann startet auch die Kommunikation. Allerdings... unter System/Kommunikation Start wird alles richtig angezeigt, in der AhoyDTU Übersicht ist die Zeitangabe aber falsch...
[image]

[image]

lumapu, 2024-01-03:
wird in der nächsten Version behoben sein, danke für den Hinweis

knickohr, 2024-01-04:
Ahhh verdammt, Morgens muß man negative Werte eintragen wenn die Kommunikation schon vor Sonnenaufgang starten soll 🤪 Ist ja auch logisch 🫣

Der oben beschriebene Bug mit der falschen Anzeige ist aber nur ein Anzeigeproblem ?

lumapu, 2024-01-04:
ja genau, wird heute noch behoben 😊

closed 2024-02-17 by lumapu