lumapu/ahoy #1311: [Bug] Wemos S2 Pico support

ldrolez, 2024-01-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.36

### Github Hash

6fb0535

### Build & Flash Method

ESP Tools (flash)

### Setup

4 inverters + mqtt + oled display (included on wemos s2 board)

### Error description

PINs for the Wemos PICO S2 are not supported in the web gui.
[sch_s2_pico_v1.0.0.pdf](https://github.com/lumapu/ahoy/files/13805407/sch_s2_pico_v1.0.0.pdf)

- For NRF pins, IO 35-36-37 should be used but they can't be found in the GUI
- The OLED display is connected to IO 8-9, but they are not available in the GUI. I edited the html in inspector mode to force the right pin values

lumapu, 2024-01-01:
ist das ein ESP32 im RP2040 (Raspberry Pi Pico 2040) Format?
Kannst du hierfür einen PR einreichen?

ldrolez, 2024-01-02:
> ist das ein ESP32 im RP2040 (Raspberry Pi Pico 2040) Format? 
50x23mm a bit larger

ldrolez, 2024-01-02:
I found another bug with this board.
Ahoy boots only when I connect a serial console on USB! Never seen this on other boards.

So if you use a power supply, no boot. If you plug it into a PC, no boot, unless I start minicom (linux).

ldrolez, 2024-01-02:
Seems to be related to serial.flush https://github.com/espressif/arduino-esp32/issues/7554

ldrolez, 2024-01-02:
Is there a way to disable the serial console in settings? that could be a quick fix

ldrolez, 2024-03-20:
I've removed serial code and recompiled, and it boots without a computer

stefan123t, 2024-10-29:
@ldrolez could you provide a PR with your changes to support ESP32-S2 out of the box, 
ie. with some preprocessor macros and the necessary `#ifdef` soup to disable the serial code ?

ldrolez, 2024-12-15:
This is my diff, don't have time to find which macros to use sorry
```diff --git a/src/app.cpp b/src/app.cpp
index 0bba14d..3e4c0eb 100644
--- a/src/app.cpp
+++ b/src/app.cpp
@@ -14,9 +14,9 @@ app::app() : ah::Scheduler {} {}
 
 //-----------------------------------------------------------------------------
 void app::setup() {
-    Serial.begin(115200);
-    while (!Serial)
-        yield();
+    //Serial.begin(115200);
+    //while (!Serial)
+    //    yield();
 
     resetSystem();
 
@@ -24,7 +24,7 @@ void app::setup() {
     mSettings.getPtr(mConfig);
     ah::Scheduler::setup(mConfig->inst.startWithoutTime);
     DPRINT(DBG_INFO, F("Settings valid: "));
-    DSERIAL.flush();
+    //DSERIAL.flush();
     if (mSettings.getValid())
         DBGPRINTLN(F("true"));
     else
```