lumapu/ahoy #1312: Inverter setup [Bug]

geronet1, 2024-01-01:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus, square dot

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.38

### Github Hash

1b73e49

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

**

### Error description

Setup->Inverter
Bei schon vorhandenen Invertern mit "disabled" überschreibt er den letzten wenn man einen neuen hinzufügt.

Ebenso wenn man mehrere mit "disabled" hinzufügen will wird nur einer abgespeichert.

lumapu, 2024-01-01:
Hallo, ich habe es gerade getestet - es geht bei mir. Es waren 3 Inverter (disabled) schon eingetragen. Ich konnte problemlos einen vierten (auch disabled) hinzufügen.
Kann es sein, dass du schon 16 Inverter in der Liste hast? Den ESP32 habe ich aktuell auf diese Anzahl begrenzt.

geronet1, 2024-01-08:
Habs grade auch nochmal geprüft - auch mit der ETH Version. War wohl irgend ein Update Fehler.. jetzt geht alles.

closed 2024-01-08 by geronet1