lumapu/ahoy #1313: [Bug]

blueline13, 2024-01-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.39

### Github Hash

*

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

*

### Error description

Bin immer wieder verblüfft wie schnell Wünsche umgesetzt werden!!!!! Doch Leider hat sich vermutlich ein Bug eingeschlichen. Betrifft die aktuelle FW.

Wenn die Limitänderung über die Live Ansicht direkt gesetzt wird, ist alles OK. "12.5" % eingegeben und es wird 12.5% angezeigt und geregelt. Bei Watt ist es auch OK.

Wird die Änderung über MQTT geschrieben darf kein Trennzeichen gesetzt werden. Für 12.5% muss ich 125 senden.  Auch wenn der Wert als "400W" für die direkte Leistung geschrieben wird, passiert das gleiche und der WR regelt auf 40W runter.

Wenn es so wäre das kein Komma angegeben werde soll würde das ja fast jede Reglung bei jedem aushebeln. Da bei mir der WR dann runter regeln kann es kein Anzeigefehler sein.

Im Manual konnte ich keinen Hinweis dazu finden wie es sein sollte. In den Infos steht ja "Breaking Change". Ist das so gewollt?

knickohr, 2024-01-02:
Vielleicht . und , vertauscht ?

lumapu, 2024-01-02:
war nicht richtig implementiert. `0.8.40` kann das besser 😉

Ollipop030, 2024-01-02:
Bei mir macht die Nulleinspeisung, sieht dann so aus:

```
Jan 02 15:00:48 Pi-Hole python3[776]: 2024-01-02 15:00:48 INFO     Ahoy: Inverter "HM-300": setting new limit from 63 Watt to 63 Watt
```

In Ahoy kommt dann das:

```
0.8.40 - 3dd4997
---------------
15:00:51.087 I: (#1) sendControlPacket cmd: 0b
15:00:51.589 I: (#1) RX  34ms | 17 CH61 | d1 81
15:00:51.590 I: (#1) has accepted power limit set point 6.30 with PowerLimitControl 0
15:00:51.590 -----
15:00:52.592 -----
15:00:52.592 I: com loop duration: 1506ms
15:00:52.593 -----
```

Das ist aber doch nicht richtig oder? Wie sende ich denn über die API jetzt das korrekte Limit? Immer .0 anhängen, also 63.0?

lumapu, 2024-01-02:
Wert `*10`, musste ich in der WebUI auch machen:
https://github.com/lumapu/ahoy/blob/f2e1e536f52a7f21efddde1f505a53c0aff29601/src/web/html/visualization.html#L450-L454

blueline13, 2024-01-02:
@lumapu 

Das Problem scheint mit der 0.8.40 gelöst! Zumindest was die Optik angeht. Für die Leistung die Gergelt wird braucht es doch etwas mehr Licht. Ich werde morgen dann wieder schauen. Vielleicht gibt es noch eine neue Version bis es morgen wieder Hell wird, die die Ansicht in der "Webserial" korrigiert :wink:

Vielleicht hast du einen Tip für mich. Ich hatte beim dem Setzen von LIMITs manchmal das Gefühl das nicht immer alle WR das Limit erhalten(Check per Webserial weil die dort schlicht fehlten). Gibt es irgendwelche vorkehrung in der DTU, das gleiche Werte nicht mehrfach übertragen werden? Wenn es da nichts gibt, dann werde ich das weiter beobachten und einen neuen PR dazu aufmachen.

Wie immer kann ich nur sagen Danke für die rasante Implementierung, Bearbeitung und Lösung! Ist echt einmalig!!!

@Ollipop030 
So wie ich das eben probiert habe ist das senden via MQTT und Live möglich wie bisher auch, also ganze Werte ohne Komma ("33" = 33%). Das Komma braucht man nur wenn etwas genauer sein muss ("33.1" = 33.1%). Ich habe das Vergügen derzeit 4 Wechselrichter zur Nulleinspeisung zu überreden. Ohne Komma ist das fast kaum möglich ruhig zu bekommen.

Ollipop030, 2024-01-02:
> So wie ich das eben probiert habe ist das senden via MQTT und Live möglich wie bisher auch, also ganze Werte ohne Komma

Ich sende absolute Limits, da kommen immer nur 10% der gesendeten Leistung an. So wie @lumapu schon erwähnt hat, der Wert ist *10 zu rechnen.

lumapu, 2024-01-02:
@blueline13 es gibt keine Prüfung, ob das gleiche nochmal gesendet wurde. Es gibt sowohl in der API als auch bei MqTT ein Acknowledge vom Wechselrichter.

blueline13, 2024-01-02:
Einen wirklich Dank an dich @knickohr für die Zusammenfassung.

Ich kann alle deine genannten Punkte nur zu 100% bestätigen durch eigene Erfahrungen. Das kann man wirklich als absolute Grundlage für die Limit Steuerung verwenden!

Dann vielleicht meine Erfahrung die vielleicht eine kleine Erweiterung zu deinen Aussagen sind. Nach einem Jahr Erfahrung mit Limits.

Die wirklichen zwei Gebot für eine funktionierende Limit Regelung sind....

- Erst das neue Limit zu prüfen ob es schon aktiv ist und dann auf keinen Fall den gleichen Wert wieder schreiben. Das mag vielleicht mal gut gehen aber sicher nicht auf Dauer und man brauch sich dann auch nicht wundern wenn der Wechselrichter regelmäßig aussteigt und sich Tod stellt. Führte bei mir auch zu den gleiche Erscheinung wie der erste Punkt aus deiner Liste, als ich damals anfing. Also niemals blindes drauf schreiben von Werten und erst recht nicht im Sekundentakt, denn das mag er überhaupt nicht.

- Was genauso wichtig ist, sind die 5 Sekunde (je nach Anzahl der Wechselrichter und Zeiten die in der DTU für die Abfrage hinterlegt sind). Darunter zu gehen mach keine Sinn und das ganze wird dadurch nur Instabil und führt zu den von dir genannten Ausfällen.

Für mich habe ich folgendes festgelegt wie eine saubere Limitregelung funktionieren kann, das lief jetzt ein Jahr mit einem TSUN M1600 ohne ein Problem.

1) Immer erst das neue Limit prüfen ob es schon aktiv ist, wenn schon aktiv dann nichts machen.

2) Wenn kein Ack für einen neuen Wertes kommt, dann ca 20 Sekunden abfragen ob doch noch ein Ack kommt.

3) Wenn dann kein Ack kommt, einen Wert mit ein Abweichung erneut schreiben (Bisher 1%).

4) Wieder auf das Ack warten wie bei Punt 2. Bei mir hatte es dann meistens nach dem zweiten aller spätestens nach dem dritten neuen Wert immer geklappt.

Erfahrungen mit alten Wechselrichtern TSUN (HM1500) und der neuen HMS Serie. Die alte Generation ist deutlich "zickiger" mit dem Annehmen von neuen Werten. Die neuen habe ich erst seit einem Monat im Einsatz, doch selbst in der kurzen Zeit zeigen die sich deutlich robuster was das annehmen des ersten neuen Limits angeht.

Zu deiner Angabe mit den 0,1% Schritten. Als ich das gestern probiert hatte auf so kleine Änderungen zu testen, um bei einem fehlendem Ack einen geändert Wert zu schreiben, hatte das sehr gut funktioniert. Hatte das zwar mit Watt Werten gemacht, und das entsprach dann 0,1%.

Macht es nicht Sinn, für alle die auch mit den Limits arbeiten wollen, das ins Manual zu übernehmen als die "Goldenen Regeln"? Da ich von Anfang an mit den Limits gearbeitet habe, kannte ich schon alle Punkt aus Tickets heraus. So kann man es anderen leichter machen dort einzusteigen.

Ich werde dann morgen mit der 0.8.41 weiter testen :smiley:

knickohr, 2024-01-02:
Gut zu wissen das es in 0,1-Schritten auch geht.

Eins habe ich noch vergessen :  niemals ein Limit unter 2% setzen. Dann besteht Gefahr das der Inverter ganz runter fährt. Ist nicht schlimm, bis auf das das er dann ewig wieder zum hochkommen braucht und eben die ganzen Startroutinen durchläuft. Besser ist es, den Inverter per Soft auszuschalten. Dann geht er in den Standby, kommt aber sofort wieder hoch wenn man ihn soft einschaltet. Ob es genügt ihn mit einem Wert >2% zu aktivieren müßte noch getestet werden.

Ja, das letze 3 Posts geschriebene sollte ein eine Wiki rein. @DanielR92 oder @stefan123t ?

blueline13, 2024-01-03:
@lumapu 

Ich habe jetzt noch etwas weiter getestet. Als erstes die 0,1% sehen gut und das klappt sehr gut! DANKE DAFÜR!

Doch beim setzen des Limit ist mir das aufgefallen, das dass ACK von der DTU sehr arg verzögert zum MQTT übertragen wird. Dazu ein Log mit zwei Beispielen.

Beispiel1:
1) 13:52:05.805 Senden des Limit per MQTT

2) 13:52:10.586 Empfang des Ack vom Wechselrichter.

3) Dann scheint der WR 2 Loops nicht zu antworten. (Nicht immer, aber häufiger)

4) 14:52:26.xxx Trudelt das Ack von der DTU erst im MQTT ein.

In dem Beispiel sind das 15 Sekunden "Totzeit" bevor ich das ACK auswerten kann obwohl es in der DTU vorhanden ist. Ich habe "Totzeiten" zwischen 7-20 Sekunden gehabt. Sicher ist das kein Problem für die Leute die nur eine Wechselrichter haben und auch nur am Abend mal eine Limit vorgeben wollen. Doch bei mehreren und sequentieller Verarbeitung(ja ich baue um auf parallel) werden das Zeiten dann sehr lang sind und schwer für einer Nulleinspeisung mit 15Sek(besser 10 Sek) zu gebrauchen.

Ich meine das vor der 0.8.x es mal anders war, ich kann mich nicht wirklich erinnern das es früher so lange gebraucht hatte bis das ACK da war. Sicher manchmal mag der Wechselrichter nicht reagieren oder länger brauchen doch ich habe auch Zeiten von nur 1 Sekunde gesehen (Beispiel2) bis das Ack da war und im zweiten Beispiel kam dann erst bei 13:45:21.xxxx das Ack von der DTU im MQTT an, also 16 Sekunden später.

Ich habe die Zeiten mit iobroker und zweiten MQTT Expolrer geprüft. In meiner DTU ist der Intervall für MQTT senden auf "0".

Es wäre eine echte Hilfe wenn das Ack von der DTU als "Priopaket" per MQTT gesendet werden könnte sobald es vom Wechselrichter empfangen wurde. Ist es möglich das mit einzubauen? 

[Limit Regelung.txt](https://github.com/lumapu/ahoy/files/13819721/Limit.Regelung.txt)

knickohr, 2024-01-03:
Das kürzeste was ich über MQTT hatte, waren tatsächlich diese 5s.

Ja, das Setzten und die Ausgabe über MQTT verzögert mit Sicherheit den Ablauf. Deshalb ist es wohl besser so etwas direkt in die DTU zu integrieren, was ja teilweise schon geschehen ist. Und nur noch die informative Ausgabe des vom Inverter zurück gelieferten Wert über MQTT anzuzeigen.

blueline13, 2024-01-03:
@knickohr 

Ahhh Ok was meinst du da genau? Was gibt es? Kann ich da schon irgendetwas testen?

knickohr, 2024-01-03:
Na, das ganze MQTT muß ja erst durch die DTU durch und aufbereitet werden. Erst dann wird es zum zum Broker/Inverter gesendet und umgekehrt.

Inverter - DTU - Broker - Deine Regelung (HA oder Node-Red oder wasauchimmer)

Da bleiben sicher Sekunden auf der Strecke.

blueline13, 2024-01-04:
Ich habe das ganze jetzt nochmal probiert und kann sagen das es ausschließlich an der verwendeten FW liegt.

1) 0.7.36
Hier wird unmittelbar(max 2 Sek eher 1 Sekunde Verzögerung) wenn im Webserial das ACK vom Wechselrichter in der DTU an kommt, an den MQTT weitergegeben.

2) 0.8.10
Hier gibt es durchschnittlich eine Verzögerung von ca 10 Sekunden nach dem eintreffen des ACK vom Wechselrichter in der DTU bis es an den MQTT weitergegeben wird.

3) 0.8.42
Hier gibt es durchschnittlich eine Verzögerung von ca 15 Sekunden nach dem eintreffen des ACK vom Wechselrichter in der DTU bis es an den MQTT weitergegeben wird.

Ich habe das Ganze auch noch mit der API getestet und es sind die gleichen Zeit an Verzögerung die zwischen eintreffen des ACK im Webserial bis zum Weitereichen an API/MQTT. Diesmal habe ich es sogar nur mit einem Inverter auf einem ESP8266 probiert TSUN M1600.

Das was ich mit dem ESP32 mit 4 Invertern beobachten konnte, dass es mit den HMS Typen schneller geht, wenn das ACK vom Wechselrichter eintrifft, bis zur Weitergabe an API/MQTT.

Bestätigt meine Erinnerung das es mit den alten Version echt rasant war im Gegensatz zu den aktuellen Version.

@lumapu
Gibt es Hoffnung wieder die Performance der alten Versionen zu bekommen? Soll ich das auf die Wunschliste mit drauf packen?

lumapu, 2024-01-04:
ich sehe es eher als bug. Das Ack ist genau für eine Nulleinspeisung wahnsinnig wichtig. Auf der anderen Seite haben wir jetzt eine 'fixere' Kommunikaitonsschleife, die hauptsächliche Ursache sein dürfte. Ich versuche es bei mir nachzustellen und hoffenltich bald Abhilfe schaffen zu können.

blueline13, 2024-01-11:
Habe heute kurz mit der 0.8.51 getestet. Das sieht für mich richtig gut aus! Vom setzen des Limits bis zum eintreffen des ACK max 1-2 Sekunden, eher unter einer Sekunde. Für mich ist das mehr als ausreichend! Daher mache ich das hier auch als gelöst zu.

Da kann ich nur sagen HERZLICHEN DANK an das gesamte Team!

stefan123t, 2024-01-11:
Eine Anmerkung noch zu den Limits für mehrere Imverter/Wechselrichter. 

Die Power Limit Commands werden mW als Prio Commands von AhoyDTU an den WR geschickt. 
Wenn man aber zu schnell die MQTT Commands über den Broker/Home Automation sendet, könnte es mE passieren dass man das vorher (zB an einen anderen WR) zu sendende Limit überschreibt und nur das neueste Limit in der TX Queue landet.

@lumapu oder haben wir hier mit Locking/Semaphoren vorgesorgt und die Priority Pipeline ist auch länger als nur das 1 Command an das ich mich erinnern kann ?

lumapu, 2024-01-11:
jedes command kann Priorität haben. Es gibt keine Sperre, kann man drüber nachdenken