lumapu/ahoy #1316: hoymills in der leistung reglen, nur tag und nacht modus

3aau, 2024-01-02:
moin, 
schönes projekt.
ich bräuchte die möglichkeit den HM per MQTT (ohne extra server) zwischen tag ( P max) und nacht (Pmin) zu steuern.
ab sonnenaufgang  Pmax und ab sonnenuntergang Pmin. die einstellungen müssen nicht permanent sein.
schön wäre auch wenn man die zeiten frei eingeben könnte.
hintergrund ist das der HM im nachtmodus aus einer batterie gespeist wird.
hat jemand sowas schon gemacht?

knickohr, 2024-01-02:
-> #1199 
Wunschliste

Andy-Op, 2024-01-12:
Ich hätte hier auch noch einen Wunsch: Dass mit meiner Nachteinspeisung nimmt Formen an. Permanentes Leistungslimit HM-300 = 100 Watt (Grundlastkompensation). An guten Tagen hätte ich noch mehr im Akku. Da würde ich gerne für 2h das Limit auf 200 Watt hochsetzten. Ich stelle mir so was wie einen Booster-Knopf vor, der zeitlich abläuft. Lässt sich das permanente Leistungslimit temporär nach oben überschreiben? Vielleicht ist das ja ein Abfallprodukt aus den hiesigen Überlegungen.

lumapu, 2024-01-12:
ja das permanente lässt sich auch nach oben korrigieren. Bisher sehe ich das hier nur offen zur Diskussion