lumapu/ahoy #1317: Gehäuse zum Drucken

Edibo63, 2024-01-02:
Hallo,
habe ein Gehäuse in das NFR24 mit ext. Antenne, ESP32 und W5500 reinpasst. Vielleicht etwas gross geraten dafür sind alle Pins leicht zugänglich.
Wenn sowas hilfreich ist stelle ich gern die stl Dateien zur Verfügung.

knickohr, 2024-01-02:
[image]

Edibo63, 2024-01-02:
[image]
[image]

jolo1581, 2024-01-04:
Warum stellst du es nicht auf printables oder thingiverse?

Dann können sich alle interessierten sich das dort herunterladen. Viele andere Gehäuse für Ahoy und openDTU sind dort schon vorhanden.

hans341, 2024-01-04:
Hi, genau so ein Gehäuse suche ich.
Wäre klasse, wenn Du die Datei zur verfügung stellst - dann könnte ich mir jemanden suchen, der es drucken kann.... :-)

Edibo63, 2024-01-05:
Ist da zu finden.
Wird aber erst in 24h freigeschaltet

https://www.thingiverse.com/thing:6417346

Henrik-89, 2024-01-08:
Andere aus diesem Github Projekt haben ihre 3D Druck Dateien unter diesem Pfad veröffentlicht. Vielleicht ist das eine zusätzliche Option neben thingiverse?

https://github.com/lumapu/ahoy/tree/main/tools/cases

Copro, 2024-02-19:
Habe einen Pull Request für @Edibo63 erstellt, der die STL Dateien für cover und bottom samt den Bildern von Thingyverse auch hier unter tools\cases\ESP32_W5500_NRF24_ext_ant_Case bereitstellt, wie @Henrik-89 vorgeschlagen hat.

stefan123t, 2024-10-29:
@lumapu um das Project (Stichwort github `.pack` Datei) leichter zu machen sollten wir die ganzen Binär Daten wie Bilder, PDFs, DOCx etc. evtl. in ein eigenes Project auslagern. Da würden sich die ganzen PCBs und STLs und BMPs natürlich ganz besonders anbieten.