lumapu/ahoy #1318: [Bug] 0.8.42 Inverter loss rate faulty

fsck-block, 2024-01-03:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.42

### Github Hash

0d10d19

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

build: esp8266-prometheus

### Error description

Die Werte der Radio statistics für "Inverter Loss rate" 65529 von 18 passen bei mir wohl nicht wirklich.

[image]

lumapu, 2024-01-03:
ja das Problem ist, dass der Zähler immer weiter läuft und wir irgendwann an die Grenze von 16 bit kommen. Da gibt es bis dato noch keine Logik, die das abfängt. Muss man in `hmInverter.h` im unteren drittel ändern.
https://github.com/lumapu/ahoy/blob/0d10d19b30ce8d48d6215bda352821fb8c3b544a/src/hm/hmInverter.h#L599-L602

knickohr, 2024-01-03:
Oh Gott, da habe ich wieder was ins Rollen gebracht 😇
Ist mir aber schon bei den inoffiziellen Testversionen im Log aufgefallen das dort hin und wieder mal -1 stand. Hab das nur ignoriert weil es eben nur im Log war und es Testversionen sind. Nun ja, da es jetzt übt in den DEVs eingebaut ist, ist das unschön 😱

beegee3, 2024-01-04:
wenn `rxCnt` kleiner als `mIvRxCnt` ist, hat es einen "overflow" gegeben und man müsste zur Kompensation in der Differenz `rxCnt - mIvRxCnt` eigentlich 65536 hinzuaddieren. Um es `uint16_t` konform zu machen heißt das:
`rxCnt + ((uint16_t)65535 - mIvRxCnt) + 1`
Analoges gilt für `txCnt` und `mIvTxCnt`.
Damit sollte folgender Code das Problem lösen (auf die schnelle, geht vielleicht auch eleganter):
```
                    radioStatistics.ivSent = mDtuTxCnt;
                    if (rxCnt < mIvRxCnt)   // overflow
                        radioStatistics.ivLoss = radioStatistics.ivSent - (rxCnt + ((uint16_t)65535 - mIvRxCnt) + 1);
                    else
                        radioStatistics.ivLoss = radioStatistics.ivSent - (rxCnt - mIvRxCnt);
                    if (txCnt < mIvTxCnt)   // overflow
                        radioStatistics.dtuSent = txCnt + ((uint16_t)65535 - mIvTxCnt) + 1;
                    else
                        radioStatistics.dtuSent = txCnt - mIvTxCnt;
                    radioStatistics.dtuLoss = radioStatistics.dtuSent - mDtuRxCnt;
```

lumapu, 2024-01-04:
danke @beegee3 habe es 1:1 übernommen 😊

closed 2024-02-17 by lumapu